networkupstools/nut #2844: all of these drivers: blazer_usb, nutdrv_qx, and blazer_ser,they didn't work for me

alhaddar, 2025-03-12:
Hello.
I have a Gtec PC615n device. I use it with an OpenWRT router. I used all of these drivers: blazer_usb, nutdrv_qx, and blazer_ser, but unfortunately, they didn't work for me.

When I use the lsusb command, I get this:

Bus 002, Device 002: ID 0665:5161 INNO TECH USB to Serial

Please help me find the solution.

Regards.

jimklimov, 2025-03-12:
Which NUT version is installed there? Is building your own an option?

A recent PR added some Gtec support to current code base; at least you could check on another computer (or VM with port pass-through) if the new NUT software likes your UPS better; then maybe engage in making a build for OpenWRT.

alhaddar, 2025-03-12:
The version is NUT version 2.8.0-3.
I actually used the program with a Raspberry Pi and used the nutdrv_qx driver, and it worked without any problems.
However, those who used the program with opwenwrt may not recognize the driver.
This information is available here.

root@OpenWrt:~# lsusb -v -d 0665:

Bus 002 Device 003: ID 0665:5161 INNO TECH USB to Serial
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0665
  idProduct          0x5161
  bcdDevice            0.02
  iManufacturer           1 INNO TECH
  iProduct                2 USB to Serial
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3 20100813
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              4 Sample HID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
          Report Descriptor: (length is 27)
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Main  ): Input, data= [ 0x82 ] 130
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Volatile Bitfield
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Main  ): Output, data= [ 0x82 ] 130
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

alhaddar, 2025-03-12:
This information is for the Raspberry Pi.
 lsusb -v -d 0665:

Bus 001 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x0665 Cypress Semiconductor
  idProduct          0x5161 USB to Serial
  bcdDevice            0.02
  iManufacturer           1 INNO TECH
  iProduct                2 USB to Serial
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          3
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              4
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

jimklimov, 2025-03-12:
Is the version also named "2.8.0-3" on OpenWRT? I thought bigger distributions did that, but not something embedded like this.

I see you've tried both serial and USB drivers; which NUT configs were involved? In particular, for serial mode experiments did you provide `port=/dev/ttyUSB0` or similar path to the serial-over-USB port device (if the OS made one for the link)? Or are there physical serial ports involved like `/dev/ttyS1`?

Can you please starting the driver programs on OpenWRT with higher debug verbosity, and post the results here, e.g. `nutdrv_qx -a UPSNAME -DDDDDD -d1`?

One thing I suspect here could be permissions: if the build (or packaging) for OpenWRT did not create and specify user/group accounts for NUT programs to use, and/or if there is no auto-configuration facility like udev/devd/..., so that the driver runs as `nobody` or even `nut` but that user has no permissions to access the device filesystem node. That bit could be rectified by adding `-u root` to experiments above (or `user=root` in `ups.conf`).

jimklimov, 2025-03-23:
Gentle bump? Did you manage to run the driver programs to collect their debug logs?

jimklimov, 2025-03-23:
Thinking some more of it, what is the CPU architecture there, especially regarding endianness? We had some possible issues addressed in #1024 (and a long stack of further PRs) for incorrect USB encoding/decoding, which is mostly due to vendors not reading the USB spec, but (in NUT versions before the fix) could also be due to wire-to-CPU bit order conversion issues.

This aspect should have been fixed in 2.8.0 and later.

I wonder if something of that fix actually misfired, and e.g. 2.7.4 would work better, as a data point? :-\

Debug logs needed wither way, though.