networkupstools/nut #2854: Persistent usbhid-ups[7209]: nut_libusb_get_report: Input/Output Error

grainsoflight, 2025-03-18:
NUT persistently output  usbhid-ups[7209]:

 nut_libusb_get_report: Input/Output Error though it appears to be running fine. 

[nut-debug-20250318181700.zip](https://github.com/user-attachments/files/19329744/nut-debug-20250318181700.zip)

jimklimov, 2025-03-19:
Looking at logs, you seem to be running NUT v2.8.0 variant. Can you please try recent versions, or master-branch builds? Maybe @desertwitch can help about that, as it seems you're running UNRAID...

grainsoflight, 2025-03-19:
Using on latest (nut-2.8.2-x86_64-7master.ssl31) gives the same errors

jimklimov, 2025-03-19:
And did you set `pollonly` in config as it recommends? Does it have any good/bad effect?

grainsoflight, 2025-03-19:
What syntax would I use to add that to the ups.conf

jimklimov, 2025-03-19:
Reading man pages and other docs generally helps :)
https://networkupstools.org/docs/man/usbhid-ups.html

In the section for the device config, add `pollonly` as a standalone line (it is a flag, so presence means enabled).

grainsoflight, 2025-03-19:
Then no, it doesnt work

grainsoflight, 2025-03-20:
Nevermind, it was not saving the flag because a setting needed to be changed, now it appears to be working

grainsoflight, 2025-03-20:
Spoke too soon Mar 19 19:03:29 cloud usbhid-ups[270750]: nut_libusb_get_report: Input/Output Error

jimklimov, 2025-03-20:
Can you also add `debug_min=6` for a while, to collect details about how it initializes the device driver and at what point it complains. (Later disable the logging, so it would not thrash your storage)

I wonder if this is related to fixes in master after 2.8.2, like #2604 (so the error is not so much of communications as of interpreting the data as usable).

grainsoflight, 2025-03-20:
Of course now that debugging is is on I dont get the error

grainsoflight, 2025-03-20:
The log is just this repeating now

> Mar 19 19:26:33 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "quiet"
Mar 19 19:26:33 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 29 bytes to socket 5 succeeded (ret=29): SETINFO driver.state "quiet"
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "updateinfo"
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 34 bytes to socket 5 succeeded (ret=34): SETINFO driver.state "updateinfo"
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D1:331365] upsdrv_updateinfo...
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D1:331365] Not using interrupt pipe...
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D1:331365] Quick update...
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D4:331365] Entering libusb_get_report
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D3:331365] Report[get]: (3 bytes) => 16 ff ff
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 1966020, PhyMin = -60, LogMax = 32767, LogMin = -1
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Unit = 00001001, UnitExp = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.Output.DelayBeforeStartup, Type: Feature, ReportID: 0x16, Offset: 0, Size: 16, Value: -60
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D4:331365] Entering libusb_get_report
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D3:331365] Report[get]: (3 bytes) => 15 ff ff
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 1966020, PhyMin = -60, LogMax = 32767, LogMin = -1
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Unit = 00001001, UnitExp = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.Output.DelayBeforeShutdown, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: -60
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D4:331365] Entering libusb_get_report
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D3:331365] Report[get]: (2 bytes) => 0b 11
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Unit = 00000000, UnitExp = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] hu_find_infoval: found online (value: 1)
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] process_boolean_info: online
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D3:331365] Report[buf]: (2 bytes) => 0b 11
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Unit = 00000000, UnitExp = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x0b, Offset: 1, Size: 1, Value: 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] hu_find_infoval: found !chrg (value: 0)
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] process_boolean_info: !chrg
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D3:331365] Report[buf]: (2 bytes) => 0b 11
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Unit = 00000000, UnitExp = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] hu_find_infoval: found !dischrg (value: 0)
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] process_boolean_info: !dischrg
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D3:331365] Report[buf]: (2 bytes) => 0b 11
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Unit = 00000000, UnitExp = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x0b, Offset: 3, Size: 1, Value: 0
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] hu_find_infoval: found !lowbatt (value: 0)
Mar 19 19:26:35 cloud usbhid-ups[331365]: [D5:331365] process_boolean_info: !lowbatt

grainsoflight, 2025-03-20:
Ok, it finally errored

> Mar 19 19:32:01 cloud usbhid-ups[331365]: [D5:331365] hu_find_infoval: found !lowbatt (value: 0)
Mar 19 19:32:01 cloud usbhid-ups[331365]: [D5:331365] process_boolean_info: !lowbatt
Mar 19 19:32:01 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "quiet"
Mar 19 19:32:01 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 29 bytes to socket 5 succeeded (ret=29): SETINFO driver.state "quiet"
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "updateinfo"
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 34 bytes to socket 5 succeeded (ret=34): SETINFO driver.state "updateinfo"
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D1:331365] upsdrv_updateinfo...
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D1:331365] Not using interrupt pipe...
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D1:331365] Quick update...
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D4:331365] Entering libusb_get_report
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D3:331365] Report[get]: (3 bytes) => 16 ff ff
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 1966020, PhyMin = -60, LogMax = 32767, LogMin = -1
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] Unit = 00001001, UnitExp = 0
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.Output.DelayBeforeStartup, Type: Feature, ReportID: 0x16, Offset: 0, Size: 16, Value: -60
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D4:331365] Entering libusb_get_report
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D3:331365] Report[get]: (3 bytes) => 15 ff ff
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] PhyMax = 1966020, PhyMin = -60, LogMax = 32767, LogMin = -1
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] Unit = 00001001, UnitExp = 0
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] Exponent = 0
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D2:331365] Path: UPS.Output.DelayBeforeShutdown, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: -60
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D4:331365] Entering libusb_get_report
**Mar 19 19:32:03 cloud usbhid-ups[331365]: nut_libusb_get_report: Input/Output Error**
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D1:331365] Can't retrieve Report 0b: Operation not permitted
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "reconnect.trying"
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 40 bytes to socket 5 succeeded (ret=40): SETINFO driver.state "reconnect.trying"
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "quiet"
Mar 19 19:32:03 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 29 bytes to socket 5 succeeded (ret=29): SETINFO driver.state "quiet"
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "updateinfo"
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 34 bytes to socket 5 succeeded (ret=34): SETINFO driver.state "updateinfo"
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D1:331365] upsdrv_updateinfo...
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D1:331365] Got to reconnect!
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D5:331365] send_to_all: SETINFO driver.state "reconnect.trying"
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D6:331365] send_to_all: write 40 bytes to socket 5 succeeded (ret=40): SETINFO driver.state "reconnect.trying"
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D4:331365] Closing comm_driver previous handle
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D4:331365] ===================================================================
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D4:331365]  device has been disconnected, try to reconnect
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D4:331365] ===================================================================
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D4:331365] Opening comm_driver ...
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] Checking device 1 of 13 (174C/3074)
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - VendorID: 174c
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - ProductID: 3074
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Manufacturer: Asmedia
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Product: ASM107x
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Serial Number: unknown
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Bus: 002
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Bus Port: 003
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Device: 002
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - Device release number: 0200
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] Trying to match device
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] match_function_subdriver (non-SHUT mode): matching a device...
**Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID**
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] Device does not match - skipping
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] Checking device 2 of 13 (1D6B/0003)
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D1:331365] nut_libusb_open: invalid libusb bus number 0
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - VendorID: 1d6b
Mar 19 19:32:05 cloud usbhid-ups[331365]: [D2:331365] - ProductID: 0003

jimklimov, 2025-03-20:
Aha, thanks. So the message is not all too benign - the driver reconnects afterwards.
* The "match ... failed" that you highlighted is not a problem, or at least not that instance of it - the driver walks all (13) devices libusb sees, and quickly discards those that did not match configuration or built-in criteria like vendor and product IDs

Seeing how you have a CPS device, check these Wiki pages:
* https://github.com/networkupstools/nut/wiki/Troubleshooting-eventual-disconnections-(Data-stale)
* https://github.com/networkupstools/nut/wiki/CyberPower-Systems-(CPS)-know-how

Notably, you may benefit from a more frequent USB polling so the UPS controller does not fall asleep.

Also checked, this does not seem to be what #2604 fixed after all, I think...

grainsoflight, 2025-03-20:
Unfortunately changing the polling interval didnt seem to help, unless I dont have a concept of what is should be. I tried 0.2, 0.5 and 1 second

jimklimov, 2025-03-21:
IIRC drivers' default is 30 sec, so reduced values around 5-10 sec should be good for CPS... assuming *that* is the issue leading to reconnects.

mercurytoxic, 2025-05-11:
Thanks for an awesome set of tools. I am facing the same issue with a new Eaton 5P.
```
$upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
USB communication driver (libusb 1.0) 0.47
Using subdriver: MGE HID 1.46
```
Out of the blue, after running fine for almost  one day:
```
journalctl -fu nut-driver@eaton.service
May 10 12:50:16 luisaranguren systemd[1]: Starting Network UPS Tools - device driver for NUT device 'eaton'...
May 10 12:50:31 luisaranguren nut-driver@eaton[3800577]: Using subdriver: MGE HID 1.46
May 10 12:50:31 luisaranguren nut-driver@eaton[3800577]: Network UPS Tools - Generic HID driver 0.53 (2.8.2)
May 10 12:50:31 luisaranguren nut-driver@eaton[3800577]: USB communication driver (libusb 1.0) 0.47
May 10 12:50:33 luisaranguren nut-driver@eaton[3800577]: Running as foreground process, but saving a PID file anyway
May 10 12:50:33 luisaranguren usbhid-ups[3800577]: Running as foreground process, but saving a PID file anyway
May 10 12:50:33 luisaranguren systemd[1]: Started Network UPS Tools - device driver for NUT device 'eaton'.
May 10 12:50:33 luisaranguren nut-driver@eaton[3800577]: upsnotify: logged the systemd watchdog situation once, will not spam more about it
May 10 12:50:33 luisaranguren usbhid-ups[3800577]: upsnotify: logged the systemd watchdog situation once, will not spam more about it
May 10 12:53:05 luisaranguren nut-driver@eaton[3800577]: sock_connect: enabling asynchronous mode (auto)
May 10 12:53:05 luisaranguren usbhid-ups[3800577]: sock_connect: enabling asynchronous mode (auto)
May 11 11:01:16 luisaranguren nut-driver@eaton[3800577]: nut_libusb_get_report: Input/Output Error
May 11 11:01:16 luisaranguren usbhid-ups[3800577]: nut_libusb_get_report: Input/Output Error
May 11 11:01:16 luisaranguren nut-driver@eaton[3800577]: Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
May 11 11:01:16 luisaranguren usbhid-ups[3800577]: 
                                                   Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
May 11 11:01:17 luisaranguren nut-driver@eaton[3800577]: libusb1: Could not open any HID devices: insufficient permissions on everything
```

I added pollonly to the usb.conf
```
[eaton]
 driver = usbhid-ups
 port = auto
 desc = "Eaton 5P 1550i AU G2"
 pollonly
```
and restarted the service `systemctl restart nut-driver@eaton.service` which seems to have worked so far.
```
May 11 11:11:56 luisaranguren systemd[1]: Stopping Network UPS Tools - device driver for NUT device 'eaton'...
May 11 11:11:56 luisaranguren nut-driver@eaton[3800577]: Signal 15: exiting
May 11 11:11:56 luisaranguren usbhid-ups[3800577]: Signal 15: exiting
May 11 11:11:57 luisaranguren nut-driver@eaton[501726]: fopen /run/nut/usbhid-ups-eaton.pid: No such file or directory
May 11 11:11:57 luisaranguren nut-driver@eaton[501726]: Network UPS Tools - UPS driver controller 2.8.2
May 11 11:11:57 luisaranguren systemd[1]: nut-driver@eaton.service: Deactivated successfully.
May 11 11:11:57 luisaranguren systemd[1]: Stopped Network UPS Tools - device driver for NUT device 'eaton'.
May 11 11:11:57 luisaranguren systemd[1]: nut-driver@eaton.service: Consumed 13.717s CPU time, 4.7M memory peak, 300K memory swap peak.
May 11 11:11:57 luisaranguren systemd[1]: Starting Network UPS Tools - device driver for NUT device 'eaton'...
GMay 11 11:12:00 luisaranguren nut-driver@eaton[501754]: Using subdriver: MGE HID 1.46
May 11 11:12:00 luisaranguren nut-driver@eaton[501754]: Network UPS Tools - Generic HID driver 0.53 (2.8.2)
May 11 11:12:00 luisaranguren nut-driver@eaton[501754]: USB communication driver (libusb 1.0) 0.47
May 11 11:12:01 luisaranguren nut-driver@eaton[501754]: Running as foreground process, but saving a PID file anyway
May 11 11:12:01 luisaranguren usbhid-ups[501754]: Running as foreground process, but saving a PID file anyway
May 11 11:12:01 luisaranguren systemd[1]: Started Network UPS Tools - device driver for NUT device 'eaton'.
May 11 11:12:01 luisaranguren nut-driver@eaton[501754]: upsnotify: logged the systemd watchdog situation once, will not spam more about it
May 11 11:12:01 luisaranguren usbhid-ups[501754]: upsnotify: logged the systemd watchdog situation once, will not spam more about it
May 11 11:12:01 luisaranguren nut-driver@eaton[501754]: sock_connect: enabling asynchronous mode (auto)
May 11 11:12:01 luisaranguren usbhid-ups[501754]: sock_connect: enabling asynchronous mode (auto)
```

Will followup if the issue comes back.

jsebt, 2025-08-11:
I am also experiencing this SMART1500LCD and "pollonly" just causes a loop of USB disconnects (but the UPS data still comes in seemingly unaffected)

rockymtngeek, 2026-09-26:
I'm seeing what appears to be the same issue on NUT 2.8.5 under Unraid with a CyberPower CP1500PFCLCDa (VID:PID 0764:0601). I noticed the issue is currently tagged as impacting 2.8.0 and 2.8.2, so I wanted to confirm that I'm still seeing the same behavior with 2.8.5.

System/driver details:
```
NUT version:               2.8.5
driver.name:               usbhid-ups
driver.version.data:       CyberPower HID 0.85
driver.version.usb:        libusb-1.0.30 (API: 0x0100010A)
driver.parameter.pollfreq: 12
driver.parameter.pollinterval: 2
UPS:                       CyberPower CP1500PFCLCDa
VID:PID:                   0764:0601
OS:                        Unraid
```
The UPS otherwise appears to be working normally. upsc continues to return complete and sensible telemetry, ups.status remains OL, and the usbhid-ups process stays running.

I also checked the kernel log for the UPS USB device. It enumerated normally at boot and I am not seeing USB disconnect/reconnect events, resets, descriptor errors, or xHCI errors associated with it while these messages occur.

The errors are frequent. On the current boot I counted over 8,000 occurrences of:
```
nut_libusb_get_report: Input/Output Error
```
I temporarily changed the running driver's driver.debug to level 5 to see if I could identify a particular HID transaction associated with the error.

One failure was captured as:
```
refresh_report_buffer: investigating path: 00840004.0084001a.00840040
Entering libusb_get_report
Can't retrieve Report 0e: Success
```
That path is subsequently identified by the driver as UPS.Input.ConfigVoltage.

What's interesting is that Report 0x0e is not permanently unreadable. On the next full polling cycle, about 14 seconds later, the same report was retrieved successfully:
```
refresh_report_buffer: investigating path: 00840004.0084001a.00840040
Report[get]: (2 bytes) => 0e 78
Path: UPS.Input.ConfigVoltage, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 8, Value: 120
```
I continued capturing for several polling cycles and Report 0x0e repeatedly returned successfully as 0e 78 / 120 V after that. So at least in this case, this does not appear to be an unsupported HID report that fails every time. Individual GET_REPORT requests appear to fail intermittently while later requests for the same report work normally.

The next report after the failed 0x0e request also completed normally. For example, UPS.Input.Voltage (ReportID 0x0f) returned 122 V, and the rest of the UPS telemetry continued updating.

From what I can see, the UPS never actually loses communication when these errors occur. It looks more like an individual HID report request occasionally fails, then the same report works normally on the next polling cycle.

I still have the debug-level-5 output available if a larger section around the failure or any additional diagnostics would be useful.

rockymtngeek, 2026-10-06:
One additional observation since my comment above:
Today the Unraid NUT plugin restarted the NUT stack. After that restart, ```usbhid-ups``` came back online normally, but ```upsc``` was missing the input/output/load telemetry that had previously been available:
```
input.voltage
input.voltage.nominal
output.voltage
ups.load
ups.power
ups.realpower
```
Battery data and ups.status: ```OL``` were still present, and the driver continued running.
While it was in this state I enabled debug level 5. A full update successfully polled the reports that were available, but I did not see any ```UPS.Input.*``` or ```UPS.Output.*``` paths in the capture.

I then restarted NUT again, without rebooting the host, reconnecting USB, or changing the configuration. The complete telemetry 
immediately returned:
```
input.voltage: 122.0
input.voltage.nominal: 120
output.voltage: 122.0
ups.load: 12
ups.power: 123
ups.realpower: 120
```
So the same NUT 2.8.5 / CyberPower HID 0.85 instance against the same CP1500PFCLCDa ```(0764:0601)``` produced two different results across driver initializations.
I didn't have debug enabled during the bad initialization, so I can't confirm what happened during discovery. But given the intermittent GET_REPORT behavior in my previous comment, I thought this might be useful additional information.