networkupstools/nut #2861: IPPON Back Power Pro II 500

individ2016, 2025-03-25:
Hello all!

Need help.. i am using Truenas Scale last version with Network UPS Tools - 2.8.0..

nut-scanner -U shows nothing, but dmesg | grep -i usb shows me:

Mar 25 01:53:28 servicehub kernel: usb 3-3: new full-speed USB device number 6 using xhci_hcd
Mar 25 01:53:28 servicehub kernel: usb 3-3: New USB device found, idVendor=0483, idProduct=a430, bcdDevice= 2.00
Mar 25 01:53:28 servicehub kernel: usb 3-3: New USB device strings: Mfr=3, Product=1, SerialNumber=2
Mar 25 01:53:28 servicehub kernel: usb 3-3: Product:   Back Power Pro II 500
Mar 25 01:53:28 servicehub kernel: usb 3-3: Manufacturer: UPS Manufacturer 
Mar 25 01:53:28 servicehub kernel: usb 3-3: SerialNumber: 00000000000000000000
Mar 25 01:53:28 servicehub kernel: hid-generic 0003:0483:A430.0005: hiddev0,hidraw0: USB HID v1.11 Device [UPS Manufacturer    Back Power Pro II 500] on usb-0000:00:14.0-3/input0

so i tried to use driver=blazer_usb or driver=usbhid-ups.. tried to use options 
vendorid = 0483
productid = a430

nothing works.. maybe u have some advice for me.. 

Thank you in advance!

jimklimov, 2025-03-25:
#2312 (NUT v2.8.2) should solve it

individ2016, 2025-03-25:
I think i cant install 2.8.2 couse it is a part of Truenas system, but i'll think about it all. Maybe i can compile driver from that thread.. Thank you!

jimklimov, 2025-03-25:
Better not from "that thread" but from the release tag or current `master`. Now that I'm more online, can link to Wiki:
* https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests
* https://networkupstools.org/docs/user-manual.chunked/_build_prerequisites_to_make_nut_from_scratch_on_various_operating_systems.html

individ2016, 2025-04-02:
Finaly i've found a solution:
https://hub.docker.com/r/instantlinux/nut-upsd

I've created new application in TrueNAS Container and then setted up TrueNAS NUT service in slave-mode and it works!
Thank you again!)

closed 2025-04-11 by individ2016