networkupstools/nut #2863: Information request - UPS not reachable standard behavior

RikyPlaza, 2025-03-26:
The request is over Phoenix Contact driver.
On some models (like QUINT4 or TRIO-2G) the USB-serial converter is placed inside the UPS, on other models (like QUINT or TRIO) the same converter is placed inside the UPS cable.
On the models where the converter is inside the cables if the UPS is not powered or not connected but the cable is connected to the NUT machine the modbus communication can be opened but all the read functions will fail. 

From my point of view this situation should stop the UPS aswell, like if the connect function fails but i think it is better to discuss on how to manage it.

RikyPlaza, 2025-03-26:
I'm opening this issue just because if the modbus communication is opened without having a UPS replying to the read requests the driver will exit with error for "Unknown UPS part number". From my point of view it could be confusing for users.

jimklimov, 2025-03-26:
To clarify and double-check, is it possible to open Modbus communications when just the smart cable is plugged into the computer (with no UPS on the other side at all)? In that case, the cable is not just an USB-to-serial adapter, but also implements the Modbus-over-serial part.

For usual UPSes, we try to read the serial/usb/... `port` and if initial connection setup fails, the device is deemed unavailable at the specified location, driver exits with failure.

With a simple USB-to-serial adapter, whether in cable or a dongle, or with a native serial port, we try to talk to `/dev/ttyUSB0` or `/dev/ttyS1` etc., and that device node does not respond (or not in a way the driver understands) - same as above, not available => exit/fail.

Here... if the cable responds but the device is not usable, I suppose it is the same. Perhaps also for re-connections (if the device is re-plugged, or the USB bus is reset, after some time of successful dialog with the device), although those cases might loop indefinitely (or for some configurable count) instead, and wait for the device to re-appear.

If you have a way to discern the absence of the device vs. "administrative OFF" (communication works, power to load is not served), the corresponding `ups.status` can be set instead. After all, perhaps the device can be administratively turned on by a command via running driver at a later time.

jimklimov, 2025-03-26:
Strange, `fatalx()` should be instant (except in one test program which overrides the `exit()` method behavior to count the expected error cases).

RikyPlaza, 2025-03-26:
My bad: `fatalx()` is not called by `mrir()` when it fails. I'll update the driver to exit if the second `mrir()` called by `upsdrv_initups()` fails. If you agree on this behavior of course.

jimklimov, 2025-03-26:
Yes, that sounds right - thanks!

RikyPlaza, 2025-03-27:
I've just implemented the behavior. I'll wait to have a UPS to test everything before opening the PR.
Thanks for the clarification :)

closed 2025-03-27 by RikyPlaza