networkupstools/nut #2864: Reverse engineering the Volt MicroUPS 1500 protocol

Jacajack, 2025-03-27:
Hi, I've been trying to communicate with a Volt MicroUPS 1500 UPS. The issue #2534 mentions that at least some of the Volt devices should work with `blazer_usb`.  Unfortunately, neither `blazer_usb` nor `nutdrv_qx` work in my case. The UPS simply does not respond and the `read()` call times out most of the time. [This Polish forum](https://www.elektroda.pl/rtvforum/topic4035334.html) also mentions that supposedly the manufacturer has changed the protocol at some point.  

I've decided to install that Viewpower software in a VM and eavesdrop the USB communication with wireshark. The exchange between these two looks like this:

```
[23.397946000] --> QPI
[23.498801000] <-- QPI
[24.933947000] --> QPI
[25.048189000] <-- QPI
[26.469912000] --> QPI
[26.573295000] <-- QPI
[28.005894000] --> QPI
[28.104543000] <-- QPI
[29.541877000] --> QPI
[29.647224000] <-- QPI
[31.077863000] --> QPI
[31.186222000] <-- QPI
[32.613844000] --> ^P004QPI
[32.727256000] <--
[34.149826000] --> ^P004QPI
[34.249285000] <--
[35.653811000] --> ^P004QPI
[35.759929000] <--
[41.381772000] --> ^P004QPI^P004QPI^P004QPIQPI
[41.488165000] <-- QPI
[42.917728000] --> QPI
[43.022976000] <-- QPI
[48.645665000] --> ^P004QPI^P004QPI^P004QPIQPI
[48.751342000] <-- QPI
[50.181674000] --> QPI
[50.296026000] <-- QPI
[51.717648000] --> QPI
[51.826604000] <-- QPI
[54.629599000] --> QPI�QPI�
[54.731579000] <--
[56.165580000] --> QPI
[56.279487000] <-- QPI
[59.109575000] --> ^P006QPI)\^P006QPI)\
[59.109909000] <-- )\
[60.549559000] --> QPI
[60.655376000] <-- QPI
[62.085516000] --> QPI
[62.187961000] <-- QPI
[63.621525000] --> QPI
[63.734792000] <-- QPI
[65.157481000] --> QPI
[65.264966000] <-- QPI
[66.693465000] --> I
[66.808900000] <-- I
[68.229446000] --> M
[68.330520000] <-- V
[69.765435000] --> QPI
[69.879289000] <-- QPI
[71.301411000] --> QS
[71.544630000] <-- 40.3 000.0 000 00.0 27.1 --.- 00001011
[72.965395000] --> F
[73.137439000] <-- #230.0 006 24.00 50.0
[74.565378000] --> QID
[74.671791000] <-- QID
[77.509372000] --> PBRPD03272025PBRPD03272025
[80.626565000] <-- 72025
[82.053293000] --> QS
[82.314460000] <-- (240.3 238.3 000.0 000 00.0 27.1 --.- 00001011
[83.749276000] --> F
[83.921518000] <-- #230.0 006 24.00 50.0
[85.349256000] --> QID
[85.453535000] <-- QID
[86.885242000] --> GM
[87.000301000] <-- GM
[88.421252000] --> QS
[88.684824000] <-- (240.3 238.3 000.0 000 00.0 27.1 --.- 00001011
[90.117231000] --> QWS
[90.222740000] <-- QWS
[91.653200000] --> QWS
[91.763315000] <-- QWS
[93.189172000] --> QWS
[96.304185000] <-- QWS
[97.733121000] --> QS
[97.998655000] <-- (240.3 240.3 000.0 000 00.0 27.1 --.- 00001011
[99.429100000] --> F
[99.607835000] <-- #230.0 006 24.00 50.0
[101.029083000] --> QID
[101.141892000] <-- QID
[102.565066000] --> GM
[102.667571000] <-- GM
[104.101049000] --> QS
[104.371969000] <-- (240.3 240.3 000.0 000 00.0 27.1 --.- 00001011
[105.797031000] --> QWS
[105.900989000] <-- QWS
[107.333040000] --> QWS
[107.441807000] <-- QWS
[108.869023000] --> QWS
[111.981443000] <-- QWS
[113.412971000] --> QS
[113.688213000] <-- (240.3 240.3 000.0 000 00.0 27.1 --.- 00001011
[113.688213000] --> F
```

All outgoing messages are `SET_REPORT` going to `1.8.0` and incoming messages are `URB_INTERRUPT` coming from `1.8.1`.

It seems that at the beginning Viewpower is also having issues talking to the UPS and is throwing some `^`  commands at it? I couldn't find anything like that in this repo. After some back and forth they seem to settle on commands which seem to be supported by `nutdrv_qx`.

I've started getting into the UPS communication lore just yesterday, so I just wanted to ask: has this protocol been seen before? In some other Chinese brands perhaps? It seems like it _could_ be supported by the `nutdrv_qx` driver with some quirks added. If that's desirable, I could try to prepare a patch when I have some time. Unless, of course, someone here recognizes this as an existing protocol.

jimklimov, 2025-03-27:
Yes, looks like a variant of prolific Megatec Qx protocol family (where x is a number like 1 or 6 based on main query).

Jacajack, 2025-03-29:
Right, thank you for the answer. I've spent some more time trying to communicate with it to learn two things. Firstly, the USB logic can retain an invalid state even for a very long time when the UPS is "powered off". And secondly, that it never responds to the first `Q1` request from `nut`. It starts sending responses after the second query, but sometimes they are mangled or incomplete. I think something goes wrong at the very beginning. I guess that just means more wiresharking for me...

jimklimov, 2025-03-29:
Check the protocols library on nut website, there were several findings and links on Q6 posted in recent months,.. maybe some are still in PR's for its repo.

I think a few subdrivers had comments about double-tapping Q1 during discovery or reconnection, maybe that's why.

desertwitch, 2025-05-04:
Judging from the logs provided, the UPS should be (more ore less fully) compatible with the `nutdrv_qx_voltronic-qs` subdriver.
You can try forcing the `nutdrv_qx` protocol with `protocol = voltronic-qs`: https://networkupstools.org/docs/man/nutdrv_qx.html

Please let us know if it worked out, ideally with logs if something fails!

Jacajack, 2025-05-05:
I've had some luck with the following options `-s volt -x port=auto -x protocol=megatec -x vendorid=0665 -x productid=5161 -x ignoresab=1 -x novendor=1 -x norating=1 -DDDDD` after I manually increased the timeout in `usb_interrupt_read()` to 10s in `nutdrv_qx.c` and added a loop to repeat the read call until it succeeds. The UPS appeared to understand the megatec protocol commands.

The communication was very unreliable, though. The UPS would only respond sometimes (to like 25-50% of the commands) without any distinct pattern. I don't think the nut driver can be blamed for that - it just looks like cheap and unreliable hardware to me. These seemingly random communication issues are what make the unmodified nut driver impossible to use in the first place. When I tried later, it refused to communicate with Viewpower either, hinting that this might be related to some invalid state retained by the UPS logic for a long time. 

I can give it another go and provide some more logs if that would help. I'm just wondering if our time is worth spending on that unit.

desertwitch, 2025-05-05:
> I've had some luck with the following options `-s volt -x port=auto -x protocol=megatec -x vendorid=0665 -x productid=5161 -x ignoresab=1 -x novendor=1 -x norating=1 -DDDDD` after I manually [...]

Hmm, also odd that while apparently the `voltronic` protocol seems to be used in communication with Viewpower (with its distinct `V` protocol and `QS` commands we can see in your communication dump) it also responds to the `megatec` protocol commands (and its `Q1`s). Maybe they use one protocol for the communication with Viewpower and the other for some kind of other (backwards) compatibility. The unstable communication probably doesn't help either getting a clear picture of how it's communicating...

At least for `voltronic-qs`, part of claiming the device involves detection of a `V` protocol, which your UPS does send:
```
[68.229446000] --> M
[68.330520000] <-- V
```
https://github.com/networkupstools/nut/blob/b03af2d4022a42b9fd5e93a412d35d35b8683c67/drivers/nutdrv_qx_voltronic-qs.c#L58-L60