networkupstools/nut #2872: Vertiv PST5 reporting incorrect status

hotsaucegrocer, 2025-04-02:
Hello,

A few weeks ago, we looked at a Vertiv PST5 and identified some configuration requirements that differentiate it from other USB HID UPSes. While it seems to be reporting more accurate information than before, it still reports "OL CHRG" while unplugged. Furthermore, it continues to report charge at 100%. After approximately three minutes, the UPS powers down and the server crashes. I have now experienced the same behavior with two different PST5 UPSes in an attempt to rule out hardware failure.

When the UPS is plugged in, powered on, and the host is powered on, CUPS shows that the battery is upwards of 90% full. Also, it shows UPS load as 0%. Even though the server attached is a small one (SuperMicro Super Server half-depth 1U), we should see UPS load above zero.

I have killpower set to 0, so my understanding is that NUT would not try to trigger a UPS poweroff in that case, but I also may be misunderstanding.

I'm not seeing anything in syslog that would suggest that NUT intentionally triggered a shutdown, but I am wondering if there is additional communication that needs to happen between NUT and the UPS to tell the UPS not to power down preemptively. For example, if the UPS is set to power off at 93% battery life (seems to be the power at which it usually goes down), is there a way to have NUT tell the UPS not to do so?

Thanks,
Jon

jimklimov, 2025-04-02:
For context, which OS and NUT version are involved?

Also, where did you "set killpower to 0"? I don't remember any such setting documented. For troubleshooting, you can set `SHUTDOWNCMD` in `upsmon.conf` though, e.g. to echo a message into a log file just to see it ran.

Not sure what you mean by CUPS - the print server? Does it report NUT status now?

Which tools do you query battery/load/status with - upsc? upslog? How often are they querying? What is the pollfreq in the driver - by default it could be between 5 to 30 sec in different versions, so if you unplug the UPS and look too soon, the info might be not yet seen or propagated...

hotsaucegrocer, 2025-04-08:
This is Ubuntu 22.04 with NUT 2.8.2 and a Vertiv PST5 UPS. The server is running Xen with approximately a dozen VMs. You had mentioned possible issues while running on a hypervisor, but this does not appear to be an issue as I've queried Xen and none of the VMs are using USB passthrough. 

Furthermore, the tests described here have been performed on TWO identical servers with TWO Vertiv PST5 UPSes. On the first server, the UPS was a few months old. On the second server, it's a brand new UPS that was purchased to rule out the possibility of a faulty UPS on the test system.

First of all, apologies for the CUPS reference. I meant to say that upsc. Here is the complete test as performed. I should note that the UPS is approximately 200 miles from me, so I am having someone at the physical location handle the testing.

1. Check UPS status with `upsc nutdev1` while UPS is plugged into mains. The UPS will report "OL CHRG" as the status, 100% battery capacity, 0% load (this is odd because the server that is running NUT is plugged into the load of the UPS).
2. Have on-site technician disconnect UPS from mains.
3. Watch NUT for regular updates with the following: `watch upsc nutdev1` which by default will run upsc every 2 seconds. I am aware that it will take up to 30 seconds for a valid driver update, but if I watch the output for long enough, I will expect to see an update.
4. I lose ssh connectivity to the server at approximately 2m30s and contact the technician at the remote site.
5. Technician confirms that the UPS has powered off.
6. I ask technician to plug the UPS back into mains, power the UPS on, and power the server on.
7. When server boots, check logs to see if there is any indication that NUT triggered the shutdown.
8. NUT did *not* trigger the shutdown. This was a hard power off.
9. Check UPS status with `upsc nutdev1`. The UPS reports "OL CHRG" as status, 93% battery capacity, 0% load.

We already determined that the PST5 is an odd UPS insofar as it has two USB endpoints and endpoint 0 seems to provide incomplete/insufficient/incorrect data whereas endpoint 1 appears to provide valid output. You offered me configuration adjustments to read from the latter USB endpoint.

At this point, it seems that there are two different things happening:
1. The UPS is abruptly shutting off at 93% battery capacity without receiving a command from NUT. This makes me think that the UPS's firmware shuts the UPS down without waiting for a command from NUT. If this is an accurate assumption, then it seems that we need some way to tell the UPS firmware not to do that. I'm not sure if it would be a one time write to NVRAM with a configuration utility or if it would be something along the lines of an initialization string that would need to be sent by the driver at connection.
2. The status seems to be "OL CHRG" regardless of whether plugged into mains or not.

hotsaucegrocer, 2025-04-09:
I'm working on getting Vertiv's proprietary software to install. It'll probably take a little while though. The system in question is a hypervisor with a number of point-of-sale VMs on it. Because of the nature of money and inventory handling of the VMs, the hypervisor is locked down more than stock Ubuntu. So, I'm working on getting the proprietary software to install in a non-standard (not in /usr) location. It may be a few days before I have an update. 

Thanks again for all help up to this point. Without your prior advice on the specific configuration for this UPS, I don't think we'd have gotten any data into NUT.

jimklimov, 2025-04-09:
Given the VMs, it may make sense to install vendor software in a VM with whatever OS they support well, and pass-through the USB port while having NUT driver on host disabled...

Also, just to clarify - why did their software get in picture? To compare what they see or do not see? To trigger a calibration (does `upscmd` offer anything of the sort on your device+driver combo, and if yes - does that work)?

Note that calibration can also cause outages, especially if the device's guess about battery health and charge vs. workload is in the wrong ballpark.

jimklimov, 2025-04-09:
Alternately (for vendor software to run in the host system directly) a chroot area may be an option... or an LXC-style container... NUT `docs/ci-farm-lxc-setup.txt` includes some hints on preparing those types of environments.

hotsaucegrocer, 2025-04-09:
On this particular UPS, `upscmd` has a pretty limited set of instant commands. The ouptut of `upscmd -l nutdev1` gives me:

- driver.killpower
- driver.reload
- driver.reload-or-error
- driver.reload-or-exit

Since I didn't see a way to call a calibration, I thought that the vendor software may be a way to trigger a calibration. Right now, I'm less concerned about a calibration causing an outage. To be clear, I'm working in my customer's development lab. However, they are a large retailer with several hundred stores that have this particular UPS deployed, so they are keen to get the UPS monitored so that it can cleanly shut down the server in a power outage.