networkupstools/nut #2874: [Support] Epyc ION UPS (lsusb listed as cyber power CP1500PFCLCD) ups.load: 0 also with devices connected

elios93, 2025-04-03:
Hello,

I’m reaching out after reading issues #2617 and #2362. I have correctly connected my Epyc ION UPS using the `usbhid-ups` driver. Here is the output from` upsc`:
```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: 1 
battery.runtime: 3600
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.7
battery.voltage.nominal: 12
device.mfr: 1 
device.model: 1000
device.serial:                 
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.2
input.transfer.high: 280
input.transfer.low: 160
input.voltage: 227.4
input.voltage.nominal: 230
output.frequency: 50.2
output.voltage: 22.6
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: 1 
ups.model: 1000
ups.productid: 0601
ups.realpower.nominal: 1200
ups.serial:                 
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```
However, I’ve noticed that the `ups.load` field always remains at `0`. Is there a way to determine where the correct value is stored and modify the driver accordingly?

Additionally, other fields might also be incorrect. For instance, I suspect that `output.voltage` should be multiplied by 10.

I’ve attached the output of the command `sudo /lib/nut/usbhid-ups -a myups -DDD`, which might provide more insight into the issue. Any help would be greatly appreciated. Let me know if any other files would be useful.

[usbhid-ups_output.txt](https://github.com/user-attachments/files/19592063/usbhid-ups_output.txt)

 Thank you in advance!

jimklimov, 2025-04-03:
Hm, in your log I see `0.937402	[D2] send_to_one: sending SETINFO ups.realpower.nominal "600"` so if it goes sour to 1200, that's at some later point :\

paolofaz, 2025-05-29:
I have the same issue (output voltage).

Another problem: the “ups.status”  is too slow (about 30 seconds) to’ switch from “on Line” to “on battery”. Is there a workaround to fix it?

jimklimov, 2025-05-30:
@paolofaz : check your `pollfreq` or similar timing settings - some of those do default to 30 sec. There might be a quicker turnaround with devices and NUT drivers that support "interrupts" (or wake up as soon as there is data in the read-buffer, as happens with serial ports etc.) - but in many cases they happen to work worse than regular polling and `pollonly` gets enabled. If you copy-pasted a configuration from some blog, and got that toggle blindly (or did need it in fact for your device), that might also be the cause of slower reactions.

jimklimov, 2025-05-30:
> I have the same issue (output voltage).

@paolofaz : did you reproduce this with current NUT (as in "a custom build of current code")?

jimklimov, 2025-05-30:
> Sorry my friend, I don't understand what I have to do :-)

For starters, read the issue you are adding posts to :) Just a few entries above was a link to Wiki article about building current NUT. Note that whatever you have via packaging of whatever OS is some older release that is set in stone that day and becomes older and more obsolete with every passing day...

adrianmihalko, 2026-09-29:
I have same problem.

```
root@proxmox:~# upsc nutdev1@127.0.0.1 
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 8830
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.3
battery.voltage.nominal: 12
device.mfr: CPS
device.model: UT850EG
device.serial:                 
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.pollonly: enabled
driver.parameter.port: auto
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.1
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
input.frequency: 499.0
input.transfer.high: 295
input.transfer.low: 167
input.voltage: 230.0
input.voltage.nominal: 230
output.voltage: 231.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.model: UT850EG
ups.productid: 0501
ups.realpower.nominal: 425
ups.serial:                 
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
root@proxmox:~# 
```

jimklimov, 2026-09-29:
@adrianmihalko : like earlier replies here - can you try with a custom build of current NUT (2.8.1 is old), see wiki link above for instructions.

adrianmihalko, 2026-10-01:
> [@adrianmihalko](https://github.com/adrianmihalko) : like earlier replies here - can you try with a custom build of current NUT (2.8.1 is old), see wiki link above for instructions.

Same, with latest git checkout:

```
root@proxmox:~# upsc nutdev1@127.0.0.1
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 8830
battery.runtime.low: 300
battery.status: 100%
battery.type: PbAcid
battery.voltage: 13.4
battery.voltage.nominal: 12
device.mfr: CPS
device.model: UT850EG
device.serial:
device.type: ups
driver.debug: 1
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 12
driver.parameter.pollinterval: 2
driver.parameter.pollonly: enabled
driver.parameter.port: auto
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0764
driver.state: updateinfo
driver.version: 2.8.5.1407-1407+g3154ca01c
driver.version.data: CyberPower HID 0.87
driver.version.internal: 0.74
driver.version.usb: libusb-1.0.28 (API: 0x0100010A)
input.frequency: 49.9
input.transfer.high: 295
input.transfer.low: 167
input.voltage: 234.0
input.voltage.nominal: 230
output.voltage: 234.0
unmapped.ups.powersummary.capacitygranularity1: 5
unmapped.ups.powersummary.capacitygranularity2: 10
unmapped.ups.powersummary.capacitymode: 2
unmapped.ups.powersummary.designcapacity: 100
unmapped.ups.powersummary.rechargeable: 1
ups.beeper.status: disabled
ups.delay.shutdown: 60
ups.delay.start: 120
ups.firmware: BF01701HBL1.y
ups.load: 0
ups.mfr: CPS
ups.model: UT850EG
ups.power.nominal: 0
ups.productid: 0501
ups.realpower.nominal: 425
ups.serial:
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
root@proxmox:~# upsc nutdev1@127.0.0.1 driver.version
Init SSL without certificate database
2.8.5.1407-1407+g3154ca01c
root@proxmox:~#
```

adrianmihalko, 2026-10-02:
I just queried NUT multiple times and now I see load is reported, but often 0. Maybe it is just a poll frequency issue? What is a recommended value?

```
[root@proxmox ~]$ upsc nutdev1@127.0.0.1 ups.load
Init SSL without certificate database
0
[root@proxmox ~]$ upsc nutdev1@127.0.0.1 ups.load
Init SSL without certificate database
13
[root@proxmox ~]$ upsc nutdev1@127.0.0.1 ups.load
Init SSL without certificate database
13
[root@proxmox ~]$ upsc nutdev1@127.0.0.1 ups.load
Init SSL without certificate database
13
[root@proxmox ~]$ upsc nutdev1@127.0.0.1 ups.load
Init SSL without certificate database
13
[root@proxmox ~]$
```