networkupstools/nut #2881: Forza FX-1500LCD (0665:5161) not supported by nutdrv_qx or blazer_usb on Windows/MSYS2 (unlike FX-2200LCD #2426)

andresg97, 2025-04-07:
## Summary

My Forza FX-1500LCD UPS, which uses the common USB ID 0665:5161, is not correctly detected or supported by either the `blazer_usb` or `nutdrv_qx` drivers in a recent NUT build on Windows using MSYS2. This is despite issue #2426 reporting success with the FX-2200LCD using the same VID/PID and the `nutdrv_qx` driver. The official Forza monitoring software *does* work with the UPS.

## Hardware Details

*   **UPS Model:** Forza FX-1500LCD
*   **USB Vendor ID (VID):** 0665
*   **USB Product ID (PID):** 5161
*   **Identification:** Appears in driver logs as "INNO TECH" "USB to Serial"

## Software Details

*   **Operating System:** Windows 11 
*   **NUT Installation:** Built from source using MSYS2 UCRT64 environment.
*   **NUT Version:** Git master branch, post-2.8.2 (specifically `2.8.2.2725-2725+g90c75da1b` from logs, built approx. April 2025)

## Steps Tried

1.  Used `blazer_usb` driver (suggested by older HCL) with `port = auto`. Failed ("No supported UPS detected").
2.  Used `blazer_usb` driver with explicit `vendorid=0665`, `productid=5161`. Failed ("No supported UPS detected").
3.  Tried forcing protocols (`megatec`, `mustek`, `megatec/old`, `zinto`) with `blazer_usb`. Failed.
4.  Tried forcing subdrivers (`cypress`, `phoenix`, `ippon`, etc.) with `blazer_usb`. Failed.
5.  Used `nutdrv_qx` driver with `port = auto`. Failed ("Device not supported!").
6.  Used `nutdrv_qx` driver with the configuration reported working for FX-2200LCD in issue #2426 (explicit VID/PID, product, vendor). **Failed ("Device not supported!")** - see logs below.
7.  Tried forcing protocols (`q1`, `megatec`, `voltronic`, `voltronic-qs`, etc.) with `nutdrv_qx`. Failed.
8.  Tried forcing subdrivers (`cypress`, `phoenixtec`, etc.) with `nutdrv_qx`. Failed.

## Expected Result

The `nutdrv_qx` driver, configured similarly to issue #2426, should successfully communicate with the UPS and report status variables (e.g., via `upsc` or direct driver execution).

## Actual Result

The driver finds the correct USB device (0665:5161) but fails to interpret the communication, resulting in "short reply", "invalid protocol", or "noncompliant reply" errors in the debug log, ultimately ending with "Device not supported!".

## Configuration (`ups.conf`) Used for Log Below

```ini
# Global section starts here
pollinterval = 1
maxretry = 3
driverpath = "/ucrt64/bin"

# UPS definition starts here
[forza_fx1500]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        product = "USB to Serial"
        vendor = "INNO TECH"
        bus = "001" # Included to match #2426, but likely ignored on Windows
        desc = "Forza FX-1500LCD (nutdrv_qx config from #2426)"

Debug Log (nutdrv_qx -DD)

Command run in MSYS2 UCRT64 terminal (as Admin): /ucrt64/bin/nutdrv_qx.exe -a forza_fx1500 -DD

Network UPS Tools 2.8.2.2725-2725+g90c75da1b (development iteration after 2.8.2) - Generic Q* USB/Serial driver 0.42
USB communication driver (libusb 1.0) 0.50
   0.000000     [D1] upsdrv_makevartable...
   0.000686     [D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010A)
   0.002070     [D1] testval_reloadable: setting 'pollinterval' exists and differs: new value '1' vs. '2'
   0.003706     [D1] Network UPS Tools version 2.8.2.2725-2725+g90c75da1b (development iteration after 2.8.2) built with gcc.exe (Rev3, Built by MSYS2 project) 14.2.0 and configured with flags: --disable-dependency-tracking
   0.006953     [D1] debug level is '2'
   0.007491     [D1] Can not become_user(<null>): not implemented on this platform
   0.008663     [D1] Signalling UPS [forza_fx1500]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.010541     WaitNamedPipe : 2
: No error [The system cannot find the file specified. ]
[... 59 lines cut ...]
   6.195548     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   6.324563     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   6.453567     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   6.582561     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   6.711552     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   6.840564     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   6.969548     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   7.098555     [D2] qx_process_answer: short reply (input.L1-N.voltage)
   7.227562     [D2] qx_process_answer: short reply (input.voltage)
   7.366546     [D2] qx_process_answer: short reply (input.voltage)
   7.496552     [D2] qx_process_answer: short reply (input.voltage)
   7.743564     [D2] qx_process_answer: short reply (input.voltage)
   7.987563     [D2] qx_process_answer: short reply (input.voltage)
   8.116560     [D2] qx_process_answer: short reply (input.voltage)
   8.117371     Device not supported!
   8.117777     Device not supported!
   8.118193     upsnotify: failed to notify about state NOTIFY_STATE_STOPPING: no notification tech defined, will not spam more about it
   8.119697     [D1] On systems without service units, consider `export NUT_QUIET_INIT_UPSNOTIFY=true`

Additional Context
The official Forza monitoring software does successfully communicate with this UPS on the same Windows machine.

I have also tried the blazer_usb driver with various configurations (including forced protocols and subdrivers) without success. Its -DD log also shows communication failures ("short reply", "No supported UPS detected").

Thanks, the forza software is horrible outdated, I hope I get this working some day !

jimklimov, 2025-04-07:
Thanks for the detailed report. A couple of quick ideas:

* the Windows port is known to be incomplete (there's a github project  dashboard to track it), and some code equivalent to POSIX is absent in the `#ifdef WIN32` blocks. I am not sure if this could bite these drivers' code paths.
* Did you try the Zadig tool (see wiki) to pin the USB VID:PID to a low-level driver used by libusb on Windows - so that other drivers don't steal bytes from data stream?

andresg97, 2025-04-07:
> Thanks for the detailed report. A couple of quick ideas:
>
> * the Windows port is known to be incomplete (there's a github project  dashboard to track it), and some code equivalent to POSIX is absent [...]

Using Zadig tool did it! After loading Winusb driver it got detected instantly.
Everything works correctly now. thank you !

sorry for the ai summary but hopefully this helps others trying to set this up:

Summary for GitHub Issue Update:

"Solution Found! After extensive troubleshooting, I've successfully got NUT working with the Forza FX-1500LCD (VID:0665 PID:5161) on Windows 11 using MSYS2 UCRT64 and a recent NUT build (2.8.2.2725-2725+g90c75da1b). The official Forza software also works, but conflicted with NUT until the Zadig step.

Here are the crucial steps that resolved the issues:

Use Zadig: The primary blocker was likely a conflict with default Windows drivers. Using Zadig ([https://zadig.akeo.ie/](https://www.google.com/url?sa=E&q=https%3A%2F%2Fzadig.akeo.ie%2F)) to replace the default driver for the USB device (0665:5161, "INNO TECH USB to Serial") with the WinUSB driver was essential. This gave libusb (used by NUT drivers) direct access.

Use nutdrv_qx Driver: Despite the HCL/scanner suggesting blazer_usb, the correct driver for this hardware (after the Zadig fix) is nutdrv_qx. The ups.conf configuration is similar to issue #2426:

# ups.conf
driverpath = "/ucrt64/bin" # Global setting

[forza_fx1500] # Or your chosen UPS name
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        # Optional but match the hardware:
        product = "USB to Serial"
        vendor = "INNO TECH"
        desc = "Forza FX-1500LCD"
Use code [with caution](https://support.google.com/legal/answer/13505487).
Ini
Bypass upsdrvctl (Bug Workaround): The upsdrvctl start <upsname> command consistently failed with a misleading No such file or directory error. Debugging revealed it was likely due to a bug in how upsdrvctl.exe constructs paths on Windows/MSYS2 (it tried to run C:\msys64\ucrt64\sbin//ucrt64/bin/nutdrv_qx.exe.exe at one point).

Workaround: Launch the driver directly instead of using upsdrvctl.

Configure Services with NSSM: Used NSSM to create three services with dependencies and specific configurations:

NUT-Driver Service:

Path: C:\msys64\ucrt64\bin\nutdrv_qx.exe

Arguments: -a forza_fx1500 (use your UPS name)

Startup directory: C:\msys64\ucrt64\sbin (or bin, where the exe is)

Environment: PATH=C:\msys64\ucrt64\bin;C:\msys64\usr\bin;C:\Windows\System32 (Crucial for finding DLLs)

NUT-Server Service (upsd):

Path: C:\msys64\ucrt64\sbin\upsd.exe

Arguments: (Optional: -DD for debug)

Startup directory: C:\msys64\ucrt64\sbin (This worked)

Environment: Same PATH as above.

Dependencies: NUT-Driver

NUT-Monitor Service (upsmon):

Path: C:\msys64\ucrt64\sbin\upsmon.exe

Arguments: (Optional: -DD for debug)

Startup directory: C:\msys64\ucrt64\sbin (This worked)

Environment: Same PATH as above.

Dependencies: NUT-Server

Use IP Address in upsmon.conf: Changed the MONITOR line in upsmon.conf to use the specific IPv4 address to avoid IPv6 connection timeouts:

# upsmon.conf
MONITOR forza_fx1500@127.0.0.1 1 monitor YourPassword master
Use code [with caution](https://support.google.com/legal/answer/13505487).
Ini
With these steps, upsc forza_fx1500@127.0.0.1 returns data correctly, and all three services start and run reliably. The key fixes were Zadig for USB access, bypassing the buggy upsdrvctl, and correctly configuring the service environment (PATH and Startup Directory) in NSSM."

jimklimov, 2025-04-07:
Thanks again for the report, confirmation, summary and a hitherto unknown bug about `upsdrvctl` :)

Also, I think `ucrt` builds were not much tested before, other than passing the build phase. So it is nice to know they are viable :)

closed 2025-04-07 by andresg97