networkupstools/nut #2885: Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release -> problem

teoberi, 2025-04-08:
`/usr/local/ups/sbin/upsdrvctl start`

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Startup timer elapsed, continuing...
> Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Duplicate driver instance detected (local Unix socket exists)! Asked the other driver nicely to self-terminate!
> Startup timer elapsed, continuing...
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Duplicate driver instance detected (PID file /var/state/ups/nutdrv_qx-PowerMust.pid exists)! Terminating other driver!
> Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Startup timer elapsed, continuing...
> Driver [PowerMust] PID 9990 initially exceeded maxstartdelay and is still starting
> root@mail:/usr/local/ups/etc# Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust

**Here the program freezes, it only ends at Enter, and this is a problem!**

`cat /var/state/ups/nutdrv_qx-PowerMust.pid`

> 9993

However, it seems to work!
I stopped and restarted the service and checked the status!

`/usr/local/ups/sbin/upsdrvctl status`

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
> UPSNAME              UPSDRV     RUNNING PF_PID  S_RESPONSIVE    S_PID   S_STATUS
> PowerMust         nutdrv_qx     RUNNING 10070   RESPONSIVE      10070   "OL"

jimklimov, 2025-04-08:
For one, note the "v2.8.3" issued tonight had some recipe problems so became 2.8.3-rc1 and a real release is expected later today.

For another, what is your OS; more specifically - why are you using `upsdrvctl` directly and not NDE-generated services (systemd/SMF); conversely, are you sure your manually launched driver instance does not conflict with same program wrapped by the service?

> I stopped and restarted the service and checked the status!

Please clarify what you meant as "service" here, because there are importantly different variants of interpretation :)

Finally, in your output I see such line as: `root@mail:/usr/local/ups/etc# Using protocol: Q1 0.08` - so apparently `upsdrvctl` with the driver did background, and eventually detached from the terminal (sending logs to syslog). So seems working like it should.

> Here the program freezes, it only ends at Enter, and this is a problem!

According to that shell prompt in the middle of the text, it did not "freeze and only end at Enter". It returned you your shell and spewed a few lines later before going quietly working.

Just in case, can you please retry like your initial attempt, to reproduce the situation where it seemed "frozen", and from ANOTHER shell session try `upsdrvctl status` and queries with `upsc` to the locally running NUT data server (`upsd`)?

teoberi, 2025-04-08:
OS: Slackware64-current (is in your CI farms as Slackware 15.0)
Slackware doesn't have systemd, I load the commands for starting NUT services into an rc.local file, I've been using it for many years!

> According to that shell prompt in the middle of the text, it did not "freeze and only end at Enter". It returned you your shell and spewed a few lines later before going quietly working.
You're right here, but it doesn't return the shell unless I press Enter after it throws those few lines!

`/usr/local/ups/sbin/upsdrvctl start`
Another shell:
`/usr/local/ups/sbin/upsdrvctl status`

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
> UPSNAME              UPSDRV     RUNNING PF_PID  S_RESPONSIVE    S_PID   S_STATUS
> PowerMust         nutdrv_qx     RUNNING 10125   NOT_RESPONSIVE  N/A

And when the shell is stuck:

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
> UPSNAME              UPSDRV     RUNNING PF_PID  S_RESPONSIVE    S_PID   S_STATUS
> PowerMust         nutdrv_qx     RUNNING 10133   RESPONSIVE      10133   "OL"

Here I press Enter to regain the shell and the status remains the same!

jimklimov, 2025-04-08:
This looks a bit weird, are you sure your descriptions to "screenshots" are correct (the second "RESPONSIVE/OL" seems healthy, not stuck)?

Also, what is the timing between `upsdrvctl start` in one shell, and `upsdrvctl status` in another? Can the first case with a known "RUNNING" PID that is however not (yet?) responsive be attributed to the driver still initializing?

A few more ideas:
* To confirm if the shell is stuck or not, try running `/usr/local/ups/sbin/upsdrvctl start ; echo "$(date) : upsdrvctl finished : $?"` -- so if you see that echo even in the middle of `upsdrvctl`'s output, you know the shell is not stuck, the program has released the terminal and in fact no longer sees your keyboard (Enter, Ctrl+C, etc. are not its input anymore).
* Bump debugging, especially during system startup if it saves any logs from the NUT drivers, to see whether and why it won't initialize then. See https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity for ideas how to set up the debug verbosity in config files.
* For command-line testing, try `upsdrvctl -DDDDDD -d -B start` or `NUT_DEBUG_LEVEL=6 upsdrvctl start` so it backgrounds as it normally does, but logs as much as it can to the console. I wonder if there are any permissions issues, or indeed a conflict with an earlier running instance of the driver program (e.g. one that your rc scripts tried to start, and then your manual invocation has to kill off).
* Just in case, can there be some fun about `/run` and `/var` mounting order, so that the driver started by rc scripts saves its PID and socket files into one filesystem, then another is mounted over that mountpoint, and the next runs of the driver program do not see those files and think they are the first copies to run (but the device is busy due to the really-first copy)?

teoberi, 2025-04-08:
Maybe some clarifications are needed!
I've been building NUT from source for many years (over 10) and after compiling and of course configuring I start the processes manually to check if everything is OK, that means:
```
/usr/local/ups/sbin/upsdrvctl start
/usr/local/ups/sbin/upsd
/usr/local/ups/bin/upsc PowerMust@localhost ups.status
/usr/local/ups/bin/upsc PowerMust@localhost
/usr/local/ups/sbin/upsmon
```
First I stop all NUT processes, check and then I start building.
There is no conflict with a previously running instance of the driver program.
I have built and am using NUT 2.8.2 on all my server systems without any problems.
Tomorrow I will try to apply the other troubleshooting suggestions.

teoberi, 2025-04-08:
`/usr/local/ups/sbin/upsdrvctl start ; echo "$(date) : upsdrvctl finished : $?"`

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Startup timer elapsed, continuing...
> Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Duplicate driver instance detected (local Unix socket exists)! Asked the other driver nicely to self-terminate!
> Startup timer elapsed, continuing...
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Duplicate driver instance detected (PID file /var/state/ups/nutdrv_qx-PowerMust.pid exists)! Terminating other driver!
> Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Startup timer elapsed, continuing...
> Driver [PowerMust] PID 10510 initially exceeded maxstartdelay and is still starting
> Tue Apr  8 20:41:53 EEST 2025 : upsdrvctl finished : 0
> root@mail:/home/netadmin# Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust

Here the execution stops and I don't get the prompt unless I press Enter!

/var/log/messages

> Apr  8 20:40:15 mail nutdrv_qx[10507]: Can't autodetect number of battery packs [-1/13.80]
> Apr  8 20:40:15 mail nutdrv_qx[10507]: Battery runtime will not be calculated (runtimecal not set)
> Apr  8 20:40:15 mail nutdrv_qx[10507]: Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Apr  8 20:40:15 mail nutdrv_qx[10508]: Startup successful
> Apr  8 20:40:18 mail nutdrv_qx[10508]: Signal -2: exiting
> Apr  8 20:41:10 mail nutdrv_qx[10509]: Can't autodetect number of battery packs [-1/13.80]
> Apr  8 20:41:10 mail nutdrv_qx[10509]: Battery runtime will not be calculated (runtimecal not set)
> Apr  8 20:41:10 mail nutdrv_qx[10509]: Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Apr  8 20:41:10 mail nutdrv_qx[10511]: Startup successful
> Apr  8 20:41:10 mail nutdrv_qx[10511]: Signal 15: exiting
> Apr  8 20:42:00 mail nutdrv_qx[10510]: Can't autodetect number of battery packs [-1/13.80]
> Apr  8 20:42:00 mail nutdrv_qx[10510]: Battery runtime will not be calculated (runtimecal not set)
> Apr  8 20:42:00 mail nutdrv_qx[10510]: Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Apr  8 20:42:00 mail nutdrv_qx[10513]: Startup successful

teoberi, 2025-04-08:
`/usr/local/ups/sbin/upsdrvctl -D -d -B start`

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
>    0.000000     [D1] upsdrvctl commanding all drivers (1 found): start
>    0.000006     [D1] Starting UPS: PowerMust
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
>    0.000000     [D1] upsdrv_makevartable...
>    0.000017     [D1] Using USB implementation: libusb-1.0.28 (API: 0x0100010A)
>    0.000157     [D1] Network UPS Tools version 2.8.3 release built with gcc (GCC) 14.2.0 and configured with flags: CPPFLAGS='-O2 -D_FORTIFY_SOURCE=3' CFLAGS='-fPIC -pie -fstack-protector-strong' CXXFLAGS='-fPIC -pie -fstack-protector-strong' LDFLAGS=-Wl,-z,now,-s,--enable-new-dtags --build=x86_64-slackware-linux --disable-static --with-user=nut --with-group=nut
>    0.000164     [D1] debug level is '1'
>    0.001233     [D1] Succeeded to become_user(nut): now UID=218 GID=218
>    0.001245     [D1] Signalling UPS [PowerMust]: driver.exit (quietly, no fuss if no driver is running or responding)
>    0.001264     Can't open /var/state/ups/nutdrv_qx-PowerMust: No such file or directory
>    0.001267     [D1] Request for other driver to exit returned code -1
>    0.001269     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
>    0.001275     [D1] PID file /var/state/ups/nutdrv_qx-PowerMust.pid not found; stat() returned -1: No such file or directory
>    0.001302     [D1] Saving PID 10543 into /var/state/ups/nutdrv_qx-PowerMust.pid
>    0.001344     [D1] upsdrv_initups...
>   45.000208     Startup timer elapsed, continuing...
>   46.647781     Using protocol: Q1 0.08
>   46.647790     [D1] upsdrv_initinfo...
>   46.873730     Can't autodetect number of battery packs [-1/13.80]
>   46.873735     Battery runtime will not be calculated (runtimecal not set)
>   46.873740     [D1] upsdrv_updateinfo...
>   46.873742     [D1] Quick update...
>   47.099786     Listening on socket /var/state/ups/nutdrv_qx-PowerMust
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
>    0.000000     [D1] upsdrv_makevartable...
>    0.000029     [D1] Using USB implementation: libusb-1.0.28 (API: 0x0100010A)
>    0.000174     [D1] Network UPS Tools version 2.8.3 release built with gcc (GCC) 14.2.0 and configured with flags: CPPFLAGS='-O2 -D_FORTIFY_SOURCE=3' CFLAGS='-fPIC -pie -fstack-protector-strong' CXXFLAGS='-fPIC -pie -fstack-protector-strong' LDFLAGS=-Wl,-z,now,-s,--enable-new-dtags --build=x86_64-slackware-linux --disable-static --with-user=nut --with-group=nut
>    0.000180     [D1] debug level is '1'
>    0.001287     [D1] Succeeded to become_user(nut): now UID=218 GID=218
>    0.001298     [D1] Signalling UPS [PowerMust]: driver.exit (quietly, no fuss if no driver is running or responding)
>    1.001735     [D1] Request for other driver to exit returned code 0
>    1.001740     Duplicate driver instance detected (local Unix socket exists)! Asked the other driver nicely to self-terminate!
>    6.001940     Can't open /var/state/ups/nutdrv_qx-PowerMust: No such file or directory
>    6.001946     [D1] Subsequent request for other driver to exit returned code -1
>    6.001951     [D1] PID file /var/state/ups/nutdrv_qx-PowerMust.pid not found; stat() returned -1: No such file or directory
>    6.001972     [D1] Saving PID 10545 into /var/state/ups/nutdrv_qx-PowerMust.pid
>    6.002000     [D1] upsdrv_initups...
>   95.000596     Startup timer elapsed, continuing...
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
>    0.000000     [D1] upsdrv_makevartable...
>    0.000027     [D1] Using USB implementation: libusb-1.0.28 (API: 0x0100010A)
>    0.000129     [D1] Network UPS Tools version 2.8.3 release built with gcc (GCC) 14.2.0 and configured with flags: CPPFLAGS='-O2 -D_FORTIFY_SOURCE=3' CFLAGS='-fPIC -pie -fstack-protector-strong' CXXFLAGS='-fPIC -pie -fstack-protector-strong' LDFLAGS=-Wl,-z,now,-s,--enable-new-dtags --build=x86_64-slackware-linux --disable-static --with-user=nut --with-group=nut
>    0.000136     [D1] debug level is '1'
>    0.001229     [D1] Succeeded to become_user(nut): now UID=218 GID=218
>    0.001240     [D1] Signalling UPS [PowerMust]: driver.exit (quietly, no fuss if no driver is running or responding)
>    0.001260     Can't open /var/state/ups/nutdrv_qx-PowerMust: No such file or directory
>    0.001262     [D1] Request for other driver to exit returned code -1
>    0.001267     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
>    0.001271     Duplicate driver instance detected (PID file /var/state/ups/nutdrv_qx-PowerMust.pid exists)! Terminating other driver!
>    0.001298     [D1] getprocname: failed to readlink() from /proc/10545/exe: Permission denied
>    0.001349     [D1] getprocname: determined process name for PID 10545: /usr/local/ups/bin/nutdrv_qx
>    0.001352     [D1] compareprocname: case-sensitive base name hit for PID 10545 of '/usr/local/ups/bin/nutdrv_qx'=>'nutdrv_qx' and checked 'nutdrv_qx'=>'nutdrv_qx'
>    0.001359     [D1] Signal sent without errors, allow the other driver instance some time to quit
>   50.001805     [D1] set_exit_flag: raising exit flag due to signal 15
>   52.230230     Using protocol: Q1 0.08
>   52.230240     [D1] upsdrv_initinfo...
[... cut ...]
>  145.001023     [D1] Completed the job of upsdrvctl tool, clean-up finished, exiting now
> root@mail:/usr/local/ups/etc#   51.646737       Using protocol: Q1 0.08
>   51.646746     [D1] upsdrv_initinfo...
>   51.872415     Can't autodetect number of battery packs [-1/13.80]
>   51.872421     Battery runtime will not be calculated (runtimecal not set)
>   51.872425     [D1] upsdrv_updateinfo...
>   51.872427     [D1] Quick update...
>   52.098035     Listening on socket /var/state/ups/nutdrv_qx-PowerMust

teoberi, 2025-04-09:
v2.8.3-rc2 has the same problem!

jimklimov, 2025-04-09:
Thanks for the tests.

In the top one of these, you can literally see the shell regain control. You get the prompt right there and then, and `upsdrvctl` (and/or the driver launched by it) also continues to write to the shell. Same as you would have with e.g. `tail -F /var/log/syslog &` eventually spewing lines over your interactive shell session. So the Enter key just has your shell re-issue its prompt, but has nothing to do with NUT programs - they don't see it nor react to it:

````
Startup timer elapsed, continuing...
Driver [PowerMust] PID 10510 initially exceeded maxstartdelay and is still starting

vvvvv THIS
Tue Apr 8 20:41:53 EEST 2025 : upsdrvctl finished : 0
root@mail:/home/netadmin# 
^^^^^ THIS
(same line contd.) Using protocol: Q1 0.08

Can't autodetect number of battery packs [-1/13.80]
Battery runtime will not be calculated (runtimecal not set)
Listening on socket /var/state/ups/nutdrv_qx-PowerMust
````

Looking at code, it may release the terminal back  to you after `maxstartdelay` configured in `ups.conf` or 45 sec by built-in default.

I am not fully sure why the entries in `/var/log/syslog` tend to show `Startup successful` and then `Signal NNN exiting` in the same second or a short time-span. May be related to this though:
```
Duplicate driver instance detected (local Unix socket exists)! Asked the other driver nicely to self-terminate!
...
Duplicate driver instance detected (PID file /var/state/ups/nutdrv_qx-PowerMust.pid exists)! Terminating other driver!
````

So either one `upsdrvctl` gets confused by launching a driver several times (could be a bug then), or you have another and it tries to keep the driver afloat. Debug enabled for `upsdrvctl` itself could expose this (e.g. `"Driver [%s] for [%s] exited soon after reload-or-exit or similar signal, restarting it"`), but I don't see it in samples above.

In your case, especially per the latter log with its time-stamping (but also implied by messages from the earlier log), it seems that the driver takes about 52 seconds to initialize until it completes protocol detection and gets to "listening on socket" - but `upsdrvctl` loses patience at 45 sec and starts another copy which kills the earlier attempt (first by protocol, but that mostly fails due to lack of socket, so then by signal).

So some mitigations to consider:
* longer `maxstartdelay` configured in `ups.conf`
* bolt the `subdriver` and/or `protocol` for this `nutdrv_qx` configuration in `ups.conf`, so less time is spent "lock-picking" the device
  * thinking of possible problems - about half a dozen subdrivers were added to nutdrv_qx since v2.8.2; not sure if detection logic of any of those involve also some sleeps/retries so that the timing would rise drastically, or it is just a very long list to walk until a hit for your device - compared to older `blazer` drivers which handle just a small subset of these dialects.

jimklimov, 2025-04-09:
For a better grasp about the timings of driver init involved here, can you please run
```
:; NUT_DEBUG_PID=true NUT_DEBUG_LEVEL=6 upsdrvctl start ; echo "$(date) : upsdrvctl finished : $?"
```

teoberi, 2025-04-09:
I like the mitigations you proposed!
I can increase maxstartdelay but I would like to know how to configure subdriver and/or protocol in ups.conf
Is there any documentation on this, I can't find anything in ups.conf about these options!

jimklimov, 2025-04-09:
Those are driver options, so rather `man nutdrv_qx` or see https://networkupstools.org/docs/man/nutdrv_qx.html

teoberi, 2025-04-09:
> Fatal error: 'maxstartdelay' is not a valid variable name for this driver.

jimklimov, 2025-04-09:
That is a global setting, outside the device section.

jimklimov, 2025-04-09:
Looking at it, `nowait` may also be a sort of useful option here (I think it makes `upsdrvctl` a sort of fire-and-forget tool, so it would not loop retrying... which may be a bad thing generally, once you get the delays set up properly, but still might help temporarily).

jimklimov, 2025-04-09:
> > Fatal error: 'maxstartdelay' is not a valid variable name for this driver.

> That is a global setting, outside the device section.

Odd, the `upsdrvctl.c` code handles it as a driver config option, similar to `sdorder`, `driver` and `port`... :\

teoberi, 2025-04-09:
If I add in ups.conf in driver section:
`protocol = Q1`
everything works extremely fast!

> Network UPS Tools upsdrvctl - UPS driver controller 2.8.3 release
> Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
> USB communication driver (libusb 1.0) 0.50
> Using protocol: Q1 0.08
> Can't autodetect number of battery packs [-1/13.80]
> Battery runtime will not be calculated (runtimecal not set)
> Listening on socket /var/state/ups/nutdrv_qx-PowerMust

And I get prompt fast!

jimklimov, 2025-04-09:
The latter is not exactly an error, since your system lacks a service management framework to notify that the daemon has initialized for proper work (so dependencies can start meaningfully) in the first place.

It is emitted by default in case this is a mis-configuration (e.g. running drivers directly where SMF or systemd are available), possibly due to obsolete packaging recipes, or if the platform has a service management framework that NUT does not yet know about (offer PRs or nudge NUT devs to integrate with it), or the platform might benefit from one (nudge distro devs/maintainers)...

If this is noise and nothing about the framework would change in a reasonably short future, `NUT_QUIET_INIT_UPSNOTIFY=true` in an init/rc script can help. See `conf/nut.conf.sample` for more such envvar toggles.

jimklimov, 2025-04-09:
And again, "getting prompt" IMHO was never a problem. As I noted above, your own posts, even the original one, showed you getting it (probably as the tool exited after 145 seconds of attempts with `maxretries=3` or so), just being obscured by lines issued by the starting driver later. But the shell was interactive at that point.

teoberi, 2025-04-09:
Quote from the link to the manual page for nutdrv_qx:

> Note | Due to historical reasons, two important tunables of this driver are somewhat inconveniently named protocol (for the Qx dialect) and subdriver (for USB to serial link conversion). Other multi-dialect drivers refer to their dialect modules as subdrivers, and actually the modular source code of nutdrv_qx also does.

so I understand that this was the solution, adding the protocol to use?

In this case, there is still the problem you exposed above with mitigations:

> thinking of possible problems - about half a dozen subdrivers were added to nutdrv_qx since v2.8.2; not sure if detection logic of any of those involve also some sleeps/retries so that the timing [...]

jimklimov, 2025-04-09:
Well, it is supposed to work out of the box, and the driver did start when left alone to its devices (pun intended) after `upsdrvctl` itself gave up and exited. So the problem is more about timing, and figuring out what takes long -- hence my queries that you time a driver start-up with your device (and no shortcuts for protocol/subdriver values).

From earlier experience, SNMP devices tend to respond slowly if there is a large MIB to walk, and/or if many devices query one SNMP UPS card (as we had in development of a monitoring solution), and/or if the appliance with NUT is overwhelmed by walking many devices while starting many drivers at once - `maxstartdelay` of 3 to 5 minutes was not unseen then, and for stress-tests we still had to batch driver startups by a few dozen at a time.

So it may well be that not specifically the driver init routine has superfluous delays that make it take longer than it could, but that the particular device's responses come slowly and it adds up as many dialect variant are tried.

teoberi, 2025-04-09:
Thanks for the support!
I'll wait for the official NUT 2.8.3 release and test again, then close this issue.

jimklimov, 2025-04-09:
Thanks for the log. So indeed, this is a matter of amount of queries made until success -- each unsupported one adds a second to the clock:
```
...
  23.043827     [D3:31109] send: 'FW?'
  24.044955     [D3:31109] read: timeout (0)
  24.044961     [D2:31109] qx_process_answer: short reply (ups.firmware)
  24.044965     [D5:31109] send_to_all: DELINFO input.voltage
...
```

A successful read takes 0.2-0.3 sec in that log.

I guess the default `maxstartdelay` should be bumped from 45 to 60 or so to avoid surprises in the field.

closed 2025-04-09 by jimklimov