networkupstools/nut #2887: Still no battery.voltage on a 3S 850 even on 2.8.3

dadodasyra, 2025-04-09:
It's a #2380 followup. Even after #2698 
I've built upsc for 2.8.3 and I'm still not getting any battery.voltage value in the upsc output : 
```
root@histeria:~# upsc eaton@localhost
battery.charge: 100
battery.charge.low: 20
battery.runtime: 1440
battery.type: PbAc
device.mfr: EATON
device.model: Eaton 3S 850 
device.serial: Blank
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 264
input.transfer.low: 161
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.status: on
outlet.1.switchable: no
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: yes
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02.08.0010
ups.load: 20
ups.mfr: EATON
ups.model: Eaton 3S 850 
ups.power.nominal: 850
ups.productid: ffff
ups.realpower: 136
ups.serial: Blank
ups.status: OL
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: offline / line interactive
ups.vendorid: 0463
root@histeria:~# upsc -V
Network UPS Tools upsc 2.8.3 release
```

jimklimov, 2025-04-09:
Did you restart the services for driver and data server, etc.? In the report above, it seems you are running older daemons: `driver.version: 2.8.0` (also the older MGE HID version)

masterwishx, 2025-04-13:
> any battery.voltage value in the upsc output

Can you post debug log?

henkiejan1, 2026-01-17:
Got also no voltage with version 2.8.3. Below is the debug info you guys want to know?

<details>
<summary>UPS info</summary>

```
battery.charge: 92
battery.charge.low: 5
battery.runtime: 2737
battery.type: PbAc
device.mfr: EATON
device.model: Eaton 3S 700 
device.serial: Blank
device.type: ups
device.usb.version: USB
driver.debug: 0
driver.flag.allow_killpower: 1
driver.flag.ignorelb: enabled
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.override.battery.charge.low: 5
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: ffff
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0463
driver.state: updateinfo
driver.version: 2.8.4
driver.version.data: MGE HID 1.55
driver.version.internal: 0.67
driver.version.usb: libusb-1.0.27 (API: 0x0100010A)
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 1
outlet.1.protect.status: protected
outlet.1.status: on
outlet.1.switchable: no
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: yes
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 02.08.001
ups.load: 8
ups.mfr: EATON
ups.model: Eaton 3S 700 
ups.power.nominal: 700
ups.productid: ffff
ups.realpower: 45
ups.serial: Blank
ups.status: OL CHRG
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: offline / line interactive
ups.vendorid: 0463
```
</details>
<details>
<summary>Debug log</summary>
```

Network UPS Tools 2.8.4 release - Generic HID driver 0.67
USB communication driver (libusb 1.0) 0.50
   0.000000	[D1] upsdrv_makevartable...
[... 1360 lines cut ...]
  32.788995	[D3] Report[buf]: (8 bytes) => 07 02 02 01 00 09 07 02
  32.789010	[D2] Path: UPS.BatterySystem.Charger.Status, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 2
  32.789029	[D4] Entering libusb_get_report
  32.834394	[D3] Report[get]: (5 bytes) => 0a ff ff ff ff
  32.834418	[D2] Path: UPS.PowerSummary.DelayBeforeStartup, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 32, Value: -1
  32.834447	[D4] Entering libusb_get_report
  32.879020	[D3] Report[get]: (5 bytes) => 09 ff ff ff ff
  32.879043	[D2] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x09, Offset: 0, Size: 32, Value: -1
  32.879067	[D4] Entering libusb_get_report
  32.921659	[D3] Report[get]: (4 bytes) => 01 25 00 00
  32.921682	[D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 1, Value: 1
  32.921700	[D3] Report[buf]: (4 bytes) => 01 25 00 00
  32.921714	[D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 4, Size: 1, Value: 0
  32.921730	[D3] Report[buf]: (4 bytes) => 01 25 00 00
  32.921745	[D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 2, Size: 1, Value: 1
  32.921763	[D3] Report[buf]: (8 bytes) => 07 02 02 01 00 09 07 02
  32.921777	[D2] Path: UPS.BatterySystem.Charger.Status, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 2
  32.921793	[D3] Report[buf]: (4 bytes) => 01 25 00 00
  32.921808	[D2] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 1, Size: 1, Value: 0
  33.985494	Signal 2: exiting
  33.985606	[D1] upsdrv_cleanup...
```
</details>

If there is some more details required please ask! :)

jimklimov, 2026-01-17:
Well, the driver tried and device did not talk:

```
   4.725004	[D4] HIDGetItemData: FindObject_with_Path() failed to locate 'UPS.BatterySystem.Voltage'
   4.725012	[D4] string_to_path: depth = 3
   4.725016	[D4] HIDGetItemData: FindObject_with_Path() failed to locate 'UPS.PowerSummary.Voltage'
   4.725023	[D4] string_to_path: depth = 3
   4.725028	[D4] HIDGetItemData: FindObject_with_Path() failed to locate 'UPS.BatterySystem.ConfigVoltage'
   4.725034	[D4] string_to_path: depth = 3
   4.725038	[D4] HIDGetItemData: FindObject_with_Path() failed to locate 'UPS.PowerSummary.ConfigVoltage'
   4.725045	[D4] string_to_path: depth = 4
```

There were also no reports with `Value: 12` or `13` :\

henkiejan1, 2026-01-17:
Yes, i saw. So i think the Eaton UPS did not report battery voltage i think? I was a little hopeful that it should be supported because i read this topic. I got this UPS yesterday  and was a little hopefull also the line voltage and current should be reported. Because the real power is reported and P = U x I. The APC Back-UPS ES700 did. But it's a budget model so maybe my expectations where to high. ;)

Little strange is btw that serial number is reported as: Blank