networkupstools/nut #2890: upssched[<pid>]: accept on unix fd failed: Too many open files

ElioDiNino, 2025-04-09:
Every ~2-3 days I am getting the following error from NUT, specifically from `upssched`: `upssched[<pid>]: accept on unix fd failed: Too many open files`. I have tried Googling, reading the mailing list, and past GitHub issues but found nothing helpful. After around a minute of this error getting repeated a few hundred times it seems to resolve itself and I also receive an email alert because it causes my heartbeat timer to fail.

I am running the version `2.8.0` of `nut` which is the latest for Proxmox 8.4.0 (Linux 6.8.12-9-pve)
My `nut` configuration files can be found here: https://github.com/ElioDiNino/Homelab/tree/main/nut

jimklimov, 2025-04-09:
Curious, haven't seen that before...

Generally, there's a system `ulimit` on various constrained resources including the amount of open file descriptors, and those can be inherited from OS hard and soft limits, or set explicitly for systemd units as well as legacy init scripts. Maybe this amount is too constrained on your system somehow, so the limit is easily reached during healthy life-cycle, or something "leaks" in NUT. I think you can monitor the open FD's via `ls -la /proc/<pid>/fd` on Linux to see how many there are and if the population changes over time or only grows (leak).

Looking at code, there is a `conn_add()` method called from `start_daemon()` which implements the main loop. I haven't had much time to discern if this concerns connections from the child processes with timers, from `upsmon`, or both.

I can't quickly say if the amount of connections is proportional to any settings (timers, notifiers), so at this point can only suggest hints like this.

Guessing from symptoms:
* Given that the problem only happens every few days, I guess this is not a leak but your setup reaching some low-ish threshold when "stars align" in a bad but rare manner. Maybe some combo/amount of alerts is raised that is just right for your setup.
* The "hundred repeats" suggests that the connections are retried (likely with little/no delay), and
* "after around a minute" suggests that something gets restarted, maybe? Or the caller gives up? Are there any log entries suggesting that a process died with error etc. around the end of such minute?

jimklimov, 2025-04-09:
Can't help but wonder if this is under the hood an issue similar to https://alioth-lists.debian.net/pipermail/nut-upsuser/2025-April/013928.html - at least, these are two upssched weirdness discussions within a week.

>  at that time I used ulimit to try increasing...

Did you do it from shell? This only impacts processes spawned by it (or its descendants), but not e.g. systemd units (independence from current shell circumstances was one of selling points for service management frameworks vs. init-scripts). The kernel-wide soft/hard defaults I think can be set via sysctl (e.g. in /etc/sysctl.conf somewhere), but for systemd units there's a `LimitNOFILE` setting. The `nut-server.service` has some detailed comments from my older experience:
````
# A busy server can be monitoring a lot of devices as well as replying
# to many clients. The "infinity" definition is actually capped by OS
# settings and hard-coded defaults; typically can be 65535+ nowadays.
# On 64-bit distros this can well be set into hundreds of thousands
# as well (though note each connection has a CPU and RAM overhead
# so one can strike physical limits upon deployment and/or bring the
# poorly sized system to a crawl, or worse). On a running system you
# can check /proc/$MAINPID/limits for active ulimits of the process.
# From my experiments, up to 1048576 can be set, but any larger value
# falls back to 65536. The systemd definition of "infinity" is 65536
# too (or maybe it falls back to that); though this may be OS/distro
# limitation and not systemd fault specifically.
#LimitNOFILE=infinity
#LimitNOFILE=65535
LimitNOFILE=1048576
````

ElioDiNino, 2025-04-09:
> Did you do it from shell? This only impacts processes spawned by it (or its descendants), but not e.g. systemd units (independence from current shell circumstances was one of selling points for [...]

Ah yeah I did, whoops. Thanks for the pointer, I have now adjusted the limit there (as well as for `nut-monitor.service`) and see the expected limits reflected in `/proc/<pid>/limits`. Hopefully that will help with my issue, time to wait a few days :)

jimklimov, 2025-04-11:
Cheers, I've automated debugging of `upssched` via NIT (NUT Integration Testing) suite as posted to https://github.com/jimklimov/nut/tree/issue-2890 branch. The mailing list thread linked above also includes results of investigation done for that other issue (seems unrelated to yours after all, but may be an interesting read anyway).

ElioDiNino, 2025-04-11:
> According to that investigation (logs in mailing list), a tracked connection from `upssched` CLI process (called as `upsmon` NOTIFYCMD) to daemon appears to process START/CANCEL TIMER events, and disappears when each such processing is finished. Probably there are more FDs added/removed when a handling CMDSCRIPT is called from `upssched`.
> 
> Check that these `CMDSCRIPT`'s actually complete and exit for you in finite time, perhaps that they do not launch any background children either.

Thanks for all the details. I read through the thread, but also agree that the issue is likely different.

I have been checking on the FDs for `upssched` today as can see that they are continually climbing. Since a restart when I created the GitHub issue, the FD count has now risen to 879 (and increases every 5 minutes, in sync with my [dummy heartbeat ups alternating on and off every 5 minutes](https://github.com/ElioDiNino/Homelab/blob/main/nut/heartbeat.conf)). Additionally, when looking into the actual FDs, it looks like they are all related to `/var/run/nut/upssched.pipe` which is what I have `PIPEFN` set to ([ref](https://github.com/ElioDiNino/Homelab/blob/main/nut/upssched.conf#L2)) - see logs below.

My `CMDSCRIPT` does not get executed for any of these dummy heartbeat events (unless the heartbeat fails to switch status which is how I discovered this issue in the first place), so I don't think it's related to that (you can see [my script here](https://github.com/ElioDiNino/Homelab/blob/main/nut/upssched-cmd.sh) though). Maybe the heartbeat I have set up is configured incorrectly somehow? I am doubtful though, since I followed instructions primarily from here (which I see you are an author of): https://rogerprice.org/NUT/ConfigExamples.A5.pdf

[fd-logs.txt](https://github.com/user-attachments/files/19711058/fd-logs.txt) (was too long to inline unfortunately)

ElioDiNino, 2025-04-14:
From my Googling, `NODE` is just the file number of the referenced local file. What I can see is that each bracketed number (within `socket:[...]`) from the `fd` output corresponds to a UNIX socket in the `lsof` output (as seen before). Furthermore, when analyzing the socket information from `ss`, I see those same sockets and the output suggests that the numbers are actually the listening port for the socket. I haven't been able to find much outside of that though, such as what is creating each and where they are being used.

<details>

<summary>Socket Analysis Logs</summary>

```
❯ ls -la /proc/4005950/fd
total 0
dr-x------ 2 root root 147 Apr 14 02:42 .
dr-xr-xr-x 9 root root   0 Apr 14 02:42 ..
lrwx------ 1 root root  64 Apr 14 02:42 0 -> /dev/null
lrwx------ 1 root root  64 Apr 14 02:42 1 -> /dev/null
lrwx------ 1 root root  64 Apr 14 02:42 10 -> 'socket:[26015824]'
lrwx------ 1 root root  64 Apr 14 07:54 100 -> 'socket:[26397315]'
lrwx------ 1 root root  64 Apr 14 07:54 101 -> 'socket:[26397316]'
lrwx------ 1 root root  64 Apr 14 08:00 102 -> 'socket:[26396500]'
lrwx------ 1 root root  64 Apr 14 08:04 103 -> 'socket:[26402730]'
lrwx------ 1 root root  64 Apr 14 08:04 104 -> 'socket:[26402731]'
lrwx------ 1 root root  64 Apr 14 08:10 105 -> 'socket:[26410139]'
lrwx------ 1 root root  64 Apr 14 08:15 106 -> 'socket:[26421137]'
lrwx------ 1 root root  64 Apr 14 08:15 107 -> 'socket:[26421138]'
lrwx------ 1 root root  64 Apr 14 08:19 108 -> 'socket:[26424933]'
lrwx------ 1 root root  64 Apr 14 08:24 109 -> 'socket:[26434539]'
lrwx------ 1 root root  64 Apr 14 02:53 11 -> 'socket:[26008912]'
lrwx------ 1 root root  64 Apr 14 08:24 110 -> 'socket:[26434540]'
lrwx------ 1 root root  64 Apr 14 08:30 111 -> 'socket:[26438695]'
lrwx------ 1 root root  64 Apr 14 08:38 112 -> 'socket:[26438780]'
lrwx------ 1 root root  64 Apr 14 08:38 113 -> 'socket:[26438781]'
lrwx------ 1 root root  64 Apr 14 08:38 114 -> 'socket:[26438782]'
lrwx------ 1 root root  64 Apr 14 08:44 115 -> 'socket:[26458886]'
lrwx------ 1 root root  64 Apr 14 08:44 116 -> 'socket:[26458887]'
lrwx------ 1 root root  64 Apr 14 08:52 117 -> 'socket:[26460829]'
lrwx------ 1 root root  64 Apr 14 08:54 118 -> 'socket:[26467835]'
lrwx------ 1 root root  64 Apr 14 08:54 119 -> 'socket:[26467836]'
lrwx------ 1 root root  64 Apr 14 02:59 12 -> 'socket:[26019187]'
lrwx------ 1 root root  64 Apr 14 08:59 120 -> 'socket:[26467837]'
lrwx------ 1 root root  64 Apr 14 09:06 121 -> 'socket:[26467840]'
lrwx------ 1 root root  64 Apr 14 09:06 122 -> 'socket:[26467841]'
lrwx------ 1 root root  64 Apr 14 09:10 123 -> 'socket:[26467873]'
lrwx------ 1 root root  64 Apr 14 09:16 124 -> 'socket:[26498169]'
lrwx------ 1 root root  64 Apr 14 09:16 125 -> 'socket:[26498170]'
lrwx------ 1 root root  64 Apr 14 09:19 126 -> 'socket:[26501719]'
lrwx------ 1 root root  64 Apr 14 09:25 127 -> 'socket:[26501729]'
lrwx------ 1 root root  64 Apr 14 09:25 128 -> 'socket:[26501730]'
lrwx------ 1 root root  64 Apr 14 09:29 129 -> 'socket:[26503016]'
lrwx------ 1 root root  64 Apr 14 03:02 13 -> 'socket:[26039351]'
lrwx------ 1 root root  64 Apr 14 09:36 130 -> 'socket:[26501740]'
lrwx------ 1 root root  64 Apr 14 09:36 131 -> 'socket:[26501741]'
lrwx------ 1 root root  64 Apr 14 09:39 132 -> 'socket:[26517444]'
lrwx------ 1 root root  64 Apr 14 09:46 133 -> 'socket:[26527865]'
lrwx------ 1 root root  64 Apr 14 09:46 134 -> 'socket:[26527866]'
lrwx------ 1 root root  64 Apr 14 09:50 135 -> 'socket:[26540550]'
lrwx------ 1 root root  64 Apr 14 09:55 136 -> 'socket:[26568915]'
lrwx------ 1 root root  64 Apr 14 09:55 137 -> 'socket:[26568916]'
lrwx------ 1 root root  64 Apr 14 09:59 138 -> 'socket:[26568935]'
lrwx------ 1 root root  64 Apr 14 10:08 139 -> 'socket:[26568960]'
lrwx------ 1 root root  64 Apr 14 03:02 14 -> 'socket:[26039352]'
lrwx------ 1 root root  64 Apr 14 10:08 140 -> 'socket:[26568961]'
lrwx------ 1 root root  64 Apr 14 10:09 141 -> 'socket:[26589388]'
lrwx------ 1 root root  64 Apr 14 10:14 142 -> 'socket:[26589392]'
lrwx------ 1 root root  64 Apr 14 10:14 143 -> 'socket:[26589393]'
lrwx------ 1 root root  64 Apr 14 10:24 144 -> 'socket:[26596268]'
lrwx------ 1 root root  64 Apr 14 10:24 145 -> 'socket:[26604753]'
lrwx------ 1 root root  64 Apr 14 10:24 146 -> 'socket:[26604754]'
lrwx------ 1 root root  64 Apr 14 03:07 15 -> 'socket:[26009563]'
lrwx------ 1 root root  64 Apr 14 03:13 16 -> 'socket:[26044776]'
lrwx------ 1 root root  64 Apr 14 03:13 17 -> 'socket:[26044777]'
lrwx------ 1 root root  64 Apr 14 03:18 18 -> 'socket:[26009575]'
lrwx------ 1 root root  64 Apr 14 03:22 19 -> 'socket:[26009578]'
[... 714 lines cut ...]
Apr 14 02:37:28 pve nut-monitor[4004994]: Network UPS Tools upsmon 2.8.0
Apr 14 02:38:17 pve systemd[1]: Starting man-db.service - Daily man-db regeneration...
Apr 14 02:38:17 pve systemd[1]: man-db.service: Deactivated successfully.
Apr 14 02:38:17 pve systemd[1]: Finished man-db.service - Daily man-db regeneration.
Apr 14 02:41:40 pve tailscaled[924]: netcheck: DetectCaptivePortal(found=false)
Apr 14 02:42:28 pve nut-monitor[4005949]: read confirmation failed, trying again
Apr 14 02:42:28 pve nut-monitor[4005949]: read confirmation failed, trying again
Apr 14 02:42:28 pve upssched[4005949]: read confirmation failed, trying again
Apr 14 02:42:28 pve upssched[4005949]: read confirmation failed, trying again
Apr 14 02:42:28 pve upssched[4005950]: Timer daemon started
[... cut ...]
Apr 14 02:47:30 pve nut-monitor[4006894]: Network UPS Tools upsmon 2.8.0
Apr 14 02:52:04 pve tailscaled[924]: netcheck: DetectCaptivePortal(found=false)
Apr 14 02:52:04 pve tailscaled[924]: magicsock: endpoints changed: 192.168.0.17:41641 (portmap), 24.85.226.112:1100 (stun), 192.168.50.132:41641 (local)
Apr 14 02:52:30 pve upssched[4005950]: Cancelling timer: heartbeat-failure-timer
Apr 14 02:52:30 pve upssched[4005950]: New timer: heartbeat-failure-timer (660 seconds)
Apr 14 02:52:30 pve nut-monitor[4007842]: Network UPS Tools upsmon 2.8.0
```

</details>

jimklimov, 2025-04-14:
Commuting now so skipped long logs, will check later with a real monitor. Regarding text :)

* what is your `zed` that's child and grandchild of upsmon? googling shows an AI editor by such name :\ An editor might I guess be invoked by mailx, if misfiring somehow (ignoring stdin?) Double-check from command line and/or other scripts/services that such `echo | mailx` works for implementations your system has.
* Are the nut run-time user/group defined in local databases (files), or ldap etc., or nowhere? Tools like mail and logger can be interested in caller's uid and block or fail on invalid-name lookups (bit me with some docker containers that all wanna be uid=1000).
* Are you sure upsmon actually fails and restarts? Does `systemctl status` or `ps -ef` confirm that the parent process is recycled often? I have a theory that notification process is forked from main upsmon (so it can continue looping), and that child may show banners of its own. Certainly did show stopping messages, that was fixed for 2.8.2 or later. Your later post has 2.8.0 so maybe suffers from that cosmetic confusing issue.

ElioDiNino, 2025-05-26:
> I agree that not-closing something is a bug, so would rather keep this issue open as a reminder. It just happens to be one I can not reproduce to investigate (or eventually confirm a fix).

Make sense, I will try to watch for when Debian releases newer stable NUT versions and see if the issue remains. Thank you for the time you have spent on this and the rest of the project!

jimklimov, 2025-05-26:
> when Debian releases newer stable NUT versions

From my understanding, that might be likely only whenever a new Debian major release comes out and they bump baselines. It may be more dynamic with experimental/"sid" repositories where such updates are marinated and live-tested by a smaller community.