networkupstools/nut #2895: Nut monitor not starting after update to 2.8.2.2784-2784+g991d39868

Phils80, 2025-04-11:
Beeing succesful previously configuting other server I tried to repeat installations steps, first intalling from aviable deb packages  and after configuration performing update from github using:

`./configure --enable-inplace-runtime --with-doc=no`  as mentioned [here](https://github.com/networkupstools/nut/issues/2741#issuecomment-2569203862)

and routine described  [here](https://networkupstools.org/docs/user-manual.chunked/_installation_instructions.html):

```
:; cd /tmp
:; git clone https://github.com/networkupstools/nut
:; cd nut
### OPTIONALLY fetch known git tags, so semantic versions look better
:; git fetch --tags --all
### Proceed with build
:; ./autogen.sh
:; ./configure --enable-inplace-runtime # --maybe-some-other-options
:; make -j 4 all && make -j 4 check && \
    { sudo systemctl stop nut-monitor nut-server || true ; } && \
    { sudo systemctl stop nut-driver.service || true ; } && \
    { sudo systemctl stop nut-driver.target || true ; } && \
    { sudo systemctl stop nut.target || true ; } && \
    sudo make install && \
    sudo systemctl daemon-reload && \
    sudo systemd-tmpfiles --create && \
    sudo systemctl disable nut.target nut-driver.target \
        nut-monitor nut-server nut-driver-enumerator.path \
        nut-driver-enumerator.service && \
    sudo systemctl enable nut.target nut-driver.target \
        nut-monitor nut-server nut-driver-enumerator.path \
        nut-driver-enumerator.service && \
    { sudo systemctl restart udev || true ; } && \
    sudo systemctl restart nut-driver-enumerator.service \
        nut-monitor nut-server
```

Unfortunetly after this operation I have lost communication with my UPS, when trying to connect using upsc, I'm reciving following message:

`upsc: symbol lookup error: upsc: undefined symbol: upscli_init_default_connect_timeout`

all services except **nut-monitor** are working and are enabled, for nut-moinitor I get:

 ```
nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/usr/lib/systemd/system/nut-monitor.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2025-04-11 08:40:25 CEST; 6h ago
   Duration: 4ms
       Docs: man:upsmon(8)
             https://www.networkupstools.org/docs/man/upsmon.html
             man:upsmon.conf(5)
             https://www.networkupstools.org/docs/man/upsmon.conf.html
             man:nut.conf(5)
             https://www.networkupstools.org/docs/man/nut.conf.html
    Process: 40445 ExecStartPre=/usr/bin/systemd-tmpfiles --create /usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, status=0/SUCCESS)
    Process: 40446 ExecStart=/usr/sbin/upsmon -F (code=exited, status=127)
   Main PID: 40446 (code=exited, status=127)
        CPU: 8ms

kwi 11 08:40:25 [***pivacy - removed***] systemd[1]: nut-monitor.service: Scheduled restart job, restart counter is at 5.
kwi 11 08:40:25 [***pivacy - removed***] systemd[1]: nut-monitor.service: Start request repeated too quickly.
kwi 11 08:40:25 [***pivacy - removed***] systemd[1]: nut-monitor.service: Failed with result 'exit-code'.
kwi 11 08:40:25 [***pivacy - removed***] systemd[1]: Failed to start nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller.
```
Any idea how to fix this problem ? I was looking for sollution, but I don't see anyone with same problem. 

P.S. Nut installed from deb package was working properly except false alarms for BX950MI (known issue not solved in 2.8.1).

jimklimov, 2025-04-11:
Seems new libupsclient.so did not get installed, or rpath baked into binaries misfired? :-\

jimklimov, 2025-04-11:
My guess is that an older library is in e.g. `/usr/lib` and it gets searched before `/usr/local/lib` for example where your custom build's library got installed. In a way then yes, it can be a flaw in `enable-inplace` if it did not detect the path. Or the "uncommon rpath" (see recent merged PRs, also for commands to check the actual values) did not get baked in or is not used correctly.

jimklimov, 2025-04-12:
Thinknig some more where I could mess it up: I suppose a newer library that exports more symbols should have got a more significant version bump, so the dynamic linker should not have confused them even if it sees both in search path. Maybe my change in `clients/Makefile.am` compared to v2.8.2 git tag was too conservative. And seems like something you can experiment with...

jimklimov, 2025-04-14:
Ran a few experiments, and it seems that on some platforms, including Debian-ish Linuxes, Solaris-derived systems, etc., libraries fall into arch-dependent paths like `(/usr)/lib/x86_64-linux-gnu/libupsclient.so.4`, but the "inplace" detection logic did not bother to create an *identical* twin and so kept `${prefix}/lib` as well as `${prefix}/bin` for drivers (something that Debian hides into `(/usr)/lib/nut`).

Essentially the same-ness of user/group names to run as, and location of config files, was what mattered until now. Maybe even more so founded, to allow removal of custom-built NUT (if disliked) to let the packaged one be used again.
* This is a bit of a gamble though, because third-party locations (systemd, augeas, python...) would be used by both builds, so the later write wins.

I suppose that your system got a new `(/usr)/lib/libupsclient.so.4` from the custom build in the generic location, but the higher-priority arch-specific preference (as far as linker is concerned) from the older package build (in reality) remained at `(/usr)/lib/x86_64-linux-gnu/libupsclient.so.4` and so was picked. And since both locations are "common", the recently added magic about "rpath" additions (to search for libs) were not done by default. I'll check what can be done about this, e.g. maybe force the rpath fiddling for in-place config (add `--enable-ldflags-nut-rpath` there), etc.

jimklimov, 2025-04-14:
Ugh, FWIW -- the `--enable-ldflags-nut-rpath{,-cxx}` options only help expose the RPATH in pkgconfig metadata, but autotools take care of linking the built binaries/libraries.

And for standard-ish paths in question, nothing gets embedded into the freshly built ELF files:
````
:; readelf -d /tmp/nut/usr/bin/upsc | grep -Ei 'R.*PATH'
# <nothing>

:; readelf -d /tmp/nut/usr/bin/upsc
Dynamic section at offset 0xeae0 contains 28 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libupsclient.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x9e34
 0x0000000000000019 (INIT_ARRAY)         0xf9d0
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0xf9d8
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x3b0
 0x0000000000000005 (STRTAB)             0xd30
 0x0000000000000006 (SYMTAB)             0x3e8
 0x000000000000000a (STRSZ)              1072 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0xfce0
 0x0000000000000002 (PLTRELSZ)           2112 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x1918
 0x0000000000000007 (RELA)               0x12b8
 0x0000000000000008 (RELASZ)             1632 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000000000001e (FLAGS)              BIND_NOW
 0x000000006ffffffb (FLAGS_1)            Flags: NOW PIE
 0x000000006ffffffe (VERNEED)            0x1228
 0x000000006fffffff (VERNEEDNUM)         1
 0x000000006ffffff0 (VERSYM)             0x1160
 0x000000006ffffff9 (RELACOUNT)          59
 0x0000000000000000 (NULL)               0x0
````

Even though the rpath options are forced-enabled (for third-party integrations):
````
...
configure:56115: checking whether NUT libdir='/usr/lib' is among common search paths
configure:56155: result: yes
configure:56166: checking whether to suggest a LDFLAGS_NUT_RPATH in pkg-config et al, and which
configure:56199: result: yes: -Wl,-rpath -Wl,${libdir}
configure:56210: checking whether to suggest a LDFLAGS_NUT_RPATH_CXX in pkg-config et al, and which
configure:56243: result: yes: -Wl,-rpath -Wl,${libdir}
...

:; cat lib/libupsclient.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include
sysconfdir=/etc/nut
statepath=/var/state/ups
nutuser=nut

Name: libupsclient
Description: UPS monitoring with Network UPS Tools
Version: 2.8.2.2806
Libs: -L${libdir} -Wl,-rpath -Wl,${libdir} -lupsclient
Cflags: -I${includedir}
Requires:
````

* For more details, see #2782 and #2865.

jimklimov, 2025-04-14:
The insufficient library `version-info` bump confirmed, thanks for bringing this up *before* the new NUT release ultimately :)

jimklimov, 2025-04-14:
I'll post a PR with assorted fixes shortly, would be great if you can test its source branch, how neatly it guesses the options to apply with an "in-place" build for your system:

```
:; git clone https://github.com/jimklimov/nut -b issue-2895 nut-issue-2895
:; cd nut-issue-2895
:; ./ci_build.sh inplace
```

Phils80, 2025-04-14:
> I'll post a PR with assorted fixes shortly, would be great if you can test its source branch, how neatly it guesses the options to apply with an "in-place" build for your system:
>
> ```
> :; git clone [...]

I confirm that for my system all went fluently. Everythng including monitor is woking perfectly, so changes seems to be efficient. Thanks a lot !

jimklimov, 2025-04-14:
Great, thanks!