networkupstools/nut #2900: [HELP/Support] Ubuntu 24.04, APC SmartUPS 2200, USB, Can't communicate, but UPS turns off without needing too.

KLeeCollins, 2025-04-14:
It's kind of a weird one.  My current setup is documented [HERE](https://github.com/KLeeCollins/NUTConfig/tree/main).

First problem:  After the machine boots and the daemons start, I issue the command:
```
upsc -L
```
or
```
ups bottom-ups
```
And I get
```
Error: Connection failure: Connection refused
```

The second issue I have right is that even though the power never is lost, NUT is sending the command to shut down the UPS (ONLY THE UPS, the system itself).  Roughly two minutes after the system starts, the UPS turns off and instantly comes back on.  It's almost like the power failure timer starts when it's not supposed to.  But I can't figure out why.

The description as to how I want it to work is in the readme on the GitHub page.  I've been working through the process of trying to get NUT to do that.  My next step is to blow away the NUT system and start over.  Any help would be appreciated.

KLeeCollins, 2025-04-14:
Thank you for the quick reply!  I apologize for the communications and docs comment.  It's just in the age of Discord, I would have been happy to hop onto a Discord server and ask my question.  But that's my lack of patience only.

My UPSes are old.  The oldest - with the serial connection - was bought in 2006.  The other two, one of which is the UPS I'm trying to monitor, were bought in late 2018.  They have been tanks and I've simply had to swap batteries every couple of years to keep them going.

That all being said, do you think that ```apc_modbus``` is the real com error issue?

I see the error in my ```upsmon.conf``` file and I'll fix that when I rebuild the system.  I was thinking of monitoring the two "new" UPSes when I first started this process, I guess I forgot that when I created that file.

I'll look into gathering more debug information and reply when I know or have more to offer.

jimklimov, 2025-04-14:
> do you think that apc_modbus is the real com error issue?

Without logs, hard to say. Ideally those with high debug verbosity (practical values up to 6) to see what devices it detects, how it goes about confurming the protocol, etc.