networkupstools/nut #2917: Cyber Power EC850LCD reports wrong output.realpower.nominal

Belphemur, 2025-04-24:
Hello, 

From the spec of the device and APCUd it's 510W.

However, nut is reporting 450W.

It's connected through USB. And I'm using nut 2.8.3 on unraid as a plugin.

jimklimov, 2025-04-24:
Hello, can you please check if you can get detailed logs (in particular the USB report descriptor dump) when starting the driver, perhaps by adding `debug_min=6` temporarily - see e.g. https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity ? CC @desertwitch 

I wonder if it serves bogus bytes which we have to re-interpret and do it incorrectly? (CPS is one of a couple of drivers where we amend wrong USB encoding from vendor firmware; perhaps different device releases need different fixes)

jimklimov, 2025-04-24:
Also, I wonder if this is related to #2728 or #2729 ?

desertwitch, 2025-04-26:
In Unraid's NUT Settings, please first stop NUT, then set these settings as shown below:

[image]

[image]

Then start NUT and wait for a bit, after which you can download the logfile and attach it to this issue.

It is important to **download** the log file (clicking the download icon) and not copy it out of the log viewer (clicking the book icon), as the log viewer truncates any parts that are exceeding 200 lines (which a lot of debug lines will probably exceed).

You can afterwards revert the above changed settings back to their original states/defaults. 😉

Belphemur, 2025-04-26:
[nut (1).log](https://github.com/user-attachments/files/19922235/nut.1.log)

Here is the log file as requested. 

Also even in debug mode there is barely anything in it... 
I did leave it running like this for 1-2 minutes.

desertwitch, 2025-04-26:
> [nut (1).log](https://github.com/user-attachments/files/19922235/nut.1.log)
> 
> Here is the log file as requested. 
> 
> Also even in debug mode there is barely anything in it... 
> I did leave it running like this for 1-2 minutes. 

You raised the debug level on the monitor, not the driver. Please check again with the screenshot I posted above.

Belphemur, 2025-04-26:
Good catch. Doing it from my phone didn't help. 

Here is the proper log file with debug values of the driver. 

[nut (3).log](https://github.com/user-attachments/files/19922340/nut.3.log)

desertwitch, 2025-04-26:
Hmm, interesting - `Report[buf]: (3 bytes) => 18 fe 01`, the device sends 510W.
This seems to get clipped by `LogMax = 450` of the descriptor (this is a firmware issue).

This is one of the `cps_fix_report_desc` models, meaning we could likely fix this descriptor up as part of that as well.
We may just need to set the LogMax to a more realistic cutoff value, so should be a relatively easy fix...

closed 2025-04-29 by jimklimov

desertwitch, 2025-05-21:
I just updated the Unraid NUT "preview" backend to the latest master, could you check if it works for you now and report back?

Belphemur, 2025-05-21:
@desertwitch Just updated now and confirm I'm getting a nice 510W now :) 

[image]

PR #2919 (2025-04-29): drivers/cps-hid.c: add report descriptor fix for ConfigActivePower
fix #2917 

Findings in the linked issue suggest that at least one model in the `cps_fix_report_desc` model range suffers also from a broken descriptor for reporting the UPS nominal power (`ConfigActivePower`). The value seemingly gets correctly sent by the UPS, but is then clipped by a too low `LogMax` of that affected descriptor:  [nut-issue-2917.log](https://github.com/user-attachments/files/19923182/nut-issue-2917.log)

```
Apr 26 08:43:04 Tower rc.nut:    0.035598#011[D3:181672] Report[get]: (3 bytes) => 18 fe 01
Apr 26 08 [... cut]