networkupstools/nut #2918: COMMBAD, NOCOMM as UPS statuses?

desertwitch, 2025-04-26:
Follow up to #2877 and #2909 

Should we consider adding UPS status codes like `COMMBAD` and/or `NOCOMM` to explicitly indicate communication failures with monitored UPS devices? Currently, `upmon` only reports these issues through notifiers and log files, making it difficult for users to distinguish between current and outdated UPS status information right away.

Since most notification events are already associated with specific UPS status flags, adding dedicated status flags for communication failures would provide immediate visual indication that displayed UPS information may be unreliable or outdated. This would allow users to quickly identify communication problems without needing to configure special notifications or search through log files.

desertwitch, 2025-04-26:
On second thought, this idea makes no sense as `driver` -> `upsd` -> `upsmon`.
`upsmon` just acts on what it queries from `upsd`, it doesn't change the representation to the user.
An individual monitoring client connection loss (due to whatever reason) should not affect global state.

closed 2025-04-26 by desertwitch

jimklimov, 2025-04-26:
Yes, communications between driver/upsd and a client like upsmon are not a concern got the driver to show. Different clients might have or not have the connection to upsd.

A "Data stale" is a state reported by `upsd` when a driver dies or freezes (typically) or is stopped in a debugger, and no refreshes come for more than X seconds.

Some drivers do expose their connection to the device (or rather a detected lack thereof, when the `driver.state` can become `reconnect.trying` and `reconnect.updateinfo`).

Your idea/question did sound like #2810 however.