networkupstools/nut #2923: Ippon's Na+ series UPS is supported by blazer_ser.

penghong123, 2025-04-28:
Hello!
      I have tested Ippon's Na+ series UPS and found that blazer_ser can support it. You can add these models to the list of Ippon-supported models.
Network UPS Tools :2.7.4

Manufacturer: **Ippon**
Model:
              **2049878 UPS Ippon Na+ TA 1000 (white)
              2055083 UPS Ippon Na+ TAE 1000 (white)
              2049879 UPS Ippon Na+ TA 2000 (white)
              2055084 UPS Ippon Na+ TAE 2000 (white)
              2049880 UPS Ippon Na+ TA 3000 (white)
              2055085 UPS Ippon Na+ TAE 3000 (white)
              2049881 UPS Ippon Na+ TA 6K (white)
              2049882 UPS Ippon Na+ TA 10K (white)
              2049895 Battery for UPS Ippon NA+ TA 6/10K
              2049883 UPS Ippon Na+ RTA 1000 (white)
              2049884 UPS Ippon Na+ RTA 2000 (white)
              2049885 UPS Ippon Na+ RTA 3000 (white)
              2049886 UPS Ippon Na+ RTA 6K (white)
              2049888 UPS Ippon Na+ RTA 10K (white)
              2058874 Battery for UPS Ippon NA+ RTA 2000
              2049899 Battery for UPS Ippon NA+ RTA 3000
              2049897 Battery for UPS Ippon NA+ RTA 6/10K**

upsc myups@localhost

```
Init SSL without certificate database
battery.charge: 100
battery.voltage: 42.30
battery.voltage.high: 39.00
battery.voltage.low: 31.20
battery.voltage.nominal: 36.0
device.mfr: 222222222222222
device.model: 2222222222
device.type: ups
driver.name: blazer_ser
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyCH341USB0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 1.57
input.current.nominal: 4.0
input.frequency: 0.0
input.frequency.nominal: 50
input.voltage: 0.0
input.voltage.fault: 0.0
input.voltage.nominal: 230
output.voltage: 229.9
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.48
ups.load: 0
ups.mfr: 222222222222222
ups.model: 2222222222
ups.status: OB
ups.temperature: 50.0
ups.type: online
```

cat /etc/nut/ups.conf

```
maxretry = 3
[myups]                                                              
     driver = blazer_ser                                 
     port = /dev/ttyCH341USB0   
     desc = "My UPS"  
```

cat /var/log/syslog | grep upsd

```
Network UPS Tools - UPS driver controller 2.7.4
Apr 28 17:02:19  upsd[1331]: mainloop: Interrupted system call
Apr 28 17:02:19  upsd[1331]: Signal 15: exiting
Apr 28 17:02:23  upsdrvctl[6494]: Supported UPS detected with megatec protocol
Apr 28 17:02:24  upsdrvctl[6494]: Vendor information read in 1 tries
Apr 28 17:02:24  upsdrvctl[6494]: No values provided for battery high/low voltages in ups.conf
Apr 28 17:02:24  upsdrvctl[6494]: Using 'guestimation' (low: 31.200000, high: 39.000000)!
Apr 28 17:02:24  upsdrvctl[6494]: Battery runtime will not be calculated (runtimecal not set)
Apr 28 17:02:24  upsdrvctl[6493]: Network UPS Tools - UPS driver controller 2.7.4
Apr 28 17:02:24  upsd[6496]: fopen /run/nut/upsd.pid: No such file or directory
Apr 28 17:02:24  upsd[6496]: listening on 0.0.0.0 port 3493
Apr 28 17:02:24  upsd[6496]: listening on 0.0.0.0 port 3493
Apr 28 17:02:24  upsd[6496]: Connected to UPS [myups]: blazer_ser-myups
Apr 28 17:02:24  upsd[6496]: Connected to UPS [myups]: blazer_ser-myups
Apr 28 17:02:24  upsd[6497]: Startup successful
Apr 28 17:02:24  upsd[6497]: User admin@127.0.0.1 logged into UPS [myups]

```

lsusb -vvv

```
Bus 003 Device 005: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 CH340 serial converter
  bcdDevice            2.64
  iManufacturer           0 
  iProduct                2 USB Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1 
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

**Shut down operation: Manually trigger shutdown**

sudo upsdrvctl stop

`Network UPS Tools - UPS driver controller 2.7.4`

sudo journalctl -u nut-monitor -f
                                               
```
Log:
18:14:25 upsmon[1383]: poll UPS [myups@localhost] failed
Driver not connected
18:14:25 upsmon[1383]: FSD set on UPS myups@localhost fail
ed: ERR ACCESS-DENIED
18:14:25  upsmon[1383]:Executing automatic power-fail shut
down
8:14:25 upsmon[1383]: Auto logout and shutdown proceeding
```

jimklimov, 2025-04-28:
Great, thanks for the report! It seems these devices (or the 10-year old build of the NUT driver) do not actually report the input voltage/current/frequency (momentary values are all zero), and I am not sure if your `ups.load` was negligible during the test, or if it was non-trivial but mis-reported by the device?

jimklimov, 2025-04-28:
I am not sure TBH if there were significant changes with this older driver since NUT v2.7.4 release a decade ago, but now that v2.8.3 is released - would you please be able to make a custom build of NUT (e.g. following https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests) and re-test these devices with the current code base?

You can run such tests right from the build workspace, without installing any build results into the system, as detailed in that wiki article.

penghong123, 2025-04-28:
When I was testing, there was only the battery, without any load, mains power, or other connections.

penghong123, 2025-04-28:
> Great, thanks for the report! It seems these devices (or the 10-year old build of the NUT driver) do not actually report the input voltage/current/frequency (momentary values are all zero), and I am not sure if your `ups.load` was negligible during the test, or if it was non-trivial but mis-reported by the device?太好了，感谢您的举报！似乎这些设备（或 NUT 驱动器的 10 年旧版本）实际上并没有报告输入电压/电流/频率（瞬时值都为零），我不确定您的 `ups.load` 在测试期间是否可以忽略不计，或者它是否非平凡但被设备错误报告了？

Input voltage/current/frequency is not reported (instantaneous values are all zero)，Because I didn't just connect the battery, I didn't take the load and connect to the mains

jimklimov, 2025-04-28:
> Because I only connected the battery, I didn't take the load and connect to the mains

Aha, so to make sure I understand: you tested the UPS when it was not plugged into the wall (`ups.status = OB` is reported indeed), so the `input.*` values are zero validly, and it fed nothing so `ups.load` is also zero correctly - neither is a mistake in the driver or device? Good to hear :)

> Need to test with v2.8.3 , right? If so, I'll do it again

If not too hard - yes, or with `master` branch from Git. The wiki article links to further resources about build prerequisites for different operating systems, although for a serial driver you would need little more than just a compiler.

And probably with "wall" power input and some load (a lamp?) connected, to check that non-zero values are also reported properly, if possible.

And same for Talegent UPS per #2924, ideally.

Thanks in advance!

penghong123, 2025-04-29:
> Aha, so to make sure I understand: you tested the UPS when it was not plugged into the wall (`ups.status = OB` is reported indeed), so the `input.*` values are zero validly, and it fed nothing so [...]

Hi！
   I'm having some problems here, and when I'm executing "./configure --with-doc=all --with-all --with-cgi", I'm always getting some missing dependencies.
   

[image]

Like what:

[image]

[image]
 
[image]

[image]

[image]

[image]

Am I mistaken? Why does it feel so cumbersome.

penghong123, 2025-04-29:
Hi！
    Not easy, not easy. After a busy day, I finally got it out. Painful and tormented, but the result was good.
    Here are the test results using the latest NUT. Since the test results are basically the same, the test results of one of the models are used here.
   Here are the test results using the latest NUT. Since the UPS test results of Ippon and Talgegent are basically the same, Ippon's "2049878 UPS Ippon Na TA 1000 (white)" test results are used here.
`
0.000001	[D5] send_to_all: SETINFO driver.state "init.starting"
**Network UPS Tools 2.8.3.64**-64+g72fbc5801 (development iteration after 2.8.3) - Megatec/Q1 protocol serial driver 1.63
   0.000437	[D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000452	[D5] do_upsconf_args: call do_global_args()
   0.000458	[D3] do_global_args: var='maxretry' val='3'
   0.000484	[D5] do_upsconf_args: confupsname=myups, var=driver, val=blazer_ser
   0.000490	[D5] do_upsconf_args: call main_arg()
   0.000496	[D3] main_arg: var='driver' val='blazer_ser'
   0.000505	[D5] do_upsconf_args: not a main_arg()
   0.000515	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000522	[D6] testval_reloadable: var=driver, oldval=blazer_ser, newval=blazer_ser, reloadable=0, reload_flag=0
   0.000556	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000569	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000581	[D5] do_upsconf_args: confupsname=myups, var=port, val=/dev/ttyCH341USB0
   0.000588	[D5] do_upsconf_args: call main_arg()
   0.000594	[D3] main_arg: var='port' val='/dev/ttyCH341USB0'
   0.000601	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/ttyCH341USB0, reloadable=0, reload_flag=0
   0.000607	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000618	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyCH341USB0"
   0.000627	[D5] do_upsconf_args: confupsname=myups, var=desc, val=My UPS
   0.000633	[D5] do_upsconf_args: call main_arg()
   0.000639	[D3] main_arg: var='desc' val='My UPS'
   0.000657	[D1] Network UPS Tools version 2.8.3.64-64+g72fbc5801 (development iteration after 2.8.3), 64-bit build for x86_64, built with gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0 and configured with flags: --prefix=/usr --sysconfdir=/etc/nut --with-user=nut --with-group=nut --with-dev --with-drivers=blazer_ser --sbindir=/usr/sbin --bindir=/usr/bin --with-drvpath=/usr/lib/nut --datadir=/usr/share/nut --libdir=/lib/x86_64-linux-gnu --enable-Wcolor --enable-configure-debug --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode --with-doc=skip PKG_CONFIG_PATH=/usr/local/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig --with-debuginfo=auto --enable-inplace-runtime='2.8.3.64-64+g72fbc5801 (development iteration after 2.8.3), 64-bit build for x86_64, built with gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0'
   0.000676	[D1] debug level is '6'
   0.000685	[D5] send_to_all: SETINFO driver.debug "6"
   0.000694	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000974	[D1] Can not become_user(nut): not root initially, remaining UID=1000 GID=1000
   0.000997	Can't chdir to /var/state/ups (but we do not require that to dump data): No such file or directory
   0.001005	[D1] Signalling UPS [myups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001016	Can't open /var/state/ups/blazer_ser-myups: No such file or directory
   0.001022	[D1] Request for other driver to exit returned code -1
   0.001028	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001038	[D5] send_to_all: SETINFO device.type "ups"
   0.001046	[D5] send_to_all: SETINFO driver.state "init.device"
   0.137527	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.137584	[D5] send_to_all: SETINFO driver.version "2.8.3.64-64+g72fbc5801"
   0.137596	[D5] send_to_all: SETINFO driver.version.internal "1.63"
   0.137606	[D5] send_to_all: SETINFO driver.name "blazer_ser"
   0.137613	[D5] send_to_all: SETINFO driver.state "init.info"
   0.137621	Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

   0.137635	[D2] Trying megatec protocol...
   0.138559	[D3] send: 'Q1'
   0.397051	[D3] read: '(235.8 000.0 229.9 013 49.9 2.50 55.0 00000001'
   0.397260	[D1] minimize_formatting_string: in-escape: unexpected formatting char: '.'
   0.397287	[D1] minimize_formatting_string: in-escape: unexpected formatting char: '1'
   0.397355	[D5] send_to_all: SETINFO input.voltage "235.8"
   0.397385	[D1] minimize_formatting_string: in-escape: unexpected formatting char: '.'
   0.397478	[D1] minimize_formatting_string: in-escape: unexpected formatting char: '1'
   0.397508	[D5] send_to_all: SETINFO input.voltage.fault "0.0"
[... cut ...]
input.current.nominal: 4.0
input.frequency: 49.9
input.frequency.nominal: 50
input.voltage: 235.7
input.voltage.fault: 0.0
input.voltage.nominal: 230
output.voltage: 230.4
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.48
ups.load: 13
ups.mfr: 222222222222222
ups.model: 2222222222
ups.status: OL
ups.temperature: 55.0
ups.type: online
   1.497202	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.501456	[D5] send_to_all: SETINFO driver.state "cleanup.exit"

`

penghong123, 2025-04-29:
> Aha, so to make sure I understand: you tested the UPS when it was not plugged into the wall (`ups.status = OB` is reported indeed), so the `input.*` values are zero validly, and it fed nothing so [...]

Hi！
    Not easy, not easy. After a busy day, I finally got it out. Painful and tormented, but the result was good.
    Here are the test results using the latest NUT. Since the test results are basically the same, the test results of one of the models are used here.
   Here are the test results using the latest NUT. Since the UPS test results of Ippon and Talgegent are basically the same, Ippon's "2049878 UPS Ippon Na TA 1000 (white)" test results are used here.

```
0.000001	[D5] send_to_all: SETINFO driver.state "init.starting"
**Network UPS Tools 2.8.3.64**-64+g72fbc5801 (development iteration after 2.8.3) - Megatec/Q1 protocol serial driver 1.63
   0.000437	[D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000452	[D5] do_upsconf_args: call do_global_args()
   0.000458	[D3] do_global_args: var='maxretry' val='3'
   0.000484	[D5] do_upsconf_args: confupsname=myups, var=driver, val=blazer_ser
   0.000490	[D5] do_upsconf_args: call main_arg()
   0.000496	[D3] main_arg: var='driver' val='blazer_ser'
   0.000505	[D5] do_upsconf_args: not a main_arg()
   0.000515	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000522	[D6] testval_reloadable: var=driver, oldval=blazer_ser, newval=blazer_ser, reloadable=0, reload_flag=0
   0.000556	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000569	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000581	[D5] do_upsconf_args: confupsname=myups, var=port, val=/dev/ttyCH341USB0
   0.000588	[D5] do_upsconf_args: call main_arg()
   0.000594	[D3] main_arg: var='port' val='/dev/ttyCH341USB0'
   0.000601	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/ttyCH341USB0, reloadable=0, reload_flag=0
   0.000607	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000618	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyCH341USB0"
   0.000627	[D5] do_upsconf_args: confupsname=myups, var=desc, val=My UPS
   0.000633	[D5] do_upsconf_args: call main_arg()
   0.000639	[D3] main_arg: var='desc' val='My UPS'
   0.000657	[D1] Network UPS Tools version 2.8.3.64-64+g72fbc5801 (development iteration after 2.8.3), 64-bit build for x86_64, built with gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0 and configured with flags: --prefix=/usr --sysconfdir=/etc/nut --with-user=nut --with-group=nut --with-dev --with-drivers=blazer_ser --sbindir=/usr/sbin --bindir=/usr/bin --with-drvpath=/usr/lib/nut --datadir=/usr/share/nut --libdir=/lib/x86_64-linux-gnu --enable-Wcolor --enable-configure-debug --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode --with-doc=skip PKG_CONFIG_PATH=/usr/local/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig --with-debuginfo=auto --enable-inplace-runtime='2.8.3.64-64+g72fbc5801 (development iteration after 2.8.3), 64-bit build for x86_64, built with gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0'
   0.000676	[D1] debug level is '6'
   0.000685	[D5] send_to_all: SETINFO driver.debug "6"
   0.000694	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000974	[D1] Can not become_user(nut): not root initially, remaining UID=1000 GID=1000
   0.000997	Can't chdir to /var/state/ups (but we do not require that to dump data): No such file or directory
   0.001005	[D1] Signalling UPS [myups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001016	Can't open /var/state/ups/blazer_ser-myups: No such file or directory
   0.001022	[D1] Request for other driver to exit returned code -1
   0.001028	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001038	[D5] send_to_all: SETINFO device.type "ups"
   0.001046	[D5] send_to_all: SETINFO driver.state "init.device"
   0.137527	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.137584	[D5] send_to_all: SETINFO driver.version "2.8.3.64-64+g72fbc5801"
   0.137596	[D5] send_to_all: SETINFO driver.version.internal "1.63"
   0.137606	[D5] send_to_all: SETINFO driver.name "blazer_ser"
   0.137613	[D5] send_to_all: SETINFO driver.state "init.info"
   0.137621	Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

   0.137635	[D2] Trying megatec protocol...
   0.138559	[D3] send: 'Q1'
   0.397051	[D3] read: '(235.8 000.0 229.9 013 49.9 2.50 55.0 00000001'
   0.397260	[D1] minimize_formatting_string: in-escape: unexpected formatting char: '.'
   0.397287	[D1] minimize_formatting_string: in-escape: unexpected formatting char: '1'
   0.397355	[D5] send_to_all: SETINFO input.voltage "235.8"
[... cut ...]
input.current.nominal: 4.0
input.frequency: 49.9
input.frequency.nominal: 50
input.voltage: 235.7
input.voltage.fault: 0.0
input.voltage.nominal: 230
output.voltage: 230.4
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.48
ups.load: 13
ups.mfr: 222222222222222
ups.model: 2222222222
ups.status: OL
ups.temperature: 55.0
ups.type: online
   1.497202	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.501456	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

jimklimov, 2025-04-29:
Great, thanks! Sorry that you found this routine "painful" though...

> The wiki article links to further resources about build prerequisites for different operating systems

Did you follow these? e.g.  https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt rendered on NUT website as https://networkupstools.org/docs/user-manual.chunked/_build_prerequisites_to_make_nut_from_scratch_on_various_operating_systems.html

Which distribution did you use that it seems to not provide a packaged net-smp etc.? I saw a few like that, mostly embedded, but it is not common...

penghong123, 2025-04-29:
> Great, thanks! Sorry that you found this routine "painful" though...
>
>
> Did you follow these? e.g. https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt rendered on NUT website [...]

Sorry for my question. I didn't look carefully at the documentation, I thought Debian 10/11/12/13 was just Debian 10/11/12/13, but I didn't expect it to contain the relevant configuration and dependency installation instructions for the Ubuntu system.

jimklimov, 2025-04-29:
Well, `--with-all` enables all (over a dozen) optional toggles to "yes" so requires quite a big dependency scope to build "everything" (which is what people looking to replace their packaged NUT installation with a modern one actually do want). The `--with-cgi` here may be an overkill as it is part of what is enabled, but it pulls in X11 so a large dependency footprint that can be easily disabled if not desired (by editing to `--with-cgi=no`).

Fair enough about the unclear title of Debian vs Ubuntu chapter, but when opposed to the likes of Solaris, MacOS, *BSD and Windows as well as several other Linux distribution families further in the chapter, these two are nearly identical :)

penghong123, 2025-04-30:
> Well, `--with-all` enables all (over a dozen) optional toggles to "yes" so requires quite a big dependency scope to build "everything" (which is what people looking to replace their packaged NUT installation with a modern one actually do want). The `--with-cgi` here may be an overkill as it is part of what is enabled, but it pulls in X11 so a large dependency footprint that can be easily disabled if not desired (by editing to `--with-cgi=no`).嗯，`--with-all` 允许所有（十几个）可选切换到 “yes”，所以需要一个相当大的依赖范围来构建“everything”（这就是那些希望用现代 NUT 安装替换打包的 NUT 安装的人真正想要的）。这里的 `--with-cgi` 可能有点矫枉过正，因为它是启用内容的一部分，但它会引入 X11，因此如果不需要，可以很容易地禁用大量的依赖项（通过编辑 `--with-cgi=no`）。
> 
> Fair enough about the unclear title of Debian vs Ubuntu chapter, but when opposed to the likes of Solaris, MacOS, *BSD and Windows as well as several other Linux distribution families further in the chapter, these two are nearly identical :)Debian 与 Ubuntu 章节的标题不明确是公平的，但是当与 Solaris、MacOS、*BSD 和 Windows 以及本章后面的其他几个 Linux 发行版系列相反时，这两个几乎是相同的:)

Got it.

penghong123, 2025-05-09:
Hi!
    I see that these models of UPS have been added to your to-do list, but it seems it will take a long time to see these UPS models on this website "https://networkupstools.org/stable-hcl.html".
    If there are others using the mentioned UPS models (Ippon's Na UPS model and Talgegent's UPS model), then they won't need to try each driver one by one.  I also want to contribute to this open-source project. I wonder what I can do to help speed up this process?

jimklimov, 2025-05-09:
Nice of you to ask and offer help - thanks and much appreciated :)

Primarily this is about changing https://github.com/networkupstools/nut/blob/master/data/driver.list.in for the HCL part, and adding structured data-dump files to https://github.com/networkupstools/nut-ddl/ for the DDL part respectively. Guidelines for the entries are in comments of the HCL file and in README of the DDL repo, so the work is about copy-pasting the bits from user reports like yours into one structured text file or another.

This is a tedious routine in a way, which keeps getting low priority because when someone has a bit of time to donate to the project, it is usually about programming - like adding new drivers or fixing bugs. So the queue of unpublished reports piles up in https://github.com/orgs/networkupstools/projects/3/views/1 :\

penghong123, 2025-05-09:
I noticed that there are many tasks listed at https://github.com/orgs/networkupstools/projects/3/views/1, which is a large workload.
If modifying https://github.com/networkupstools/nut/blob/master/data/driver.list.in and https://github.com/networkupstools/nut-ddl/ files is allowed, I can add the UPS models I submitted earlier to the above files according to the rules. This should help alleviate your workload.
Of course, I also hope that others can see the above UPS models at https://networkupstools.org/stable-hcl.html, as it brings a joy of helping others.Because the first time I used the NUT tool, I went to this https://networkupstools.org/stable-hcl.html website to look for the UPS model I was using and to check its corresponding driver. However, I found that there was none, and the driver configuration took a long time. So I thought that when others use the UPS model I mentioned earlier, they won’t have to spend as much time.

penghong123, 2025-05-09:
> Nice of you to ask and offer help - thanks and much appreciated :)
> 
> Primarily this is about changing https://github.com/networkupstools/nut/blob/master/data/driver.list.in for the HCL part, and adding structured data-dump files to https://github.com/networkupstools/nut-ddl/ for the DDL part respectively. Guidelines for the entries are in comments of the HCL file and in README of the DDL repo, so the work is about copy-pasting the bits from user reports like yours into one structured text file or another.
> 
> This is a tedious routine in a way, which keeps getting low priority because when someone has a bit of time to donate to the project, it is usually about programming - like adding new drivers or fixing bugs. So the queue of unpublished reports piles up in [NUT HCL/DDL (view)](https://github.com/orgs/networkupstools/projects/3/views/1) :\

I noticed that there are many tasks listed at https://github.com/orgs/networkupstools/projects/3/views/1, which is a large workload.
If modifying https://github.com/networkupstools/nut/blob/master/data/driver.list.in and https://github.com/networkupstools/nut-ddl/ files is allowed, I can add the UPS models I submitted earlier to the above files according to the rules. This should help alleviate your workload.
Of course, I also hope that others can see the above UPS models at https://networkupstools.org/stable-hcl.html, as it brings a joy of helping others.Because the first time I used the NUT tool, I went to this https://networkupstools.org/stable-hcl.html website to look for the UPS model I was using and to check its corresponding driver. However, I found that there was none, and the driver configuration took a long time. So I thought that when others use the UPS model I mentioned earlier, they won’t have to spend as much time.

penghong123, 2025-05-11:
> Nice of you to ask and offer help - thanks and much appreciated :)很高兴您提出请求并提供帮助 - 谢谢并非常感谢:)
> 
> Primarily this is about changing https://github.com/networkupstools/nut/blob/master/data/driver.list.in for the HCL part, and adding structured data-dump files to https://github.com/networkupstools/nut-ddl/ for the DDL part respectively. Guidelines for the entries are in comments of the HCL file and in README of the DDL repo, so the work is about copy-pasting the bits from user reports like yours into one structured text file or another.这主要是关于更改 HCL 部分的 https://github.com/networkupstools/nut/blob/master/data/driver.list.in，并分别将结构化数据转储文件添加到 DDL 部分的 https://github.com/networkupstools/nut-ddl/。这些条目的准则位于 HCL 文件的注释和 DDL 存储库的 README 中，因此工作是将用户报告中的位复制粘贴到一个或另一个结构化文本文件中。
> 
> This is a tedious routine in a way, which keeps getting low priority because when someone has a bit of time to donate to the project, it is usually about programming - like adding new drivers or fixing bugs. So the queue of unpublished reports piles up in [NUT HCL/DDL (view)](https://github.com/orgs/networkupstools/projects/3/views/1) :\在某种程度上，这是一个乏味的例行公事，它一直变得越来越低优先级，因为当有人有一点时间为项目捐款时，通常是关于编程的——比如添加新的驱动程序或修复错误。因此，未发布的报告队列堆积在 [NUT HCL/DDL (view)](https://github.com/orgs/networkupstools/projects/3/views/1) ：\

Do you mean that I should prepare the corresponding content in advance based on the requirements from https://github.com/networkupstools/nut/blob/master/data/driver.list.in and https://github.com/networkupstools/nut-ddl/, and then reply to you in the chat window? If there are no issues, will the corresponding UPS models I submitted be visible at https://networkupstools.org/stable-hcl.html?

jimklimov, 2025-05-11:
Ideally that would be a pull request - you fork the project on GitHub, check out the Git code base from your replica, make a branch and changes in it, post that branch back to GitHub and make the pull request in its Web-UI.

No rocket science, but a big time-hog to find and copy the data, adapt the comments from issue discussions into DDL structured notes, etc.

closed 2025-06-03 by jimklimov