networkupstools/nut #2924: Talegent UPS is supported by blazer_ser.

penghong123, 2025-04-28:
Hello!
      I have tested Talegent's  series UPS and found that blazer_ser can support it. I browsed the UPS models that have been cataloged, but I didn't find any UPS with the manufacturer and the corresponding model.
You can include the manufacturer and the corresponding UPS model.

nut :2.7.4
Manufacturer: **Talegent**
Model:
              **
Vitality   1KS
Vitality   2KS
Vitality   3KS
Vitality 6KS
Vitality 10KS
EBP6K-UP
EBP10K-UP
RT01S-36HV-Na
RT02S-72HV-Na
RT03S-96HV-Na
RT06H-192HV     EBP6K/10K-RT     PDU
RT10H-240HV     EBP6K/10K-RT     PDU
EBP2K-RT
EBP3K-RT
EBP6K/10K-RT
**

upsc myups@localhost

```
Init SSL without certificate database
battery.charge: 100
battery.voltage: 40.50
battery.voltage.high: 39.00
battery.voltage.low: 31.20
battery.voltage.nominal: 36.0
device.mfr: 222222222222222
device.model: 2222222222
device.type: ups
driver.name: blazer_ser
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyCH341USB0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 1.57
input.current.nominal: 4.0
input.frequency: 0.0
input.frequency.nominal: 50
input.voltage: 0.0
input.voltage.fault: 0.0
input.voltage.nominal: 230
output.voltage: 229.9
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.48
ups.load: 0
ups.mfr: 222222222222222
ups.model: 2222222222
ups.status: OB
ups.temperature: 49.0
ups.type: online
```

cat /etc/nut/ups.conf

```
maxretry = 3
[myups]                                                              
     driver = blazer_ser                                 
     port = /dev/ttyCH341USB0   
     desc = "My UPS"  
```

cat /var/log/syslog | grep upsd

```
Network UPS Tools - UPS driver controller 2.7.4
  upsd[1331]: mainloop: Interrupted system call
 upsd[1331]: Signal 15: exiting
  upsdrvctl[6494]: Supported UPS detected with megatec protocol
  upsdrvctl[6494]: Vendor information read in 1 tries
upsdrvctl[6494]: No values provided for battery high/low voltages in ups.conf
  upsdrvctl[6494]: Using 'guestimation' (low: 31.200000, high: 39.000000)!
upsdrvctl[6494]: Battery runtime will not be calculated (runtimecal not set)
 upsdrvctl[6493]: Network UPS Tools - UPS driver controller 2.7.4
upsd[6496]: fopen /run/nut/upsd.pid: No such file or directory
  upsd[6496]: listening on 0.0.0.0 port 3493
  upsd[6496]: listening on 0.0.0.0 port 3493
  upsd[6496]: Connected to UPS [myups]: blazer_ser-myups
  upsd[6496]: Connected to UPS [myups]: blazer_ser-myups
  upsd[6497]: Startup successful
  upsd[6497]: User admin@127.0.0.1 logged into UPS [myups]

```

lsusb -vvv

```
Bus 003 Device 005: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 CH340 serial converter
  bcdDevice            2.64
  iManufacturer           0 
  iProduct                2 USB Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1 
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

**Shut down operation: Able to perform the corresponding actions normally.**

sudo upsdrvctl stop

`Network UPS Tools - UPS driver controller 2.7.4`

sudo journalctl -u nut-monitor -f
                                               
```
Log:
4April 28 18:33:17 tianxia-virtual-machine upsmon[1438]: Poll UPS [myups@localhost] failed - Driver not connected
4April 28 18:33:17 tianxia-virtual-machine upsmon[1438]: FSD set on UPS myups@localhost failed: ERR ACCESS-DENIED
4April 28 18:33:17 tianxia-virtual-machine upsmon[1438]: Executing automatic power-fail shutdown
4April 28 18:33:17 tianxia-virtual-machine upsmon[1438]: Auto logout and shutdown proceeding

```

jimklimov, 2025-04-28:
Thanks for this report too! Same comments as #2923 about a few invalid readings seem to apply here.

penghong123, 2025-04-29:
> Thanks for this report too! Same comments as [#2923](https://github.com/networkupstools/nut/issues/2923) about a few invalid readings seem to apply here.

Okay.

jimklimov, 2025-06-03:
DDL and HCL updates handled by PRs linked above.