networkupstools/nut #2925: usbhid-ups - USB often disconnects when losing power or recovering power

N-Mi, 2025-04-28:
Hi,

I recently turned my USB Eaton UPSes into network-aware ones by plugging them into a Raspberry Pi model B running OpenWRT. This very old Pi hardware is nonetheless powerful enough to handle such task, and boots in a few seconds thanks to OpenWRT.

But...

I had a micro-cut this morning just before getting up, and found later that a Raspberry in my attic and the home server powered by this UPS were shutdown.
I did a few tests by unplugging/plugging power on the UPS, and it turns out NUT was shutting down all hosts because it lost USB connection while on battery.

It might be a libusb issue, but I think there might be some settings to tweak (polling interval, ....) to avoid this situation. Can someone help me how to do it ?

OS : 
```
root@ups-grenier1:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="24.10.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0"
VERSION_ID="24.10.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28427-6df0e3d02a"
OPENWRT_BOARD="bcm27xx/bcm2708"
OPENWRT_ARCH="arm_arm1176jzf-s_vfp"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0 r28427-6df0e3d02a"
OPENWRT_BUILD_DATE="1738624177"
```

NUT version :
```
root@ups-grenier1:~# opkg list-installed | grep nut
luci-app-nut - 25.116.36015~cca4654
nut - 2.8.1-r2
nut-common - 2.8.1-r2
nut-driver-usbhid-ups - 2.8.1-r2
nut-server - 2.8.1-r2
nut-upsc - 2.8.1-r2
nut-upscmd - 2.8.1-r2
nut-upsmon - 2.8.1-r2
nut-upsrw - 2.8.1-r2
nut-web-cgi - 2.8.1-r2
```

NUT config :
```
root@ups-grenier1:/etc/nut# cat nut.conf 
MODE=netserver
root@ups-grenier1:/etc/nut# cat ups.conf 
# Config file automatically generated from UCI config
[5p1150]
driver = usbhid-ups
port = auto
productid = ffff
serial = G116E47016
vendorid = 0463

root@ups-grenier1:/etc/nut# cat upsd.conf 
# Config file automatically generated from UCI config
LISTEN 0.0.0.0 
root@ups-grenier1:/etc/nut# cat upsmon.conf 
# Config file automatically generated from UCI config
RUN_AS_USER nutmon
MINSUPPLIES 1
[... 134 lines cut ...]
Mon Apr 28 18:46:25 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Mon Apr 28 18:46:25 2025 daemon.err upsmon[2256]: 39679.563183	[D1] Exiting upsmon immediately after initiating shutdown, by default
Mon Apr 28 18:46:25 2025 daemon.err upsmon[2256]: mount: mounting /overlay on /overlay failed: Invalid argument
Mon Apr 28 18:46:26 2025 kern.info kernel: [ 3556.640096] EXT4-fs (mmcblk0p2): re-mounted ff313567-e9f1-5a5d-9895-3ba130b4a864 ro. Quota mode: disabled.
Mon Apr 28 18:46:27 2025 daemon.err upsmon[2256]: Can't open /var/run/nut/usbhid-ups-5p1150: No such file or directory
Mon Apr 28 18:46:27 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Mon Apr 28 18:46:27 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Mon Apr 28 18:46:27 2025 daemon.err upsmon[2256]: Can't claim USB device [0463:ffff]@0/1: Entity not found
Mon Apr 28 18:46:27 2025 daemon.err upsmon[2256]: upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Mon Apr 28 18:46:28 2025 daemon.info procd: - shutdown -
Mon Apr 28 18:46:28 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Mon Apr 28 18:46:28 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Mon Apr 28 18:46:29 2025 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Mon Apr 28 18:46:30 2025 daemon.notice procd: /etc/rc.d/K28nut-monitor: Network UPS Tools upsmon 2.8.1
Mon Apr 28 18:46:30 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Mon Apr 28 18:46:30 2025 daemon.debug upsd[1750]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Mon Apr 28 18:46:31 2025 daemon.err upsd[1750]: mainloop: Interrupted system call
Mon Apr 28 18:46:31 2025 daemon.info upsd[1750]: Signal 15: exiting
Mon Apr 28 18:46:31 2025 daemon.debug upsd[1750]: [D1] driver_free: forgetting UPS [5p1150] (FD -1)
Connection to 10.0.0.30 closed by remote host.
```

I got the issue either when losing power, or when recovering power, but the consequences are more severe on the latter case, as the shutdown is triggered when losing USB when on battery.

Sometimes after a few retries (I've seen it happen 20s after "New USB device found"), the driver manages to reconnect to the UPS:

jimklimov, 2025-04-28:
Can you please check if you can try with a more recent NUT, perhaps a build from the master branch or recent v2.8.3 release? See https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for details. There were some changes about reconnection in recent releases (one still queued with #1630).
* We also have a backlogged idea for OpenWRT builds at #2586 - if you end up having some suggestions about setting up the build (dependency/tool package names for the platform, etc.), feel welcome to comment there :)

As for the time after an UPS is detected "stale" until `upsmon` triggers the shutdown, it should be managed as `DEADTIME` (default 15s).

jimklimov, 2025-04-30:
Note there are several moving parts here:
* USB link (and the UPS controller on the other side... and/or motherboard side of USB hub/chipset) going awry at the time-point we don't really know when. It may well be that the NUT driver's view of the data from the device was not updated from the moment of outage, for example, but the OS took time to detect the loss of link and removal/recreation of the device. Or it may well be that the `dmesg` report and timing reflects the truth, and the link was useful until that second the power came back :)
  * Possibly higher debug levels when running the driver can help, showing how it receives updates from the device and announces them on local UNIX socket (for `upsd`)
* The NUT driver which at some point no longer got and forwarded the device data updates (and so the `upsd` clock started ticking for "Data Stale"'ness). Also it should detect the `LIBUSB_ERROR` when the device context becomes invalid, and should try to reconnect (or perhaps exit, hoping that systemd or similar would revive another instance with an even cleaner re-connection). This part did evolve a lot in recent NUT releases and even after v2.8.3 already.
* The `upsd` NUT data server which talks to the currently running driver, it refreshes the last-activity time stamp that is considered for "Data Stale" verdict. If the driver does fully restart (or at least reconnect to `upsd`) this may also play a role in raising the verdict quickly.
  * This suggests the driver exited or crashed: `Can't connect to UPS [5p1150] (usbhid-ups-5p1150): No such file or directory`, so more logs are needed to discern.
* The `upsmon` client which decides the situation is serious enough to begin the shutdowns at some point.
* See https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity for more suggestions

N-Mi, 2025-04-30:
Just compiled new packages of NUT 2.8.3 for OpenWRT : 

```
root@ups-grenier1:/tmp/packages# opkg list-installed | grep nut
luci-app-nut - 25.116.36015~cca4654
nut - 2.8.3-r1
nut-common - 2.8.3-r1
nut-driver-usbhid-ups - 2.8.3-r1
nut-server - 2.8.3-r1
nut-upsc - 2.8.3-r1
nut-upscmd - 2.8.3-r1
nut-upsmon - 2.8.3-r1
nut-upsrw - 2.8.3-r1
nut-web-cgi - 2.8.3-r1
```12:59:28

I retried to unplug the UPS power socket, wait a minute, then powering it again, and got the same behaviour as with 2.8.1 : 

```
Wed Apr 30 12:58:02 2025 daemon.err upsmon[7800]:  370.740632	UPS 5p1150@localhost on battery
Wed Apr 30 12:58:25 2025 daemon.warn odhcpd[1467]: No default route present, overriding ra_lifetime to 0!
Wed Apr 30 12:59:03 2025 kern.info kernel: [151886.182423] usb 1-1.3: USB disconnect, device number 6
Wed Apr 30 12:59:04 2025 kern.info kernel: [151887.247741] usb 1-1.3: new low-speed USB device number 7 using dwc_otg
Wed Apr 30 12:59:05 2025 kern.info kernel: [151887.892165] usb 1-1.3: New USB device found, idVendor=0463, idProduct=ffff, bcdDevice= 2.02
Wed Apr 30 12:59:05 2025 kern.info kernel: [151887.907448] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Wed Apr 30 12:59:05 2025 kern.info kernel: [151887.921824] usb 1-1.3: Product: Eaton 5P
Wed Apr 30 12:59:05 2025 kern.info kernel: [151887.932790] usb 1-1.3: Manufacturer: EATON
Wed Apr 30 12:59:05 2025 kern.info kernel: [151887.943524] usb 1-1.3: SerialNumber: G116E47016
Wed Apr 30 12:59:07 2025 kern.info kernel: [151890.077777] hid-generic 0003:0463:FFFF.0004: hiddev96,hidraw2: USB HID v1.10 Device [EATON Eaton 5P] on usb-20980000.usb-1.3/input0
Wed Apr 30 12:59:07 2025 daemon.notice upsd[7717]: Data for UPS [5p1150] is stale - check driver
Wed Apr 30 12:59:07 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:07 2025 daemon.err upsd[7717]: Can't connect to UPS [5p1150] (/tmp/run/nut/usbhid-ups-5p1150): No such file or directory
Wed Apr 30 12:59:07 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:07 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:07 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:07 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:07 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:07 2025 daemon.err upsmon[7800]:  435.862071	Poll UPS [5p1150@localhost] failed - Driver not connected
Wed Apr 30 12:59:07 2025 daemon.err upsmon[7800]:  435.862437	Communications with UPS 5p1150@localhost lost
Wed Apr 30 12:59:07 2025 daemon.err upsmon[7800]:  435.866139	[D1] UPS [5p1150@localhost] was last known to be not fully online and currently is not communicating, just so you know (waiting for DEADTIME to elapse)
Wed Apr 30 12:59:08 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:08 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:10 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:10 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:12 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:12 2025 daemon.err upsmon[7800]:  440.895464	Poll UPS [5p1150@localhost] failed - Driver not connected
Wed Apr 30 12:59:12 2025 daemon.err upsmon[7800]:  440.895853	[D1] UPS [5p1150@localhost] was last known to be not fully online and currently is not communicating, just so you know (waiting for DEADTIME to elapse)
Wed Apr 30 12:59:13 2025 daemon.err procd: instance 5p1150 not found
Wed Apr 30 12:59:13 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:13 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:15 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 12:59:15 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 12:59:17 2025 daemon.debug upsd[7717]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
[... cut ...]
Wed Apr 30 13:02:36 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:02:36 2025 daemon.info upsd[1765]: Connected to UPS [5p1150]: usbhid-ups-5p1150
Wed Apr 30 13:02:36 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is now connected as FD 5
Wed Apr 30 13:02:39 2025 daemon.err upsmon[2269]: 61473.344120	Communications with UPS 5p1150@localhost established
```
(plugged back at 13:02:22)

I will try to run `usb_resetter` to see if it helps.

I will also try to increase log verbosity.

arnaudquette-eaton, 2025-04-30:
standard bells @jimklimov 
@N-Mi
* which exact UPS models?
*  do you have a USB hub between the Pi and the device? if so, self or bus powered?
* have you tried a different USB cable?

N-Mi, 2025-04-30:
>    which exact UPS models?

5P1150 IR (gen1)

>    do you have a USB hub between the Pi and the device? if so, self or bus powered?

No

>    have you tried a different USB cable?

Yes

N-Mi, 2025-04-30:
Seems usb_resetter.py triggers the same behaviour : 

Lauched this command : `logger "##### Launching usb_resetter.py" ; ./usb_resetter.py  --reset-hub --device 0463:ffff ; logger "##### Done running usb_resetter.py"`

Logs :
```
Wed Apr 30 13:53:50 2025 user.notice root: ##### Launching usb_resetter.py
Wed Apr 30 13:53:54 2025 kern.info kernel: [ 3223.535078] hid-generic 0003:0463:FFFF.0007: hiddev96,hidraw2: USB HID v1.10 Device [EATON Eaton 5P] on usb-20980000.usb-1.3/input0
Wed Apr 30 13:53:54 2025 user.notice root: ##### Done running usb_resetter.py
Wed Apr 30 13:53:57 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:53:57 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:53:59 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:53:59 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:00 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:00 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:01 2025 daemon.info usbhid-ups[10278]: Network UPS Tools 2.8.3 release - Generic HID driver 0.62
Wed Apr 30 13:54:01 2025 daemon.info usbhid-ups[10278]: USB communication driver (libusb 1.0) 0.50
Wed Apr 30 13:54:02 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:02 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:02 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:02 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:02 2025 daemon.err upsmon[2269]: 64556.392098	Poll UPS [5p1150@localhost] failed - Driver not connected
Wed Apr 30 13:54:02 2025 daemon.err upsmon[2269]: 64556.392497	Communications with UPS 5p1150@localhost lost
Wed Apr 30 13:54:02 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:02 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:04 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:04 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:05 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:05 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:07 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:07 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:07 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:07 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:07 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:07 2025 daemon.err upsmon[2269]: 64561.421025	Poll UPS [5p1150@localhost] failed - Driver not connected
Wed Apr 30 13:54:07 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:09 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:09 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:10 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:10 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is still not connected (FD -1)
Wed Apr 30 13:54:12 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is not currently connected, trying to reconnect
Wed Apr 30 13:54:12 2025 daemon.info upsd[1765]: Connected to UPS [5p1150]: usbhid-ups-5p1150
Wed Apr 30 13:54:12 2025 daemon.debug upsd[1765]: [D1] mainloop: UPS [5p1150] is now connected as FD 5
Wed Apr 30 13:54:12 2025 daemon.err upsmon[2269]: 64566.436511	Poll UPS [5p1150@localhost] failed - Variable not supported by UPS
Wed Apr 30 13:54:17 2025 daemon.err upsmon[2269]: 64571.440699	Communications with UPS 5p1150@localhost established
```

N-Mi, 2025-04-30:
Same test on another UPS in my house : 
 - model : Eaton Ellipse ECO 1200 USB FR (serial number : G030M29294, firmware "02" according to upsc)
 - Raspberry Pi model B (identical to the other one), running also OpenWRT 24.10, NUT 2.8.1, known-good USB cable

Unplugging/plugging USB cable between the UPS and the Rpi :
```
Wed Apr 30 14:04:13 2025 kern.info kernel: [160819.939369] usb 1-1.3: USB disconnect, device number 5
Wed Apr 30 14:04:15 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:15 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:17 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:17 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:17 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:17 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:17 2025 daemon.err procd: instance el1200 not found
Wed Apr 30 14:04:19 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:19 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:21 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:21 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:22 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:22 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:24 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:24 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:26 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:26 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:26 2025 kern.info kernel: [160833.031407] usb 1-1.3: new full-speed USB device number 6 using dwc_otg
Wed Apr 30 14:04:27 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:27 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:28 2025 kern.info kernel: [160834.986792] usb 1-1.3: New USB device found, idVendor=0463, idProduct=ffff, bcdDevice= 1.00
Wed Apr 30 14:04:28 2025 kern.info kernel: [160835.000945] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=4
Wed Apr 30 14:04:28 2025 kern.info kernel: [160835.014250] usb 1-1.3: Product: Ellipse ECO
Wed Apr 30 14:04:28 2025 kern.info kernel: [160835.021531] usb 1-1.3: Manufacturer: EATON
Wed Apr 30 14:04:28 2025 kern.info kernel: [160835.028554] usb 1-1.3: SerialNumber: 000000000
Wed Apr 30 14:04:29 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:29 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:29 2025 daemon.info usbhid-ups[7290]: Network UPS Tools - Generic HID driver 0.52 (2.8.1)
Wed Apr 30 14:04:29 2025 daemon.info usbhid-ups[7290]: USB communication driver (libusb 1.0) 0.46
Wed Apr 30 14:04:31 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:31 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:31 2025 kern.info kernel: [160838.362272] hid-generic 0003:0463:FFFF.0002: hiddev96,hidraw0: USB HID v10.10 Device [EATON Ellipse ECO] on usb-20980000.usb-1.3/input0
Wed Apr 30 14:04:32 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:32 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:34 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:34 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:36 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:36 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:37 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:37 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is still not connected (FD -1)
Wed Apr 30 14:04:39 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is not currently connected, trying to reconnect
Wed Apr 30 14:04:39 2025 daemon.info upsd[1737]: Connected to UPS [el1200]: usbhid-ups-el1200
Wed Apr 30 14:04:39 2025 daemon.debug upsd[1737]: [D1] mainloop: UPS [el1200] is now connected as FD 5
```

Same behaviour, but connection is established 11-12 seconds (I made several tests) with this UPS, whereas 20 seconds with the 5P1150.

N-Mi, 2025-05-01:
Bank holiday today, so I had some time to do a few other tests with the 5P1150 today.

After setting DEADTIME to 30s, I could unplug/replug the power inlet of the UPS 3 times in a row without triggering emergency shutdown, as the driver could reconnect in time with the UPS. Each time the USB connection was lost only when powering back the UPS  (not when losing power).

Interestingly, the 5 next tries to unplug/plug the power inlet, I had no USB disconnection either when unplugging or plugging power on the UPS. 

I suppose the relay that I can hear triggering when the UPS swiches between online and battery had a bit of corrosion, that cleaned itself after repeating cycles during my tests. This might have caused a delay for supplying power to UPS's mainboard, which in turn caused the USB disconnect.

Setting DEADTIME  was a correct workaround for me, and the disappearing of the USB disconnection was an unexpected bonus caused by repeating tests. The time taken to reconnect to the UPS might still be an issue from a code point-of-view, so if it can help, I can still perform some tests (this time by unplugging the USB cable) with more verbosity as suggested by @jimklimov