networkupstools/nut #2926: Re: APC Netbotz250 Environmental Monitoring

commswire, 2025-04-28:
I was wondering although its not a UPS is it possible to add in an APC Netbotz250 over an snmp connection into Nut-Server as a UPS
I've tried to add it into the ups.conf file and I get a no matching MIB error when using the snmps-ups driver
All i want to pull from the Netbotz is the room temperature so I can use that value to initiate a shutdown trigger to my other UPS's
I can do an snmpwalk to the Netbotz to pull down a lot of MIB values so I was thinking is it possible to build an snmps-ups subdriver to talk to the Netbotz 250 device or am I better to cut my losses and get a supported APC AP9641 network card which has temperature builtin

jimklimov, 2025-04-28:
Technically and logically - probably possible.

There is already support for e.g. Eaton non-UPS devices (ePDUs, ATS/STS switches, etc.) including detected ambient senors and even daisy-chaining of devices (so several ePDUs are represented on one IP address, talking to each other with dedicated serial links). There are also drivers for e.g. IPMI in a blade server chassis serving as power source for individual blades - so logically a PDU in a way too.

This rack monitor would logically fit as a PDU with zero outlets but some sensors available :) In case of `upsmon` monitoring, take care to assign a powervalue=0 to this entry.

A subdriver should be possible, see https://networkupstools.org/docs/developer-guide.chunked/new-drivers.html#snmp-subdrivers for details.

commswire, 2025-04-29:
OK thats great news, now I know I'm not wasting my time with an impossible mission I'll proceed and will invest time into trying to get this to work with our Netbotz250. I only came across NUT a few weeks ago so I'm still very new to this, thanks for taking time to reply back to me its much appreciated

closed 2025-04-29 by commswire