networkupstools/nut #2935: `upsmon -K` always returns true when mode=none

baconwaifu, 2025-05-01:
`upsmon`'s mode check always returns `EXIT_SUCCESS` when MODE=none, which conflicts with the expected and documented behavior of `upsmon -K` only returning success during UPS-related shutdown events, resulting in hook scripts expecting the documented behavior always shutting down the UPS, even on a normal reboot.

`upsmon -K` should either bypass the mode check, or the mode check should always return `EXIT_FAILURE` when this parameter is present.

jimklimov, 2025-05-02:
I don't think any NUT programs parse `nut.conf` directly - it is a set of envvars considered (or not) by init scripts, systemd methods and such.

The `upsmon -K` normally runs in a `nutshutdown` hook script with systemd, or similar on other platforms; it is a sysadmin (or packager's) homework to consider MODE there; it may be toxic to do so though (you can run NUT programs directly e.g. to trigger FSD when leaving for vacation, and not care that you disabled the services for whatever reason).

Puzzled why you have an UPS and a NUT deployment configured with it, but disabled it. You probably have reasons.  :)

The `upsmon -K` return value should only depend on presence of a killpower file in the configured location.

Historically those were in `/etc`, survived a reboot and were deleted by init scripts of a next uptime. I suppose with your disabled services, it never gets removed, so any shutdown is treated as FSD.

Packagers and end-users were long recommended to put it in a tmpfs (e.g. `/run/nut/...`) if their OS has one, so it disappears by virtue of a reboot.

baconwaifu, 2025-05-02:
...And just confirmed that it's a debian packaging issue. They added a stub script in front of `upsmon` that gives it the "exit on mode=none" behavior. My apologies for reporting this to the wrong team. 

And it's not that I disabled it, it's more that I hadn't gotten around to setting it up before I needed to reboot.

closed 2025-05-02 by baconwaifu

baconwaifu, 2025-05-02:
To clarify further, it's 100% due to debian's packaging. The shutdown hook (which runs at every shutdown) calls `upsmon -K` (which debian hijacked with a wrapper script) to check for killpower, and because `mode=none`, the wrapper hits the "print info and return success" branch instead of running the actual `upsmon`, so the hook will *always* think it's a killpower shutdown, regardless of what the real `upsmon` has to say.

It turns out that I had also set up the driver config for it in `ups.conf`, but hadn't gotten around to actually turning on upsmon (which I have since done, "solving" the issue)

jimklimov, 2025-05-02:
Aha, thanks fir analysis!

CC @bigon

bigon, 2025-07-02:
@baconwaifu Can you please open a but in debian for this if it's not already the case?

jimklimov, 2025-07-08:
Thinking of it, while https://github.com/networkupstools/nut/blob/master/scripts/systemd/nutshutdown.in has had a lot of improvements in recent years, and does consult `nut.conf` for some settings, it does not check/care about a value of `MODE` yet. So the bug as reported could also be addressed by adding a check there.