networkupstools/nut #2941: add qt6 support

fow0ryl, 2025-05-05:
please ass support for qt6 only systems

jimklimov, 2025-05-05:
Is this about NUT-Monitor UI client? PRs welcome... need to find someone who knows QT, and Python...

fow0ryl, 2025-05-07:
Yes, the only GUI I'm using ...
NUT-Monitor seems to be available as GTK2 and/or QT5 version only. See  scripts/python/app.

jimklimov, 2025-05-07:
Yes, ok, So for context - the Qt5 variant was introduced in NUT v2.8.0 release (for better or worse, as a separate script file although most of the text is the same) largely to handle the move from obsolete and de-supported Python2 to Python3 (which dropped GTK2 and had Qt5 instead), and with a different approach to localization which proved to be a difficult entanglement when some changes were needed for NUT v2.8.3 release.

That change for Qt5 was graciously contributed by @luke-jr - hence as for many other features before and after that, I have to say "PRs welcome" as someone needs to dig into this and understand what they are doing, usually starting as a way to "scratch their own itch", as it goes :)

I might assist with the surrounding integration (recipes, shell dispatching script, docs, etc.) to complete a PR, but someone else should investigate and make the actual GUI code change.

FWIW, the older (py2) versions are still provided because existing systems might have only the older software ecosystem available "forever" until their capacitors dry out and they no longer boot, but until that time they may well be protected by new UPSes and current NUT builds.

PR #2946 (2025-05-09): Port NUT Monitor to Py3Qt6
This should be in time for the EoS of latest Qt5 version 5.15 on 26 May 2025.
I have ported the NUT Monitor to Py3Qt6, as an addition to the existing Py3Qt5.
The build scripts I have also updated to build either or both, depending on availability.
I tested the new addition with Windows and Ubuntu, as well as `make distcheck`, and all seemed well.

closes #2941
closed 2025-05-09 by jimklimov