networkupstools/nut #2943: Riello IDG 1600 not reporting battery charge

dragmammut, 2025-05-07:
Hi!

I use  Riello IDG 1600 on my unraid server and it's not reporting battery charge.

[image]
[image]

Can anybody help please?

dragmammut, 2025-05-07:
[nut.log](https://github.com/user-attachments/files/20092418/nut.log)

I read to add localcalculation flag, how do I do that?

desertwitch, 2025-05-07:
> [nut.log](https://github.com/user-attachments/files/20092418/nut.log)
> 
> I read to add localcalculation flag, how do I do that?

You raised the wrong debug level, please check again with my screenshot and re-post the log for the **driver** and not the monitor.

desertwitch, 2025-05-07:
> [nut.log](https://github.com/user-attachments/files/20092418/nut.log)
> 
> I read to add localcalculation flag, how do I do that?
> 
> EDIT: I figured it out!
> 
> [image]
> 
> Thanks for the help!

If you could, could you still post the correct driver log please (and not the monitor log) - see my screenshot above.
This would help us maybe identify something that we can change in the driver to have an actual (not guessed) reading.

dragmammut, 2025-05-07:
[image]

I would,ve already but I can't find USB Driver DEbug level setting. ?

desertwitch, 2025-05-08:
Ah, since you're in manual configuration mode, please add to the `ups.conf` (where you added `localcalculation`):

`debug_min 6`

Then follow the instructions above to get the log file for us, thank you! 😎

dragmammut, 2025-05-08:
[nut.log](https://github.com/user-attachments/files/20097476/nut.log)

here you go