networkupstools/nut #2958: nutdrv_qx not detecting GREEN CELL UPS01LCD 600VA - works with blazer_usb

m-stefanski, 2025-05-15:
Working configuration:

```
[nutdev1]
  driver = blazer_usb
  port = auto
  norating
  novendor
  protocol = megatec
  default.battery.voltage.high = 13.8
  default.battery.voltage.low = 10.5
```

I cannot make nutdrv_qx work in any case, but closest to showing something is following config:

```
[nutdev1]
  driver = "nutdrv_qx"
  protocol = "hunnox"
  port = "auto"
  vendorid = "0001"
  productid = "0000"
  product = "MEC0003"
  vendor = "MEC"
  bus = "003"  
```

which gives following output:

```
root@pve:~# /usr/lib/nut/nutdrv_qx -a nutdev1 -DDD
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '3'
   0.000590     [D1] upsdrv_initups...
   0.002361     [D2] Checking device 1 of 6 (1D6B/0003)
   0.002376     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.002378     [D2] Checking device 2 of 6 (0001/0000)
   0.003263     [D2] - VendorID: 0001
   0.003267     [D2] - ProductID: 0000
   0.003269     [D2] - Manufacturer: MEC
   0.003270     [D2] - Product: MEC0003
   0.003272     [D2] - Serial Number: unknown
   0.003274     [D2] - Bus: 003
   0.003275     [D2] - Device: unknown
   0.003277     [D2] - Device release number: 0100
   0.003278     [D2] Trying to match device
   0.003280     [D3] match_function_regex: matching a device...
   0.003322     [D2] Device matches
   0.003326     [D2] Reading first configuration descriptor
   0.003332     [D3] libusb_kernel_driver_active() returned 0
   0.003348     [D2] Claimed interface 0 successfully
   0.003351     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.003356     [D2] Skipping protocol Voltronic 0.06
   0.003359     [D2] Skipping protocol Voltronic-QS 0.07
   0.003360     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.003362     [D2] Skipping protocol Mustek 0.07
   0.003364     [D2] Skipping protocol Megatec/old 0.07
   0.003366     [D2] Skipping protocol BestUPS 0.06
   0.003367     [D2] Skipping protocol Mecer 0.07
   0.003372     [D3] send: Q1
   0.226152     [D3] read: (242.9 000.0 242.7 001 50.1 13.7 --.- 00001001
   0.226175     [D3] send: I
   0.410897     [D3] read: #                                     
   0.410912     [D1] ups_infoval_set: non significant value [ups.firmware]
   0.410917     [D3] send: Q1
   0.633018     [D3] read: (242.9 000.0 243.0 001 50.1 13.7 --.- 00001001
   0.633058     [D3] send: I
   0.817726     [D3] read: #                                     
   0.817742     [D1] ups_infoval_set: non significant value [ups.firmware]
   0.817747     [D3] send: Q1
   1.040236     [D3] read: (243.0 000.0 242.9 001 50.1 13.7 --.- 00001001
   1.040258     [D3] send: I
   1.224779     [D3] read: #                                     
   1.224795     [D1] ups_infoval_set: non significant value [ups.firmware]
   1.224800     [D2] Skipping protocol Zinto 0.06
   1.224802     [D2] Skipping protocol Masterguard 0.02
   1.224808     [D2] Skipping protocol Hunnox 0.01
   1.224809     [D2] Skipping protocol Ablerex 0.01
   1.224811     [D2] Skipping protocol Q1 0.07
   1.224813     Device not supported!
   1.224814     Device not supported!
```

nut-scanner:

```
root@pve:~# nut-scanner -U
Scanning USB bus.
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        vendor = "MEC"
        bus = "003"
```

jimklimov, 2025-05-15:
The first-order question is whether you can try with current NUT code base? Release v2.8.3 is out for some time, and there were some master-branch fixes after it already.

See e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for details; note that to test the built code you don't have to install it until you are satisfied that it works (better) :)

m-stefanski, 2025-05-15:
Thank you - I have just checked with 2.8.3 and using command `./drivers/nutdrv_qx -a nutdev1 -d1 -DDDDDD -x protocol=q1` and configuration from post 1 i got it to display parameters:

```
root@pve:~/nut-2.8.3/nut# ./drivers/nutdrv_qx -a nutdev1 -d1 -DDDDDD -x protocol=q1
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.3.285-285+g6393115cb (development iteration after 2.8.3) - Generic Q* USB/Serial driver 0.43
USB communication driver (libusb 1.0) 0.50
   0.000089     [D1] upsdrv_makevartable...
   0.000108     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.26 (API: 0x01000109)"
   0.000117     [D1] Using USB implementation: libusb-1.0.26 (API: 0x01000109)
   0.000192     [D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000195     [D5] do_upsconf_args: call do_global_args()
   0.000201     [D3] do_global_args: var='maxretry' val='3'
   0.000242     [D5] do_upsconf_args: confupsname=nutdev1, var=driver, val=nutdrv_qx
   0.000245     [D5] do_upsconf_args: call main_arg()
   0.000250     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000253     [D5] do_upsconf_args: not a main_arg()
   0.000256     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000264     [D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.000266     [D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000270     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000275     [D5] do_upsconf_args: confupsname=nutdev1, var=protocol, val=megatec
   0.000278     [D5] do_upsconf_args: call main_arg()
   0.000281     [D3] main_arg: var='protocol' val='megatec'
   0.000284     [D5] do_upsconf_args: not a main_arg()
   0.000287     [D5] do_upsconf_args: process as value
   0.000290     [D6] testvar_reloadable: searching for var=protocol, vartype=2, reload_flag=0
   0.000294     [D6] testvar_reloadable: found var=protocol, val='(null)' => 'megatec', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000297     [D6] testvar_reloadable: verdict for (re)loading var=protocol value: 1
   0.000302     [D5] send_to_all: SETINFO driver.parameter.protocol "megatec"
   0.000306     [D5] do_upsconf_args: confupsname=nutdev1, var=port, val=auto
   0.000309     [D5] do_upsconf_args: call main_arg()
   0.000311     [D3] main_arg: var='port' val='auto'
   0.000315     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000317     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000320     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000325     [D5] do_upsconf_args: confupsname=nutdev1, var=vendorid, val=0001
   0.000327     [D5] do_upsconf_args: call main_arg()
   0.000330     [D3] main_arg: var='vendorid' val='0001'
   0.000332     [D5] do_upsconf_args: not a main_arg()
   0.000334     [D5] do_upsconf_args: process as value
   0.000337     [D6] testvar_reloadable: searching for var=vendorid, vartype=2, reload_flag=0
   0.000340     [D6] testvar_reloadable: found var=vendorid, val='(null)' => '0001', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000346     [D6] testvar_reloadable: verdict for (re)loading var=vendorid value: 1
   0.000350     [D5] send_to_all: SETINFO driver.parameter.vendorid "0001"
   0.000354     [D5] do_upsconf_args: confupsname=nutdev1, var=productid, val=0000
   0.000357     [D5] do_upsconf_args: call main_arg()
   0.000359     [D3] main_arg: var='productid' val='0000'
   0.000362     [D5] do_upsconf_args: not a main_arg()
   0.000365     [D5] do_upsconf_args: process as value
   0.000369     [D6] testvar_reloadable: searching for var=productid, vartype=2, reload_flag=0
   0.000372     [D6] testvar_reloadable: found var=productid, val='(null)' => '0000', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000374     [D6] testvar_reloadable: verdict for (re)loading var=productid value: 1
   0.000378     [D5] send_to_all: SETINFO driver.parameter.productid "0000"
   0.000382     [D5] do_upsconf_args: confupsname=nutdev1, var=product, val=MEC0003
   0.000385     [D5] do_upsconf_args: call main_arg()
   0.000387     [D3] main_arg: var='product' val='MEC0003'
   0.000391     [D5] do_upsconf_args: not a main_arg()
   0.000395     [D5] do_upsconf_args: process as value
   0.000398     [D6] testvar_reloadable: searching for var=product, vartype=2, reload_flag=0
   0.000401     [D6] testvar_reloadable: found var=product, val='(null)' => 'MEC0003', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000404     [D6] testvar_reloadable: verdict for (re)loading var=product value: 1
   0.000407     [D5] send_to_all: SETINFO driver.parameter.product "MEC0003"
   0.000411     [D5] do_upsconf_args: confupsname=nutdev1, var=vendor, val=MEC
   0.000413     [D5] do_upsconf_args: call main_arg()
   0.000416     [D3] main_arg: var='vendor' val='MEC'
   0.000419     [D5] do_upsconf_args: not a main_arg()
   0.000421     [D5] do_upsconf_args: process as value
   0.000427     [D6] testvar_reloadable: searching for var=vendor, vartype=2, reload_flag=0
   0.000430     [D6] testvar_reloadable: found var=vendor, val='(null)' => 'MEC', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
[... 323 lines cut ...]
   0.003805     [D2] - Manufacturer: MEC
   0.003807     [D2] - Product: MEC0003
   0.003808     [D2] - Serial Number: unknown
[... cut ...]
   0.003888     [D2] Kernel driver already detached
   0.003891     [D2] failed to claim USB device: Resource busy
   0.003898     [D2] Kernel driver already detached
   0.003900     [D2] failed to claim USB device: Resource busy
   0.003902     [D2] Kernel driver already detached
   0.003906     Can't claim USB device [0001:0000]@0/0/0: Entity not found
   0.003911     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

Old blazer_usb with 2.8.0 still works, I am not sure what causes this inconsistency

jimklimov, 2025-05-15:
Most likely the "Resource busy" means that a systemd managed instance of the driver appeared and grabbed the devfs node (see wiki for NDE, I can't link right now).

Although the driver does seem to try contacting an earlier instance by socket and pid file, and can't...

Maybe the controller hung, so if there is no other driver program running - try to re-plug the cable and/or a software equivalent of that (e.g. check `scripts/usb_resetter`)...

jimklimov, 2025-05-23:
Cheers, one thing that looks concerning here is `Can't chdir to /var/state/ups` - did you create and chown it properly?

My current theory, based on `failed to claim USB device: Resource busy` in later attempts, is that once you've saved the working `protocol=q1` into `ups.conf` ("HOWEVER when I change protocol in config file to q1 I get the following error..."), then NDE https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE) kicks in to finally generate a successfully working unit named like `nut-driver@ups.service` which grabs the libusb device and so your subsequent runs can not. And due to lack of location for socket/PID files of the driver, this can not be detected either (`Can't open /var/state/ups/nutdrv_qx-ups: No such file or directory`).

Although I'd expect the service-wrapped driver to fail and die when it can't create the socket, so maybe thanks to systemd you have a looping attempt to spawn the driver, it dies, systemd tries again - so much of the time the USB device is busy with a new short-lived driver instance?