networkupstools/nut #2960: ina219 driver µW instead of mW

renedis, 2025-05-17:
Stating from #2430 / #2378 I finally got the ina219 driver working with the [Waveshare CM4-POE-UPS-BASE](https://www.waveshare.com/wiki/CM4-POE-UPS-BASE). Kudos to the people making this work.

How ever, i found out that my device is stating watt usage as "µW" instead of "mW".

When i exectue "sensors":
```
# sensors
ina219-i2c-10-43
Adapter: i2c-22-mux (chan_id 1)
in0:           0.00 V
in1:           4.19 V
power1:        0.00 W
curr1:         0.00 A
```

Not a real big problem cause I can easily convert it with a dummy-ups. Can @jviki or @3dti or @desertwitch replicate this issue?
And if there any interest for re-implementing this in the driver so it shows as "mW" instead of "µW"?
.
.
.
Anyway, I've created a dummy UPS and also included more statistics. Maybe a bit overkill for some, needs some knowledge of bash coding and understanding battery load etc. It's a one-time-run script so i've made a service that runs each 15 seconds. Can image that implenting this in to existing driver is to much of a long shot because of the dependency of a UniFi switch.

In short; it's a batch script that reads from original ina219 driver and from a UniFi PoE port (API) and injects it. When detected that PoE is down (no voltage income) it changes to the values of the total battery capacity. In this way I can have "load" detection and correct wattage usage.
.
.
.
Here's the script if anyone is interested:
```
#!/bin/bash

# === CONFIGURATION ===

REAL_UPS="xxxxxxxxx"                            # The name of the real INA219 UPS device defined in your NUT ups.conf. Used to pull actual sensor values
VIRTUAL_FILE="/etc/nut/virtual-cm401.status"    # Path to the output file used by the dummy UPS driver. Recommended to keep it under /etc/nut/

# UniFi Controller Access
CONTROLLER="https://x.x.x.x"                    # IP or hostname of your UniFi controller (e.g. UDM Pro)
USERNAME="username"                             # UniFi local user account (not SSO)
PASSWORD="password"                             # Password for the local user
SITE="default"                                  # Site name in UniFi. Use "default" if you only have one site
TARGET_SWITCH="switch"                          # Name of the switch as shown in the UniFi controller
TARGET_PORT=1                                   # Port number where the CM4 is connected (e.g. port 1)

# Battery characteristics (3x 18650 3500 mAh in parallel)
BATTERY_CAPACITY_WH=37.8                        # Total battery capacity in watt-hours. Formula: 3 cells × 3500 mAh = 10.5 Ah × 3.6 V = 37.8 Wh
BATTERY_DERATING=0.85                           # Efficiency derating factor (to account for aging, inverter losses, or imperfect batteries)
REALPOWER_NOMINAL=15.0                          # Max PoE power expected when running on a PoE switch. 15.0 W is standard for Waveshare PoE UPS HAT/CM4
MAX_BATTERY_DISCHARGE_W=37.0                    # Max power draw when running purely on battery. Typical 1C cells do a maximum wattage draw of ~37.0 W. Conversative safe setting is ~25.0 W

# === GET INA219 VALUES ===
INA_OUT=$(upsc "$REAL_UPS@localhost" 2>/dev/null)

if [[ -z "$INA_OUT" ]] || ! echo "$INA_OUT" | grep -q "^battery\.voltage:"; then
  echo "Failed to read INA219 data from $REAL_UPS"
  exit 1
fi

# === Fix battery percentage ===
reported_charge=$(echo "$INA_OUT" | grep '^battery.charge:' | awk '{print $2}')
battery_charge=$(awk "BEGIN {
  if ($reported_charge == \"\") c = 0;
  else {
    c = ($reported_charge / 94) * 100;
    if (c > 100) c = 100;
  }
  printf \"%.0f\", c
}")

# === UniFi API: Login and Retrieve PoE Info ===
COOKIE_JAR=$(mktemp)
[... 219 lines cut ...]
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: dummy-ups
driver.parameter.mode: dummy
driver.parameter.pollinterval: 2
driver.parameter.port: /etc/nut/virtual-cm401.status
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.3
driver.version.internal: 0.20
input.current: 0.000
input.power: 3.08
input.voltage: 0.00
output.realpower: 3.08
ups.firmware: v1.0.0
ups.load: 8
ups.mfr: Texas Instruments
ups.mode: line-interactive
ups.model: INA219
ups.realpower: 3.08
ups.realpower.nominal: 37.0
ups.status: OB DISCHRG
ups.temperature: 42.4
ups.type: standby
```

jimklimov, 2025-05-17:
NOTE: NUT drivers should show stuff in units (mostly SI) listed in `docs/nut-names.txt` for respective data points; possibly floating-point with leading zeroes if the measurement is way below 1 unit. Some drivers do have run-time config flags to apply certain factors (e.g. some UPSes needed 10x or 1x depending on firmware revision).

jviki, 2025-05-17:
> Stating from [#2430](https://github.com/networkupstools/nut/pull/2430) / [#2378](https://github.com/networkupstools/nut/issues/2378) I finally got the ina219 driver working with the [Waveshare [...]

Hi, quite frankly, I do not see any issue here. The hwmon_ina219 driver does not work with power at all, it only reads current and voltage:

https://github.com/networkupstools/nut/blob/f0bb5ad1fd74cda390630b12d053f30dc7e1c716/drivers/hwmon_ina219.c#L70

By the way, the reported power is a derived value, the INA219 device itself measures current and voltage while the power value is computed by multiplication (see datasheet: https://www.ti.com/document-viewer/ina219/datasheet for details).

Anyway, the output from sensors is not relevant here (these are interpreted values for better human readability), it is better to look directly into /sys. This output on my device when running on battery:

```
$ cat /sys/class/hwmon/hwmon2/{in0_input,in1_input,power1_input,curr1_input,shunt_resistor}
-4
4228
176000
-40
100000
```

Output from sensors:

```
$ sensors
ina219-i2c-10-43
Adapter: i2c-22-mux (chan_id 1)
in0:          -0.00 V  
in1:          +4.22 V  
power1:      174.00 mW 
curr1:        -0.04 A
```

You can see that the raw value from /sys contains voltage in mV, power in uW and current in mA (and resistance in uOhms). This is according to the Linux Kernel documentation here: https://docs.kernel.org/hwmon/ina2xx.html.

If you have any issue with the hwmon_ina219 driver, could you please clarify what exactly is wrong?

renedis, 2025-05-18:
> Hi, quite frankly, I do not see any issue here. The hwmon_ina219 driver does not work with power at all, it only reads current and voltage:
> 
> [nut/drivers/hwmon_ina219.c](https://github.com/networkupstools/nut/blob/f0bb5ad1fd74cda390630b12d053f30dc7e1c716/drivers/hwmon_ina219.c#L70)
> 
> Line 70 in [f0bb5ad](/networkupstools/nut/commit/f0bb5ad1fd74cda390630b12d053f30dc7e1c716)
> 
>  /** 
> [nut/drivers/hwmon_ina219.c](https://github.com/networkupstools/nut/blob/f0bb5ad1fd74cda390630b12d053f30dc7e1c716/drivers/hwmon_ina219.c#L225)
> 
> Line 225 in [f0bb5ad](/networkupstools/nut/commit/f0bb5ad1fd74cda390630b12d053f30dc7e1c716)
> 
>  static int update_voltage(void) 
> By the way, the reported power is a derived value, the INA219 device itself measures current and voltage while the power value is computed by multiplication (see datasheet: https://www.ti.com/document-viewer/ina219/datasheet for details).
> 
> Anyway, the output from sensors is not relevant here (these are interpreted values for better human readability), it is better to look directly into /sys. This is output on my device when running on battery:
> 
> ```
> $ cat /sys/class/hwmon/hwmon2/{in0_input,in1_input,power1_input,curr1_input,shunt_resistor}
> -4
> 4228
> 176000
> -40
> 100000
> ```
> 
> Output from sensors:
> 
> ```
> $ sensors
> ina219-i2c-10-43
> Adapter: i2c-22-mux (chan_id 1)
> in0:          -0.00 V  
> in1:          +4.22 V  
> power1:      174.00 mW 
> curr1:        -0.04 A
> ```
> 
> You can see that the raw value from /sys contains voltage in mV, power in uW and current in mA (and resistance in uOhms). This is according to the Linux Kernel documentation here: https://docs.kernel.org/hwmon/ina2xx.html.
> 
> If you have any issue with the hwmon_ina219 driver, could you please clarify what exactly is wrong?

I found it weird that power is in uW instead of mW and was assuming that it should show mW. I came across this when I was building the bash script and it didnt made sense to me. But if this correct according to the Linux Kernel then it's setup correctly as is :)

Thank you for clarifying.

closed 2025-05-18 by renedis