networkupstools/nut #2964: Cannot configure Tescom UPS

grego1981, 2025-05-20:
Hi all!

I have replaced an APC UPS with a Tescom one on my raspberry nut server. I have ran nut-scanner and it produces the following output:

```
[nutdev1]
        driver = "nutdrv_atcl_usb"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        bus = "001"
```

I tried to use the proposed config but it does not work (I'm missing something obviously) anyway I used the following config and it seems to work

```
[nutdev1]
driver = nutdrv_qx
override.battery.packs = 1
port = auto
```

But all values come up zeroed and only status is shown (OL) see below. Is there a way to get data from this UPS? 

```
root@upssrv02:~# upsc nutdev1@localhost
Init SSL without certificate database
battery.packs: 1
battery.voltage: 0.00
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Q1 0.07
driver.version.internal: 0.28
input.frequency: 0.0
input.voltage: 0.0
input.voltage.fault: 0.0
output.voltage: 0.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 0
ups.productid: 0000
ups.status: OL
ups.temperature: 0.0
ups.type: online
ups.vendorid: 0001
```

Thank you and best regards to all!

jimklimov, 2025-05-20:
Try to start the driver with a higher debug verbosity to see what strings the UPS sends back and if it spews zeroes or your driver build misinterprets something. For details see https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity

With NUT v2.7.4 (note it is antique) it may be easier to stop the driver running under a service umbrella (if any), e.g. `systemctl stop nut-driver` and start it from command line like `/lib/nut/nutdrv_qx -a nutdev1 -DDDDDD` and then "Ctrl+C" it after it begins regularly looping for data updates - and post here whatever it wrote to console until then.

Consider custom-building a newer NUT version, see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about testing a newly built driver and eventually replacing the existing installation if the build is better. It is possible your device talks some newer Qx dialect, support for which could have been added in the past decade since NUT v2.7.4 was released, and would convey more data via those newer queries.

grego1981, 2025-05-20:
Hi! Below the output of the proposed command, it really shows zeros...

```
root@upssrv02:~# /lib/nut/nutdrv_qx -a nutdev1 -DDDDDD
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '6'
   0.002232     upsdrv_initups...
   0.003205     Checking device (0001/0000) (001/003)
   0.004682     - VendorID: 0001
   0.004861     - ProductID: 0000
   0.005037     - Manufacturer: unknown
   0.005195     - Product: MEC0003
   0.005367     - Serial Number: unknown
   0.005520     - Bus: 001
   0.005868     - Device release number: 0100
   0.006061     Trying to match device
   0.006221     Device matches
   0.006636     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.006834     send_to_all: SETINFO ups.vendorid "0001"
   0.007011     send_to_all: SETINFO ups.productid "0000"
   0.007315     send: QGS
   0.007528     read: QGS
   0.007710     qx_process_answer: short reply (input.voltage)
   0.007967     send: QGS
   0.008138     read: QGS
   0.008303     qx_process_answer: short reply (input.voltage)
   0.008462     send: QGS
   0.008525     read: QGS
   0.008672     qx_process_answer: short reply (input.voltage)
   0.008861     send: M
   0.009014     read: M
   0.009202     voltronic_qs_protocol: invalid protocol [M]
   0.009343     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.009405     send: M
   0.009462     read: M
   0.009520     voltronic_qs_protocol: invalid protocol [M]
   0.009577     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.009636     send: M
   0.009692     read: M
   0.009750     voltronic_qs_protocol: invalid protocol [M]
   0.009807     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.009865     send: M
   0.009921     read: M
   0.009981     voltronic_qs_hex_protocol: invalid protocol [M]
   0.010038     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.010097     send: M
   0.010153     read: M
   0.010212     voltronic_qs_hex_protocol: invalid protocol [M]
   0.010270     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.010328     send: M
   0.010384     read: M
   0.010442     voltronic_qs_hex_protocol: invalid protocol [M]
   0.010500     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.010559     send: QS
   0.010616     read: QS
   0.010676     qx_process_answer: short reply (input.voltage)
   0.010734     send: QS
   0.010790     read: QS
   0.010943     qx_process_answer: short reply (input.voltage)
   0.011098     send: QS
   0.011251     read: QS
   0.011405     qx_process_answer: short reply (input.voltage)
   0.011520     send: D
   0.011671     read: D
   0.011832     qx_process_answer: short reply (input.voltage)
   0.011967     send: D
   0.012024     read: D
   0.012081     qx_process_answer: short reply (input.voltage)
   0.012140     send: D
[... 337 lines cut ...]
  12.112904     update_status: !LB
  12.113006     update_status: !CAL
  12.113144     update_status: !FSD
  14.110811     upsdrv_updateinfo...
  14.110978     Quick update...
  14.111099     send: Q1
  14.114170     received 47 (40)
  14.114322     read: (47 bytes) => 28 30 30 30 2e 30 20 30 30 30 2e 30 20 30 30 30 2e 30
  14.114462      20 30 30 30 20 30 30 2e 30 20 30 2e 30 30 20 30 30 2e 30 20 30 30 30 30 30
  14.114573      30 30 30 0d
  14.114659     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
  14.114766     update_status: OL
  14.114883     update_status: !LB
  14.115003     update_status: !CAL
  14.115108     update_status: !FSD
  15.444198     send_to_one: PONG
  16.111544     upsdrv_updateinfo...
  16.111685     Quick update...
  16.111814     send: Q1
  16.115163     received 47 (40)
  16.115340     read: (47 bytes) => 28 30 30 30 2e 30 20 30 30 30 2e 30 20 30 30 30 2e 30
  16.115477      20 30 30 30 20 30 30 2e 30 20 30 2e 30 30 20 30 30 2e 30 20 30 30 30 30 30
  16.115583      30 30 30 0d
```
Do you think I should try upgrading to the latest and see?

desertwitch, 2025-05-21:
It does really seem to be sending the zero values, I'd still try updating to the latest NUT version.
There are some protocol related errors, so maybe some later change has better understanding of what's sent.
You see the NUT 2.7.4 is years old at this point and there's been a ton of useful changes since that was released...

grego1981, 2025-05-22:
I have tested all protocol values to no avail.. I also try to compile the 2.8.3 version on a raspiZero2W which has a minimal raspbian OS and when running make following the article https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests the system crashes at some point... any ideas or tested newer versions that 2.7.4 for raspiZero2W?

grego1981, 2025-06-04:
Yes the system keeps crashing, I made the make with parallel threads and monitored to temperature so as to use Ctrl+S/Ctrl+Q but there is a time the sytem crashes... Is there a binary file to use for raspbian Bullseye (11)?

jimklimov, 2025-06-04:
Also, try to `make` without parallelization e.g. `make -j 1` or no `-j` / `-l` (GNU Make limit by load average) arguments at all. That should reduce load on the raspiZero too.

Although if the stress from the build really does cause it to crash, other stresses during its "normal" work might end up doing so too. You might want to mitigate this before putting it into production anyway.

I had something like this with a larger Raspberry Pi5 in the NUT CI farm, where the first tried USB sources were not 5V/5A so the system felt under-voltaged (said so in `dmesg`) and sometimes reset itself not just when building but merely `apt upgrade`'ing or copying files.

If it is about temperature - consider a heat sink (and good thermal paste/pad). Bigger Raspberries are nowadays routinely actively cooled, not sure about zeroes.

grego1981, 2025-06-04:
> Not immediately, I think. Probably can spin up a container to create and share a build (in lieu of better approach like using OBS, at the moment), but can't make time for that today.

This would be awesome and of course I expect it on your own time, no rush, I appreciate your involvement!

grego1981, 2025-06-04:
> Also, try to `make` without parallelization e.g. `make -j 1` or no `-j` / `-l` (GNU Make limit by load average) arguments at all. That should reduce load on the raspiZero too.
>
> Although if the [...]

I'll try it again, disabling all running processes this time as well!

tzalistar, 2025-09-08:
Ill try rebuilding from master later on. Thanks for the tip!

tzalistar, 2025-10-01:
> NOTE: The empty `battery.runtime` may be related to [#3075](https://github.com/networkupstools/nut/issues/3075) and presumed solved by [#3076](https://github.com/networkupstools/nut/pull/3076). That PR was merged - try rebuilding NUT from GitHub master branch.

Just letting you know, that compiling from the master branch has indeed fixed the issue, many thanks for your suggestion!