networkupstools/nut #2965: CI farm issue?

desertwitch, 2025-05-21:
Pending - Running 380 'slow build' dynamatrix stages: Started:733 (of which Restarted:**353**) Completed:**732** FinishedOK:358

Just noticed this on the last commit while I was building from master, just thought I'd ping you about it @jimklimov, if not already aware.

jimklimov, 2025-05-21:
Oh, that is not common; maybe disk space on some worker ran out so it's trying to re-queue and gets the same one. Thanks!

desertwitch, 2025-05-21:
> Oh, that is not common; maybe disk space on some worker ran out so it's trying to re-queue and gets the same one. Thanks!

Thanks, did just see this in one of the Eaton PR runs, if that helps: `java.nio.file.FileSystemException: /home/abuild/jenkins-nutci-netbsd-92-amd64/workspace/nut_nut_PR-2956@tmp/durable-e28915cb: No space left on device`

jimklimov, 2025-05-21:
Hm, this one pops put occasionally. When unwanted/obsoleted build jobs are killed off, this agent tends to not drop their half-filled workspaces :\

closed 2025-05-22 by desertwitch

jimklimov, 2025-05-22:
Probably the effect of being under load. Gotta check if ZFS got "too full", sometimes much time is spent in kernel as it looks for big enough holes to place new writes into sequentially.

jimklimov, 2025-05-23:
FWIW, cleaned up older snapshots and rebooted for good measure; should crawl less :)