networkupstools/nut #2967: Missing Salicru driver support on Unraid (Slackware)

Angryhellfire, 2025-05-22:
I got two Salicru **SLC-700-TWIN RT2** and would like to use them with Unraid.
apcupsd recognizes my UPS but displays incorrect or missing values. Therefore, I wanted to switch to NUT. However, NUT doesn't recognize the UPS at all. After some analysis, I noticed that driver support for Salicru is missing in the Slackware version.

Of course, salicru-hid.c and salicru-hid.h can't simply be imported into Unraid/Slackware.

What options are there? NUT-master has 279 driver files, while NUT Slackware only has 68.

The UPS has Vendor ID 0102. IDs 0101, 0201, 0202, 0203, 0300 and 0302 are listed in the Salicru driver.

**Unraid VendorID:**
Bus 002 Device 002 Port 2-2 ID 2e66:**0102** Salicru TWIN700-RT

**Unraid /dev:**
/dev/bus/usb/002/002 - Salicru_TWIN700-RT_0000000000000000

**NUT master driver:**
/* USB IDs device table */
static usb_device_id_t salicru_usb_device_table[] = { 
/* Salicru SPS 3000 ADV RT2 */ 
/* https://www.salicru.com/sps-3000-adv-rt2.html */ 
{ USB_DEVICE(SALICRU_VENDORID, 0x**0101**), NULL }, 

/* TWINPRO3/TWINRT3 (SLC-1500-TWIN PRO3) via https://github.com/networkupstools/nut/issues/1142 */ 
/* SLC TWIN PRO2<=3KVA via https://github.com/networkupstools/nut/issues/450 */ 
{ USB_DEVICE(SALICRU_VENDORID, 0x**0201**), NULL }, 
{ USB_DEVICE(SALICRU_VENDORID, 0x**0202**), NULL }, 
{ USB_DEVICE(SALICRU_VENDORID, 0x**0203**), NULL }, 

/* Salicru SPS 850 HOME via https://github.com/networkupstools/nut/pull/1199 */ 
/* https://www.salicru.com/sps-home.html */ 
{ USB_DEVICE(SALICRU_VENDORID, 0x**0300**), NULL }, 

/* Salicru SPS 850 ADV T, see https://github.com/networkupstools/nut/issues/1416 */ 
/* https://www.salicru.com/sps-850-adv-t.html */ 
{ USB_DEVICE(SALICRU_VENDORID, 0x**0302**), NULL }

[image]

Thanks in advance!

jimklimov, 2025-05-22:
NOTE: *VendorID* is `2e66` shared by all devices of that vendor.

The NUT driver source you quote is at https://github.com/networkupstools/nut/blob/7e36b658e5f3ecae6a017974b91f0440d0fd38e0/drivers/salicru-hid.c#L42-L64

It is 2 years old now (released in NUT v2.8.1) though still the latest change about it. Also with NUT v2.8.1 or newer you can use the `subdriver=salicru` (or similar) option along with a `productid=0102` in the `ups.conf` as a field work-around for drivers to try supporting devices they did not know at time of coding/building.

If this does work, we'd welcome a PR to include a line with the new Product ID for this to work out of the box with future NUT releases.

CC @desertwitch regarding Unraid and Slackware, if there would be specific questions about that (e.g. adding the `subdriver` option via UI?)

jimklimov, 2025-05-22:
> NUT-master has 279 driver files, while NUT Slackware only has 68.

I hope you do realize the project is maintained in a compiled (not interpreted etc.) language, so many source files can be combined into a different amount of ultimate binary programs and/or libraries? ;)

I've checked with a local rebuild, there seem to be 72 programs made currently under `drivers`, although some could have been added after NUT v2.8.3 release (or what have you in unmentioned NUT package).

> Of course, salicru-hid.c and salicru-hid.h can't simply be imported into Unraid/Slackware.

As such - no. But now that you mention it, there is a long-brewing plan for drivers based on standard low-level protocols like SNMP or USB HID, and mere mapping tables (no/little custom logic) needed to support specific devices, to actually dynamically load such tables from "DMF" resource files. Then the data needed to support a newer device in an older NUT installation could be a matter of copying or writing a text file.

There is a prototype waiting to be reviewed and upstreamed for this, queued and getting bumped back in priority for a few years now. But this is not something that would help you today anyway (the code made already covers the base DMF library and an SNMP support on top of it, not USB).

desertwitch, 2025-05-22:
Not near a computer right now, but what are you basing that the driver is missing on? I'm almost certain it has it, you might need to set "UPS Driver" to "Other (not listed)" in the GUI and write the driver name into the textbox to the right - the dropdown only includes the more popular drivers for now. I'll check back when I'm home with more info.

Angryhellfire, 2025-05-22:
> The screenshot is from apcupsd

Yes I know, just to show the reason why I would like to use NUT ;)

> use the `subdriver=salicru` (or similar) option along with a `productid=0102` in the `ups.conf` as a field work-around for drivers to try supporting devices they did not know at time of coding/building.

Thank you! For now it recognized at least the UPS.

> I hope you do realize the project is maintained in a compiled (not interpreted etc.) language, so many source files can be combined into a different amount of ultimate binary programs and/or libraries? ;)

No I wasn't aware =) I just saw the missing files in the nut folder on Unrraid.
So they are compiled/included in the usbhid-ups?

[image]

Now I need to figure out how to map the values correct. For now it thinks it is on battery and initiates a shutdown ;)

> May 22 18:37:44 SRVxxx ool www[2249959]: /usr/local/emhttp/plugins/nut-dw/scripts/start
> May 22 18:37:45 SRVxxx rc.nut: Writing NUT configuration...
> May 22 18:37:46 SRVxxx rc.nut: Updating permissions for NUT...
> May 22 18:37:46 SRVxxx rc.nut: Checking if the NUT Runtime Statistics Module should be enabled...
> May 22 18:37:46 SRVxxx rc.nut: Disabling the NUT Runtime Statistics Module...
> May 22 18:37:47 SRVxxx rc.nut: Network UPS Tools upsdrvctl - UPS driver controller 2.8.3.313-313+g8f4340afa (development iteration after 2.8.3)
> May 22 18:37:47 SRVxxx rc.nut: Network UPS Tools 2.8.3.313-313+g8f4340afa (development iteration after 2.8.3) - Generic HID driver 0.65
> May 22 18:37:47 SRVxxx rc.nut: USB communication driver (libusb 1.0) 0.50
> May 22 18:37:48 SRVxxx rc.nut: Using subdriver: Salicru HID 0.4
> May 22 18:37:49 SRVxxx rc.nut: Listening on socket /var/run/nut/usbhid-ups-ups
> May 22 18:37:49 SRVxxx usbhid-ups[2251897]: Listening on socket /var/run/nut/usbhid-ups-ups
> May 22 18:37:49 SRVxxx usbhid-ups[2251949]: Startup successful
> May 22 18:37:50 SRVxxx rc.nut: Network UPS Tools upsd 2.8.3.313-313+g8f4340afa (development iteration after 2.8.3)
> May 22 18:37:50 SRVxxx upsd[2252017]: listening on 0.0.0.0 port 3493
> May 22 18:37:50 SRVxxx rc.nut: listening on 0.0.0.0 port 3493
> May 22 18:37:50 SRVxxx upsd[2252017]: Connected to UPS [ups]: usbhid-ups-ups
> May 22 18:37:50 SRVxxx rc.nut: Connected to UPS [ups]: usbhid-ups-ups
> May 22 18:37:50 SRVxxx usbhid-ups[2251949]: sock_connect: enabling asynchronous mode (auto)
> May 22 18:37:50 SRVxxx upsd[2252017]: Found 1 UPS defined in ups.conf
> May 22 18:37:50 SRVxxx rc.nut: Found 1 UPS defined in ups.conf
> May 22 18:37:50 SRVxxx upsd[2252018]: Startup successful
> May 22 18:37:50 SRVxxx rc.nut: Network UPS Tools upsmon 2.8.3.313-313+g8f4340afa (development iteration after 2.8.3)
> May 22 18:37:50 SRVxxx rc.nut: UPS: ups@127.0.0.1 (primary) (power value 1)
> May 22 18:37:50 SRVxxx rc.nut: Using power down flag file /etc/nut/no_killpower
> May 22 18:37:50 SRVxxx upsmon[2252021]: Startup successful
> May 22 18:37:50 SRVxxx upsmon[2252021]: Warning: running as one big root process by request (upsmon -p)
> May 22 18:37:50 SRVxxx upsd[2252018]: User monuser@127.0.0.1 logged into UPS [ups]
> May 22 18:37:50 SRVxxx upsmon[2252021]: **UPS ups@127.0.0.1 on battery**
> May 22 18:37:50 SRVxxx xnut-notify-hooks.sh: ONBATT commands are now executing in background...
> May 22 18:37:50 SRVxxx nut-notify: [ups] **UPS is on battery. The system will shutdown in 240 seconds.**
> May 22 18:38:29 SRVxxx ool www[2252092]: /usr/local/emhttp/plugins/nut-dw/scripts/write_config
> May 22 18:38:29 SRVxxx rc.nut: Writing NUT configuration...
> May 22 18:38:30 SRVxxx rc.nut: Updating permissions for NUT...
> May 22 18:38:30 SRVxxx rc.nut: Checking if the NUT Runtime Statistics Module should be enabled...
> May 22 18:38:30 SRVxxx rc.nut: Disabling the NUT Runtime Statistics Module...
> May 22 18:38:43 SRVxxx emhttpd: cmd: /usr/local/emhttp/plugins/nut-dw/scripts/usbdiagnostics

Thank you so far!

desertwitch, 2025-05-22:
Another thing, in the meantime, which I usually don't recommend is trying an older iteration of NUT. For more exotic devices sometimes ancient NUT versions do provide other readings, especially if later iterations changed driver code based on newer and/or entirely different models. For what it's worth, there is a report of a `TWIN` device (although that may not mean much) with NUT 2.7.4 providing somewhat sane readings. You could try switching the "backend": NUT Settings -> NUT Backend Switch -> legacy (2.7.4...) -> Reinstall the plugin or reboot your server -> Confirm you're on nut-2.7.4... and try again. If it doesn't work, just switch back.

In any case, please do provide us with a debug log of the most recent version (the one you're on now) first. Ideally we can fix something up that'll work for you with the latest NUT version, which really is a lot more advanced than 2.7.4 was years ago. 😉

Angryhellfire, 2025-05-22:
Thank you again for your efforts and help.
Attached the nut.log file. It seems communications timed out and it can't get the needed values?

[nut.log](https://github.com/user-attachments/files/20400731/nut.log)

jimklimov, 2025-05-23:
Did it determine that the UPS is on battery and the system should shutdown - correctly?
Try to set the `pollonly` flag in `ups.conf` so it would not rely on interrupts for data transfer.

desertwitch, 2025-05-23:
> Did it determine that the UPS is on battery and the system should shutdown - correctly? Try to set the `pollonly` flag in `ups.conf` so it would not rely on interrupts for data transfer.

I think the `OB` was actually wrong if I didn't misread his above posts, and possibly a result of the absence of anything power:
```
May 22 22:00:22 SRVxxx rc.nut:    0.239226	[D4:2727135] HIDGetItemData: FindObject_with_Path() failed to locate 'UPS.Output.ConfigActivePower'
May 22 22:00:22 SRVxxx rc.nut:    0.239231	[D5:2727135] hid_lookup_usage: UPS -> 00840004
May 22 22:00:22 SRVxxx rc.nut:    0.239237	[D5:2727135] hid_lookup_usage: PowerSummary -> 00840024
May 22 22:00:22 SRVxxx rc.nut:    0.239241	[D5:2727135] hid_lookup_usage: PresentStatus -> 00840002
May 22 22:00:22 SRVxxx rc.nut:    0.239249	[D5:2727135] hid_lookup_usage: ACPresent -> 008500d0
May 22 22:00:22 SRVxxx rc.nut:    0.239253	[D4:2727135] string_to_path: depth = 4
May 22 22:00:22 SRVxxx rc.nut:    0.239257	[D4:2727135] HIDGetItemData: FindObject_with_Path() failed to locate 'UPS.PowerSummary.PresentStatus.ACPresent'
May 22 22:00:22 SRVxxx rc.nut:    0.239262	[D5:2727135] hid_lookup_usage: UPS -> 00840004
May 22 22:00:22 SRVxxx rc.nut:    0.239268	[D5:2727135] hid_lookup_usage: PowerSummary -> 00840024
May 22 22:00:22 SRVxxx rc.nut:    0.239273	[D5:2727135] hid_lookup_usage: PresentStatus -> 00840002
May 22 22:00:22 SRVxxx rc.nut:    0.239280	[D5:2727135] hid_lookup_usage: Charging -> 00850044
```

Angryhellfire, 2025-05-24:
> Did it determine that the UPS is on battery and the system should shutdown - correctly?
> Try to set the `pollonly` flag in `ups.conf` so it would not rely on interrupts for data transfer.

The UPS is connected to power. However, since NUT cannot read any values ​​(neither power, remaining runtime, nor status), NUT initiates a shutdown.

**Log with excerpt "pollonly" enabled:**
> May 22 22:00:22 SRVxxx rc.nut:    0.236432	[D2:2727135] Checking device 11 of 12 (2E66/0102)
May 22 22:00:22 SRVxxx rc.nut:    0.237555	[D2:2727135] - VendorID: 2e66
May 22 22:00:22 SRVxxx rc.nut:    0.237585	[D2:2727135] - ProductID: 0102
May 22 22:00:22 SRVxxx rc.nut:    0.237590	[D2:2727135] - Manufacturer: Salicru
May 22 22:00:22 SRVxxx rc.nut:    0.237593	[D2:2727135] - Product: TWIN700-RT
May 22 22:00:22 SRVxxx rc.nut:    0.237596	[D2:2727135] - Serial Number: 0000000000000000
May 22 22:00:22 SRVxxx rc.nut:    0.237599	[D2:2727135] - Bus: 002
May 22 22:00:22 SRVxxx rc.nut:    0.237601	[D2:2727135] - Bus Port: 002
May 22 22:00:22 SRVxxx rc.nut:    0.237604	[D2:2727135] - Device: 002
May 22 22:00:22 SRVxxx rc.nut:    0.237607	[D2:2727135] - Device release number: 0002
May 22 22:00:22 SRVxxx rc.nut:    0.237610	[D2:2727135] Trying to match device
May 22 22:00:22 SRVxxx rc.nut:    0.237614	[D2:2727135] match_function_subdriver_name: matching a subdriver by explicit name/regex: 'salicru'...
May 22 22:00:22 SRVxxx rc.nut:    0.237618	[D2:2727135] match_function_subdriver_name: retry matching by regex 'as is'
May 22 22:00:22 SRVxxx rc.nut:    0.237669	[D2:2727135] match_function_subdriver_name: retry matching by regex with added '.*'
May 22 22:00:22 SRVxxx rc.nut:    0.237737	[D2:2727135] match_function_subdriver_name: found a match: Salicru HID 0.4
May 22 22:00:22 SRVxxx rc.nut:    0.237749	[D3:2727135] match_function_regex: matching a device...
May 22 22:00:22 SRVxxx rc.nut:    0.237788	[D2:2727135] Device matches
May 22 22:00:22 SRVxxx rc.nut:    0.237796	[D2:2727135] Reading configuration descriptor 1 of 1
May 22 22:00:22 SRVxxx rc.nut:    0.237814	[D3:2727135] libusb_kernel_driver_active() returned 0: Success
May 22 22:00:22 SRVxxx rc.nut:    0.237850	[D2:2727135] Claimed interface 0 successfully
May 22 22:00:22 SRVxxx rc.nut:    0.237860	[D3:2727135] nut_libusb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
May 22 22:00:22 SRVxxx rc.nut:    0.238043	[D2:2727135] Retrieved HID descriptor (expected 9, got 9)
May 22 22:00:22 SRVxxx rc.nut:    0.238052	[D3:2727135] HID descriptor, method 1: (9 bytes) => 09 21 10 01 19 01 22 1b 00
May 22 22:00:22 SRVxxx rc.nut:    0.238055	[D3:2727135] HID descriptor length (method 1) 27
May 22 22:00:22 SRVxxx rc.nut:    0.238059	[D4:2727135] i=0, extra[i]=09, extra[i+1]=21
May 22 22:00:22 SRVxxx rc.nut:    0.238063	[D3:2727135] HID descriptor, method 2: (9 bytes) => 09 21 10 01 19 01 22 1b 00
May 22 22:00:22 SRVxxx rc.nut:    0.238069	[D3:2727135] HID descriptor length (method 2) 27
May 22 22:00:22 SRVxxx rc.nut:    0.238072	[D2:2727135] HID descriptor length 27
May 22 22:00:22 SRVxxx rc.nut:    0.238312	[D2:2727135] Report Descriptor size = 27
May 22 22:00:22 SRVxxx rc.nut:    0.238320	[D3:2727135] Report Descriptor: (27 bytes) => 06 00 ff 09 01 a1 01 09 02 15 00 26 ff 00
May 22 22:00:22 SRVxxx rc.nut:    0.238326	[D3:2727135]  75 08 95 08 81 82 09 02 95 08 91 82 c0
May 22 22:00:22 SRVxxx rc.nut:    0.238430	[D2:2727135] match_function_subdriver_name: matching a subdriver by explicit name/regex: 'salicru'...
May 22 22:00:22 SRVxxx rc.nut:    0.238437	[D2:2727135] match_function_subdriver_name: retry matching by regex 'as is'
May 22 22:00:22 SRVxxx rc.nut:    0.238457	[D2:2727135] match_function_subdriver_name: retry matching by regex with added '.*'
May 22 22:00:22 SRVxxx rc.nut:    0.238497	[D2:2727135] match_function_subdriver_name: found a match: Salicru HID 0.4
May 22 22:00:22 SRVxxx rc.nut:    0.238504	Using subdriver: Salicru HID 0.4
May 22 22:00:22 SRVxxx rc.nut:    0.238509	[D1:2727135] 2 HID objects found
May 22 22:00:22 SRVxxx rc.nut:    0.238516	[D4:2727135] Entering libusb_get_report
May 22 22:00:22 SRVxxx rc.nut:    0.238643	nut_libusb_get_report: Success
May 22 22:00:22 SRVxxx rc.nut:    0.238654	[D1:2727135] Can't retrieve Report 00: Success
May 22 22:00:22 SRVxxx rc.nut:    0.238658	[D5:2727135] hid_lookup_path: ff000001 -> not found in lookup table
May 22 22:00:22 SRVxxx rc.nut:    0.238662	[D5:2727135] hid_lookup_path: ff000002 -> not found in lookup table
May 22 22:00:22 SRVxxx rc.nut:    0.238666	[D1:2727135] Path: ff000001.ff000002, Type: Input, ReportID: 0x00, Offset: 0, Size: 8
May 22 22:00:22 SRVxxx rc.nut:    0.238669	[D4:2727135] Entering libusb_get_report
May 22 22:00:22 SRVxxx rc.nut:    0.238794	nut_libusb_get_report: Success
May 22 22:00:22 SRVxxx rc.nut:    0.238802	[D1:2727135] Can't retrieve Report 00: Success
May 22 22:00:22 SRVxxx rc.nut:    0.238806	[D5:2727135] hid_lookup_path: ff000001 -> not found in lookup table
May 22 22:00:22 SRVxxx rc.nut:    0.238809	[D5:2727135] hid_lookup_path: ff000002 -> not found in lookup table
May 22 22:00:22 SRVxxx rc.nut:    0.238812	[D1:2727135] Path: ff000001.ff000002, Type: Output, ReportID: 0x00, Offset: 0, Size: 8
May 22 22:00:22 SRVxxx rc.nut:    0.238828	[D5:2727135] send_to_all: SETINFO ups.mfr "Salicru"
May 22 22:00:22 SRVxxx rc.nut:    0.238832	[D5:2727135] send_to_all: SETINFO ups.model "TWIN700-RT"
[... cut ...]
May 22 22:00:23 SRVxxx rc.nut:    0.990863	[D5:2727135] send_to_all: SETINFO driver.parameter.pollinterval "2"
May 22 22:00:23 SRVxxx rc.nut:    0.990875	[D5:2727135] send_to_all: SETINFO driver.parameter.synchronous "auto"
May 22 22:00:23 SRVxxx rc.nut:    0.990887	[D5:2727135] send_to_all: SETINFO device.mfr "Salicru"
May 22 22:00:23 SRVxxx rc.nut:    0.990899	[D5:2727135] send_to_all: SETINFO device.model "TWIN700-RT"
May 22 22:00:23 SRVxxx rc.nut:    0.990911	[D5:2727135] send_to_all: SETINFO device.serial "0000000000000000"
May 22 22:00:23 SRVxxx usbhid-ups[2727162]: Startup successful

Angryhellfire, 2025-05-26:
Old NUT installed and unraid rebooted.

So I tried with device path and salicru sub-driver:

> May 26 17:54:36 FGI-SRV01 rc.nut: Network UPS Tools - UPS driver controller 2.7.4.1
May 26 17:54:46 FGI-SRV01 emhttpd: cmd: /usr/local/emhttp/plugins/nut-dw/scripts/nutscan_soft
May 26 17:55:17 FGI-SRV01 emhttpd: cmd: /usr/local/emhttp/plugins/nut-dw/scripts/nutscan_soft
May 26 17:55:29 FGI-SRV01 ool www[19657]: /usr/local/emhttp/plugins/nut-dw/scripts/start
May 26 17:55:30 FGI-SRV01 rc.nut: Writing NUT configuration...
May 26 17:55:32 FGI-SRV01 rc.nut: Updating permissions for NUT...
May 26 17:55:32 FGI-SRV01 rc.nut: Checking if the NUT Runtime Statistics Module should be enabled...
May 26 17:55:32 FGI-SRV01 rc.nut: Disabling the NUT Runtime Statistics Module...
May 26 17:55:33 FGI-SRV01 rc.nut: Can't start /usr/libexec/nut/**salicru-hid: No such file or directory**
May 26 17:55:33 FGI-SRV01 rc.nut: Network UPS Tools - UPS driver controller 2.7.4.1

Device path but no sub-driver:

> May 26 17:55:33 FGI-SRV01 rc.nut: Network UPS Tools - UPS driver controller 2.7.4.1
May 26 17:56:17 FGI-SRV01 emhttpd: cmd: /usr/local/emhttp/plugins/nut-dw/scripts/nutscan_soft
May 26 17:56:26 FGI-SRV01 ool www[24146]: /usr/local/emhttp/plugins/nut-dw/scripts/stop
May 26 17:56:27 FGI-SRV01 rc.nut: Writing NUT configuration...
May 26 17:56:31 FGI-SRV01 rc.nut: Updating permissions for NUT...
May 26 17:56:31 FGI-SRV01 rc.nut: Checking if the NUT Runtime Statistics Module should be enabled...
May 26 17:56:31 FGI-SRV01 rc.nut: Disabling the NUT Runtime Statistics Module...
May 26 17:56:32 FGI-SRV01 rc.nut: Stopping the NUT services... 
May 26 17:56:35 FGI-SRV01 rc.nut: Can't open /var/run/nut/**usbhid-ups-ups.pid: No such file or directory**
May 26 17:56:35 FGI-SRV01 rc.nut: Can't open /var/run/nut/**usbhid-ups-002.pid either: No such file or directory**

usbhid withoud sub-driver or device path (auto):

> May 26 17:56:45 SRVxxx ool www[24146]: /usr/local/emhttp/plugins/nut-dw/scripts/start
May 26 17:56:46 SRVxxx rc.nut: Writing NUT configuration...
May 26 17:56:48 SRVxxx rc.nut: Updating permissions for NUT...
May 26 17:56:48 SRVxxx rc.nut: Checking if the NUT Runtime Statistics Module should be enabled...
May 26 17:56:48 SRVxxx rc.nut: Disabling the NUT Runtime Statistics Module...
May 26 17:56:49 SRVxxx rc.nut: Network UPS Tools - Generic HID driver 0.55 (2.7.4.1)
May 26 17:56:49 SRVxxx rc.nut: USB communication driver 0.43
May 26 17:56:49 SRVxxx rc.nut: **No matching HID UPS found**
May 26 17:56:49 SRVxxx rc.nut: Driver failed to start (exit status=1)
May 26 17:56:49 SRVxxx rc.nut: Network UPS Tools - UPS driver controller 2.7.4.1

jimklimov, 2025-05-26:
> May 26 17:55:33 SRVxxx rc.nut: Can't start /usr/libexec/nut/salicru-hid: No such file or directory

This looks like you specified `driver=salicru-hid` instead of `driver=usbhid-ups`? (Note a `subdriver` option would be at best ignored, did not exist in v2.7.4)

During a NUT program start, a "Can't open ....pid: No such file or directory" may be benign - if there is no older copy running. (Or it may highlight a fault, if the older copy was expected to run but is not, or if something is in fact running but could/did not leave a PID file - so the message is always issued; in new releases it is worded hopefully better).