networkupstools/nut #2968: Documentation link for Augeas is wrong

kstuddard, 2025-05-23:
In the below text/page, "Augeas" is a hyperlink to a dead page
"""
https://networkupstools.org/docs/user-manual.chunked/Configuration_notes.html
Note
NUT does not currently provide proper graphical configuration tools. However, there is now support for Augeas, which will enable the easier creation of configuration tools.
"""

The current link is to https://networkupstools.org/docs/developer-guide.chunked/index.html#augeas_user but I think it is meant to point to https://networkupstools.org/docs/developer-guide.chunked/augeas.html

jimklimov, 2025-05-23:
Thanks! The pages are generated, and there was recently a re-arrangement of some source texts into a new book, so maybe some asciidoc anchor got lost :\

jimklimov, 2025-05-23:
Hm, historic site builds led to e.g. https://networkupstools.org/historic/v2.8.1/docs/developer-guide.chunked/index.html without even the anchor. So it was broken before the recent change.

jimklimov, 2025-05-23:
Thanks, I think I've got it :)

PR #2971 (2025-05-27): Various documentation updates
* New NUT Overview page did not get published on nut-website
* Improve upsmon docs about root/unprivileged nuances
* Fix man page cross-links (external tools, NUT configs)

Closes: #2968
Closes: #2970
Follows-up from: #2950 and #2945