networkupstools/nut #2989: macos 2.8.3 build error

daeho-ro, 2025-06-15:
Hello, this is a homebrew maintainer and we got some error during build for v2.8.3.

```
  make[3]: Nothing to be done for `install-data-am'.
  sed: RE error: illegal byte sequence
  make[2]: *** [/private/tmp/nut-20250615-6006-2b19w8/nut-2.8.3/docs/.prep-src-docs] Error 1
  make[1]: *** [install-recursive] Error 1
  make: *** [install-recursive] Error 1
```
- https://github.com/Homebrew/homebrew-core/pull/218729

Not only this but there are many invalid option for some commands.
Can you take a look this?

Thanks,

jimklimov, 2025-06-15:
Thanks for the report. And great to see package maintainers engage in general :)

In cloudy NUT CI farm agents we have, I think, Monterey and whatever CircleCI has on tap that day.

I wonder what exactly it complains about, and with which tools (e.g. sed - gnu, bsd, sun, ...?)

A verbose sequential make log can help: `make -v 1 V=1 -k` ... Possibly after another `make -ks -j 8` that would build whatever works, to minimize the error report, but this may dilute the error visibility (if a recipe is misinterpreted and some goal relies on a file not made before it, a re-run might jist pick it up as present at that time).

jimklimov, 2025-06-15:
Also check GitHub release notes, there were a few errata fixed after 2.8.3 - should be small patches to backport for non-rolling packages.

kevinruddy, 2025-06-16:
I'm not skilled enough to find what change caused the issue. But I configured and tried to build the current version on macOS Sequoia. It fails in the `for F in $(PREP_SRC)` loop of the `prep-src-docs` section of `docs/Makefile`, on this line:
```
sed \
    -e 's,\(home page:\) https://www.networkupstools.org/*$$,\1 $(NUT_WEBSITE_BASE)/,' ; \
```
It dumps out when it encounters a file in `PREP_SRC` that contains a slash (`images/asciidoc.png`). I'm not sure why that triggers it since the replacement seems to be using comma as a delimiter. But it loops through the first 50 pathnames (no slashes) without a problem, and if I remove all the pathnames that include subdirectories it completes successfully.

jimklimov, 2025-06-16:
If it really does, must be some typo. I think those should only be text sources :\

Will double-check (later)...

kevinruddy, 2025-06-16:
I think the change that triggered this was [c1ae32c](https://github.com/networkupstools/nut/commit/c1ae32cb24c36c48846d11464b09e389488acc10). Code was added that has `sed` reading the contents of the files and outputting to -prepped suffixed files. Prior to that I think it was only working on the pathnames, so the fact that binary files were in the list didn't affect anything.

jimklimov, 2025-06-19:
Cheers, can you please check if #2990 nails this?

jimklimov, 2025-06-20:
Great, thanks for checking and reporting (in the first place, and now)! :)