networkupstools/nut #2991: [HCL] Makelsan, Lion 850VA Line Interactive supported by nutdrv_qx

giraypultar, 2025-06-20:
root@lahana:/home/giray# upsc lahanaups
Init SSL without certificate database
battery.voltage: 13.40
device.type: ups
driver.name: nutdrv_qx
driver.parameter.bus: 005
driver.parameter.langid_fix: 0x409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.version: 2.8.0
driver.version.data: Q1 0.07
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.3
input.voltage: 221.4
input.voltage.fault: 222.0
output.voltage: 221.4
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 5161
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
ups.vendorid: 0665

root@lahana:/home/giray# upsrw -l  lahanaups
[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: RANGE NUMBER
Option: "12-600" SELECTED

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: RANGE NUMBER
Option: "0-599940" SELECTED

root@lahana:/home/giray# upscmd -l  lahanaups
Instant commands supported on UPS [lahanaups]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test

shutdown appears to work ok.

https://www.makelsan.com.tr/en/ups/urundetay/lion-serisi-650---2200-va/index-500.html

UPS works without langid_fix as well

giraypultar, 2025-06-20:
From two different versions of the same model ups

Host 1
----------------
# from /etc/nut/ups.conf
[lahanaups]
	driver = "nutdrv_qx"
	port = "auto"
	vendorid = "0665"
	productid = "5161"
	# poduct = "USB to Serial"
	# serial = "20100826"
	# vendor = "INNO TECH"
	bus = "005"
 	langid_fix = "0x409" # not necessary

root@lahana:/#  lsusb|grep Serial
Bus 005 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial

Host 2
-----------------
# from /etc/nut/ups.conf
[patatesups]
	driver = "nutdrv_qx"
	port = "auto"
	vendorid = "0001"
	productid = "0000"
	# 	product = "MEC0003"
	# 	vendor = "MEC"
	bus = "002"
	langid_fix = "0x409" # not necessary 

root@patates:/home/giray# lsusb|grep MEC
Bus 002 Device 003: ID 0001:0000 Fry's Electronics MEC0003

jimklimov, 2025-07-03:
Thanks for posting!