networkupstools/nut #2994: CyberPower UPS w/ RMCARD205 not reporting UPS Status

sdjernes, 2025-06-22:
Hello,

I am connecting my CP1500PFCLCDa with a RMCARD205 in it to NUT and via that to Peanut and Home Assistant.

In the process of setting it up I had to do some troubleshooting.  It was not showing an Online status in either Home Assistant or Peanut.

Below is the output of upsc cyber-lvroom.

battery.charge: 100
battery.current: 0.40
battery.runtime: 11400
battery.runtime.elapsed: 0
battery.voltage: 27.40
battery.voltage.nominal: 24
device.contact: Shawn
device.description: UPS SNMP Card
device.location: Living Room TV Stand
device.mfr: CYBERPOWER
device.model: CP1500PFCLCDa
device.serial: CXXLP7002521
device.type: ups
driver.name: snmp-ups
driver.parameter.mibs: cyberpower
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.1.97
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: cyberpower MIB 0.51
driver.version.internal: 1.21
input.frequency: 60
input.voltage: 118.60
output.current: 0.40
output.frequency: 60
output.voltage: 118.60
ups.delay.reboot: 0
ups.firmware: CR01802CBH11
ups.load: 3
ups.mfr: CYBERPOWER
ups.model: CP1500PFCLCDa
ups.serial: CXXLP7002521
ups.status: 

Below is my section from ups.conf:
[cyber-lvroom]
        driver = "snmp-ups"
        port = "192.168.1.97"
        desc = "CP1500PFCLCDa"
        mibs = "cyberpower"
        community = "public"

The RMCARD205 is running 1.4.1 of the firmware.

I have a CP1500AVRLCD3 that does not have a network card yet and is using usb-hid and reports status fine.

If anyone could point me to the proper documentation that would help.

Thank You,
Shawn

sdjernes, 2025-07-12:
> Regarding NUT version
>
>
> https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests
> is always an option ;)
>
> SNMP traps
>
> At the moment no NUT driver actively listens on the network, they only
> initiate connections. Comments in sources indicate that SNMP trap support
> was considered as a future idea, but for over a decade nobody stepped up to
> introduce the code - so we rely on regular polling instead.
>
> As a side note, I think a similar limitation concerns netxml-ups (Eaton
> NMCs can also message subscribed clients); though maybe
> @arnaudquette-eaton <https://github.com/arnaudquette-eaton> would
> disprove me on this statement.
>
> Something similar also applies to USB HID drivers, where in some cases
> "interrupt" mode works and in others it is toxic and breaks things - so
> there are options to require it or conversely limit to "pollonly" mode.
>
> snmpwalk
>
> Try the man page for the version you have installed. Generally speaking,
> https://linux.die.net/man/1/snmpwalk and
> https://linux.die.net/man/1/snmpcmd may apply - you need to pass a
> SNMPv1/2 community string or SNMPv3 login (same credentials as you have in
> ups.conf), possibly protocol version, the host/IP address, and the OID to
> query, e.g.
>
> :; snmpwalk -c public -v1 192.168.1.97 ".1.3.6.1.4.1.3808.1.1.1.4.1.1.0"
>
> (repeat for other OID strings in that table excerpt)

.1.3.6.1.4.1.3808.1.1.1.4.1.1.0
Returns
iso.3.6.1.4.1.3808.1.1.1.4.1.1.0 = INTEGER: 2

.1.3.6.1.4.1.3808.1.1.1.2.1.1.0
Returns
iso.3.6.1.4.1.3808.1.1.1.2.1.1.0 = INTEGER: 2

.1.3.6.1.4.1.3808.1.1.1.7.2.7.0
Returns
iso.3.6.1.4.1.3808.1.1.1.7.2.7.0 = INTEGER: 1

.1.3.6.1.4.1.3808.1.1.1.2.2.5.0"Returns
iso.3.6.1.4.1.3808.1.1.1.2.2.5.0 = INTEGER: 1

I am working on building a Debian 12 box to try to build a newer
version of NUT as I have several UPSes.  I have found that both of my
CP1500 w/ the RMCARD205 on firmware 1.4.1 or 1.4.4 are broken in the
same way.  I am waiting for an outlet to be installed before I can
plug in my OR3000 with a 30amp plug.  It has an RMCARD205 with much
older firmware.

sdjernes, 2025-07-13:
Jim,

The problem seams to have been resolved in the HEAD branch of the NUT from
Github.  I built it on a test Debian WSL VM and it shows "OL" in the
ups.status field when I use "upsc".  Now I just have to figure out how to
build a .deb package so I can install it on my ProxMox server.  I am very
rusty at building .deb packages so it may take me some time to figure it
out.

Can an upsmon from 2.8.0 talk to an upsd from 2.8.3?  I may take an old
Raspberry PI 2b and make it the UPSD server for now.

Thanks,

jimklimov, 2025-07-14:
Great to hear! As for mixing two versions - should work for network protocol, post a bug please if there are any discrepancies.