networkupstools/nut #2996: Port to ESP32

banoz, 2025-06-24:
I would like to ask for suggestions on porting to ESP32 MCU. Here's my [rough attempt](
https://github.com/banoz/nut/tree/esp32-alpha) on PoC that at least retrieves UPS model/SN but it turned out quite butchered. I won't pretend I know a lot about either NUT or ESP32, there might be very dumb mistakes in the code. At first I'd try to avoid moving the driver and server to src. Also there's still some issue in HID driver that prevents retrieving actual running values.

jimklimov, 2025-06-24:
At face value, NUT code itself tries to maintain compatibility with C99 and to some extent (few warnings) with C89/ANSI C, and most of it is in plain C (few recent optional parts in C++, UI in python, etc.)

Much of the interface with the world however uses respective third-party libraries (whether platform-specific implementations of libc, or stuff like libusb, etc.) or tools (e.g, ability to spell-check docs or render man pages) - so is subject to their availability and functionality there.

banoz, 2025-06-24:
ESP32-S3 that I used is the Xtensa arch MCU and ESP-IDF offers POSIX compatible API so most things just work. The main problems I faced were a missing libusb port and lack of support of AF_UNIX in lwip. I tried to switch from libusb to a native ESP32 stack that's built on tinyusb. For AF_UNIX I just rerouted the socket to use loopback. Both approaches work somehow but there's still something preventing the live data flow. I knew little to nothing about both NUT and ESP-IDF when I got the idea of porting this, and still feel like this all is way over my head :)

jimklimov, 2025-06-24:
FYI : Now that NUT supports two generations of libusb (0.1 and 1.0), I suppose you might try adding tinyusb as a backend in a similar fashion - as a starting point, see `drivers/libusb{0,1}.c` and `usb-common` sources for inspiration.

banoz, 2025-06-24:
I'd also appreciate if anyone could take a look at the [log](https://gist.github.com/banoz/0442a168f53437ea34286fc99ae165ab) to see if any obvious issue pops out

banoz, 2026-09-23:
[esp32-beta-0.1.0](https://github.com/banoz/nut/releases/tag/esp32-beta-0.1.0)