networkupstools/nut #3000: Wrong protocol for Powercom RAPTOR series

aleka204, 2025-07-02:
I have RAPTOR RPT-1500AP LCD on my linux server and default powercom driver doesn't work with it. I've done a lot of experiments for a several weeks, and i can say that RPT series have the same usb protocol like Smart KING Pro series. In this protocol, we can't set arbitrary value of shutdown delay in seconds like in other powercom upses, shutdown delay can be set only from table "Table of possible delays for Shutdown commands" specified at page 17: https://networkupstools.org/protocols/powercom/Software_USB_communication_controller_SKP_series.doc.
I checked this by changing the powercom-hid.c driver. I changed function "powercom_shutdown_nuf" so it set "command" variable value from func input(fast bypass :D). After changing, i compiled it and execute `./upscmd -u username -p password powercom1 shutdown.return 8` and ups start beeping and turned off exactly after 54 seconds:
```
static double powercom_shutdown_nuf(const char *value)
{
	const char	*s = dstate_getinfo("ups.delay.shutdown");
	uint8_t	val, command;
	int iv;

	iv = atoi(value ? value : s);
	if (iv < 0 || (intmax_t)iv > (intmax_t)UINT8_MAX) {
		upsdebugx(0, "%s: value = %d is not in uint16_t range", __func__, iv);
		return 0;
	}

	command = (uint8_t)iv;

	// val = (uint16_t)iv;
	// val = val ? val : 1;    /* 0 sets the maximum delay */
	// if (powercom_sdcmd_byte_order_fallback) {
	// 	/* Legacy behavior */
	// 	command = ((uint16_t)((val % 60) <<  8)) + (uint16_t)(val / 60);
	// 	command |= 0x4000;	/* AC RESTART NORMAL ENABLE */
	// } else {
	// 	/* New default */
	// 	command = ((uint16_t)((val / 60) << 8)) + (uint16_t)(val % 60);
	// 	command |= 0x0040;	/* AC RESTART NORMAL ENABLE */
	// }

	upsdebugx(3, "%s: value = %s, command = %02X", __func__, value, command);

	return command;
}
```
The same issue with "powercom_stayoff_nuf" function:
```
static double powercom_stayoff_nuf(const char *value)
{
	const char	*s = dstate_getinfo("ups.delay.shutdown");
	uint16_t	val, command;
	int iv;

	iv = atoi(value ? value : s);
	if (iv < 0 || (intmax_t)iv > (intmax_t)UINT16_MAX) {
		upsdebugx(0, "%s: value = %d is not in uint16_t range", __func__, iv);
		return 0;
	}

	val = (uint16_t)iv;
	val = val ? val : 1;    /* 0 sets the maximum delay */
	if (powercom_sdcmd_byte_order_fallback) {
		/* Legacy behavior */
		command = ((uint16_t)((val % 60) << 8)) + (uint16_t)(val / 60);
		command |= 0x8000;	/* AC RESTART NORMAL DISABLE */
	} else {
		/* New default */
		command = ((uint16_t)((val / 60) << 8)) + (uint16_t)(val % 60);
		command |= 0x0080;	/* AC RESTART NORMAL DISABLE */
	}

	upsdebugx(3, "%s: value = %s, command = %04X", __func__, value, command);

	return command;
}
```
Also, the "powercom_startup_nuf" function doesn't work either. Worked version: 
```
static double powercom_startup_nuf(const char *value)
{
	const char	*s = dstate_getinfo("ups.delay.start");
	uint32_t	val, command;
	int iv;

	//iv = atoi(value ? value : s) / 60;
	iv = atoi(value ? value : s);
	if (iv < 0 || (intmax_t)iv > (intmax_t)UINT32_MAX) {
		upsdebugx(0, "%s: value = %d is not in uint16_t range", __func__, iv);
		return 0;
	}

	/* COMMENTME: What are we doing here, a byte-swap in the word? */
	// val = (uint16_t)iv;
	// command =  (uint16_t)(val << 8);
	// command += (uint16_t)(val >> 8);
	command = (uint32_t)iv;
	upsdebugx(3, "%s: value = %s, command = %04X", __func__, value, command);

	return command;
}
```
After compile, if i execute `./upscmd -u username -p password powercom1 load.on.delay X` - in case of powered off ups and the presence of the input voltage, it will turn on after 1minute + X seconds (Idk why, but it`s work).

I found this bugs 2 month ago, but i haven't time to create issue, so maybe I missed some details.
lsusb log: 

[powercom_usbhid.log](https://github.com/user-attachments/files/21018167/powercom_usbhid.log)

jimklimov, 2025-07-08:
@aleka204 : just in case, do you have a chance to test this on systems with different CPU endianness (e.g. an ARM like Raspberry PI)? Looking at proposed changes again, I am now also worried that the older code did some byte-swapping tricks (I suppose so that bytes on the USB wire are in certain order expected by UPS firmware, regardless of maths in PC/server CPU).

I'll post a PR shortly that would try to mix your proposal with older code, as an option, so that there would be least surprise for existing users by default (until/unless we prove the old driver was broken for all Powercom USB HID models).

jimklimov, 2025-07-08:
It looks like your changes trivialized the older conversions from seconds in dstate values, so effectively you pass the numbers received as is, just cast to a (different) int type; in case of `uint8_t` it might also truncate the range (e.g. hard to request 10 minutes with a 255-second value). And then expanded to double, as the method return type.

The table you linked to at first suggests that for this protocol there are 18 index values meaning certain timings, so I'd expect that the fix would determine the requested seconds from strings, and return the nearest index value, or something similar. Probably same is needed for `*_fun` methods to report one of those index values as a specific amount of seconds (as a string).

In your tests after the "fix", did you specify `X` timing as seconds or as one of those index numbers? If the latter, the experiments confirm that such numbers should be passed around for these models, but more work is needed before the change is acceptable into NUT code base :)

* [ ] Map protocol timing table index to seconds *and back* (from/to numeric/string values)
* [ ] Very much not sure the change of int types was needed at all
* [ ] Add a user-configurable toggle to enable this discrete logic or keep doing whatever was done before (for existing deployments, maybe older models - assuming some did fit the protocol dialect/variant as coded in NUT now)

jimklimov, 2025-07-08:
Ultimately, mapping the time back from the table seems not needed - the UPS apparently saves the time encoded by `DelayBeforeShutdown` into "value memorized in USB controller by `DelayBeforeStartup` command", which is odd since the former may specify seconds, and the latter is in whole(?) minutes. And they are different delays for different logical purposes.

Reading from `DelayBeforeShutdown` should yield shutdown status encoded across different bits for various situations, not delays :-\

@aleka204 : can you please check if https://github.com/jimklimov/nut/tree/issue-3000 (PR #3009) addresses this issue reasonably? At least as far as setting the values from seconds, and sending the commands? Reading back seems complicated, gotta run to dayjob at the moment.

aleka204, 2025-07-10:
@jimklimov Hi! Sorry for long answering.

> [@aleka204](https://github.com/aleka204) : just in case, do you have a chance to test this on systems with different CPU endianness (e.g. an ARM like Raspberry PI)

No i haven`t systems with ARM cpu, only x86-64.

> > The same issue with "powercom_stayoff_nuf" function
> 
> The code you posted seems to have no changes vs. what is in git, did you pick the wrong version from the editor?

Indeed, I did not change powercom_stayoff_nuf, I forgot about it. There was a lot of attempts to get work version of driver so I'm a little confused.

> It looks like your changes trivialized the older conversions from seconds in dstate values, so effectively you pass the numbers received as is, just cast to a (different) int type; in case of [...]

I don't suggest working solution, it's just a temporary patch to highlight problem essence)

> In your tests after the "fix", did you specify `X` timing as seconds or as one of those index numbers?

I specifying X like "index" number or "Sent to UPS" number from table, forgot what exactly. Will check it after work.

>@aleka204 : can you please check if https://github.com/jimklimov/nut/tree/issue-3000 (PR https://github.com/networkupstools/nut/pull/3009) addresses this issue reasonably?

Of course! I will definitely try it one of these days.

jimklimov, 2025-07-24:
My sympathies... sure thing, quality of life goes first :)

aleka204, 2025-07-30:
> In your tests after the "fix", did you specify X timing as seconds or as one of those index numbers?

I checked what number from table i send to ups. It`s "Index", not "Sent to UPS".
I tryed to compile your fork with fixed powercom driver and I broke my linux usb subsystem :D. At august i will have't enough time to check it(and fix my server), but at september I guess come back.

jimklimov, 2025-07-30:
Wow, I hope there was nothing NUT did to break the OS setup (Linux USB subsystem)?

aleka204, 2025-07-30:
No, i think its not nut. I installed 2 packages for compile configure file, then compile nut and execute ./upsc powercom1 (it defined in config file). Upsc return: no usb found(or something like that). I rebooted linux and it stuck on boot: cant enumerate usb devices and some other usb errors. Next, watchdog timer triggered and linux autorebooted and starts successfully but with usb errors. I think it`s not a difficult problem but it will take time. If I will find time on these week, I could resolve this and test your fork, if not, only at September;)

aleka204, 2025-07-30:
Sorry, missclicked

closed 2025-11-06 by jimklimov