networkupstools/nut #3011: snmp driver not setting the write date for apc ups

sirmeili, 2025-07-09:
Hopefully this is the write place to put this. I have 2 APC s20BLK UPSes that I have monitored via SNMP via NUT. I ran a self test and noticed that NUT had the wrong date for the self date. It's not even consistent:

configs:

```
[houseRackNetwork]
driver = snmp-ups
mibs = apcc
port = 10.0.0.27
community = public
snmp_version = v1
pollfreq = 15
desc = "APC S20blk (Network)"

[houseRackAV]
driver = snmp-ups
mibs = apcc
port = 10.0.0.26
community = public
snmp_version = v1
pollfreq = 15
desc = "APC S20blk (AV)"
```

when  I run upsc houseRackNetwork I get 5/7/2025 for the ups.test.date.  I checked the UPS, it says 7/9/2025 (I justran it today). I used an snmp tester and it shows 7/9/2025. I also ran /lib/nut/snmp-ups -a houseRackNetwork -DD and it showed the following for that OID:

```
 2.286434     [D1] snmp_ups_walk: processing unitary device (1)
   2.287517     [D1] get_and_process_data: ups.test.date (.1.3.6.1.4.1.318.1.1.1.7.2.4.0)
   2.287808     [D2] su_ups_get: ups.test.date .1.3.6.1.4.1.318.1.1.1.7.2.4.0
   2.288085     [D2] su_ups_get: requesting nut_snmp_get_str(), without daisy template originally
   2.310172     [D1] su_find_strval: no result value for this OID string value (07/09/2025)
   2.311342     [D1] entering su_setinfo(ups.test.date, 07/09/2025)
   2.312834     [D1] su_setinfo: using info_type 'ups.test.date'
   2.313267     [D2] => value: 07/09/2025
```

The same is for my AV one, but oddly enough the date shows 5/9/2025 when the real date is 7/8/2025. They aren't even off the same amount.

So long story short, SNMP is sending out the right date, NUT seems to get the right date, but when I ask for the values it returns the wrong date?

jimklimov, 2025-07-09:
Thanks for the report. Just in case - which NUT version is this? By log format I can see 2.8.x, but not more specific.

Also, how long is that driver instance running? Looking at code, the current mapping in `drivers/apc-mib.c` is:
```
snmp_info_default("ups.test.date", ST_FLAG_STRING | ST_FLAG_RW, 8, ".1.3.6.1.4.1.318.1.1.1.7.2.4.0", "", SU_FLAG_OK | SU_FLAG_STATIC, NULL),
```

The `SU_FLAG_STATIC` part concerns me, probably should be semi-static. My guess is that the driver has successfully read the value once in its life-time, and did not try since.

It also seems to skip such entries quietly, so hard to confirm or deny the guess by looking at logs alone:
```
                        /* skip static elements in update mode */
                        if ((mode == SU_WALKMODE_UPDATE) && (su_info_p->flags & SU_FLAG_STATIC))
                                continue;
```

sirmeili, 2025-07-10:
So the version I have installed is: 2.8.0-7

and the driver overall has been there for over a year, but yesterday I updated the base linux install and rebooted the pi that it's running on. Not sure how static/semi-static plays into that.

Perhaps back in May I did run self tests, but I don't think so.

PR #3012 (2025-07-11): `snmp-ups`: debug when static entries are skipped; fix `ups.test.date` in `apc-mib` to be semi-static
Closes: #3011 

Wondering if `upsdebugx(1, ...)` verbosity level used to report activity on other entry flags would be too "loud" with skipped statics?..

@sirmeili: would you have a chance to test this fix? See e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about building custom NUT from sources (using this PR's source branch in my fork, if it is not merged by then). You would probably need to leave the driver runnin [... cut]