networkupstools/nut #3015: Basic functionality working for GoldenMate LIPO UPS - ddl-dump output attached.

jonpaterson, 2025-07-13:
Apologies if I have submitted this incorrectly.  let me know if you need something else.

# Please add if relevant: DEVICE:URL:REPORT: <GitHub issue/PR or Mailing list archive entry URL>
# Please add if relevant: DEVICE:URL:VENDOR: <Vendor/Manufacturer/Product page URL>

GoldenMate LIPO 1500VA/1000W
https://goldenmateenergy.com/products/goldenmate-back-ups-1500va-1000w-lifepo4-battery-backup-and-surge-protector

# Please add comments for humans here and perhaps raise concerns about the data points,
# see https://networkupstools.org/ddl/#devseq-files for comment-tag syntax
# strip sensitive data (passwords, SNMP community, serial number, IP address, host name...)
# and post as a pull request to https://github.com/networkupstools/nut-ddl/
# DEVICE:COMMENT:
# Device dump generated by ./nut-ddl-dump.sh on Sun, 13 Jul 2025 16:20:26 -0500
# DEVICE:EOC

# :; upsc myups
battery.charge: 100
battery.runtime: 0
battery.type: Lion
device.mfr: -BMS-
device.model: Smart-Battery
device.serial: J954E1A365FBE554041:23
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: iDowell HID 0.2
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.delay.shutdown: 20
ups.delay.start: 30
ups.mfr: -BMS-
ups.model: Smart-Battery
ups.productid: 0300
ups.serial: J954E1A365FBE554041:23
ups.status: OL
ups.timer.shutdown: 1
ups.timer.start: 257
ups.vendorid: 075d

# DEVICE:COMMENT-BLOCK:FIXME:UPSRW: myups
# [ups.delay.shutdown]
# Interval to wait after shutdown with delay command (seconds)
# Type: STRING
# Maximum length: 10
# Value: 20
#
# [ups.delay.start]
# Interval to wait before (re)starting the load (seconds)
# Type: STRING
# Maximum length: 10
# Value: 30
# DEVICE:EOC

# DEVICE:COMMENT-BLOCK:FIXME:UPSCMD: myups
# Instant commands supported on UPS [myups]:
#
# load.off - Turn off the load immediately
# load.off.delay - Turn off the load with a delay (seconds)
# load.on - Turn on the load immediately
# load.on.delay - Turn on the load with a delay (seconds)
# shutdown.return - Turn off the load and return when power is back
# shutdown.stayoff - Turn off the load and remain off
# shutdown.stop - Stop a shutdown in progress
# DEVICE:EOC

jimklimov, 2025-08-03:
Essentially yes - what the device says at all, and if it does so in a way the driver code (and particular sub-driver with data point mapping) understands.

Many data points on USB HID side have effectifely synonyms, so the device may be reporting one "path" and we're looking for another. In that case, it is a bit of trial and error, aided by high-verbosity debug in the NUT driver, to code up a next iteration of those mappings that would also understand this device.

If feeling adventurous, see wiki about custom-building NUT (anyhow would be newer than what you have from packages), and docs in sources about subdriver development.

sovanyio, 2025-08-05:
Here's the driver debug data from my 1000VA/600W version:

```
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
   0.000000	[D1] Built-in default or configured user for drivers 'root' was ignored due to 'root' specified on command line
   0.000240	[D3] main_arg: var='explore' val='<null>'
   0.000381	[D5] send_to_all: SETINFO driver.flag.explore "enabled"
   0.000515	[D3] main_arg: var='vendorid' val='075d'
   0.000589	[D5] send_to_all: SETINFO driver.parameter.vendorid "075d"
   0.000756	[D3] main_arg: var='port' val='auto'
   0.000839	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000875	[D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28784-155eea44e7) 13.3.0 and configured with flags: --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --sysconfdir=/etc/nut --datadir=/usr/share/nut --with-dev --with-usb --without-avahi --without-snmp --with-serial --without-doc --with-neon --without-powerman --without-wrap --with-hotplug-dir=/etc/hotplug --with-cgi --without-ipmi --without-freeipmi --without-linux-i2c --without-ssl --without-libltdl --without-macosx_ups --with-statepath=/var/run/nut --with-pidpath=/var/run --with-drvpath=/lib/nut --with-user=root --with-group=root --with-gd-includes='-I/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include/freetype2 -I/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include/libpng16' --with-gd-libs='-L/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/lib -lgd'
   0.000978	[D1] debug level is '5'
   0.001043	[D5] send_to_all: SETINFO driver.debug "5"
   0.001150	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001653	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001745	[D5] send_to_all: SETINFO device.type "ups"
   0.001810	[D5] send_to_all: SETINFO driver.state "init.device"
   0.001869	[D1] upsdrv_initups (non-SHUT)...
   0.001931	[D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.004187	[D2] Checking device 1 of 9 (075D/0300)
   0.006638	[D2] - VendorID: 075d
   0.006700	[D2] - ProductID: 0300
   0.006872	[D2] - Manufacturer: -BMS-
   0.007045	[D2] - Product: Smart-Battery
   0.007178	[D2] - Serial Number: XXXXXXXX
   0.007240	[D2] - Bus: 007
   0.007370	[D2] - Bus Port: 001
   0.007432	[D2] - Device: 002
   0.007599	[D2] - Device release number: 0001
   0.007660	[D2] Trying to match device
   0.007818	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.007943	[D3] match_function_regex: matching a device...
   0.008016	[D2] Device matches
   0.008149	[D2] Reading first configuration descriptor
   0.008220	[D3] libusb_kernel_driver_active() returned 0: Success
   0.008315	[D2] Claimed interface 0 successfully
   0.008377	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.008820	[D2] Retrieved HID descriptor (expected 9, got 9)
   0.008891	[D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 23 01
   0.008954	[D3] HID descriptor length (method 1) 291
   0.009013	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.009080	[D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 23 01
   0.009144	[D3] HID descriptor length (method 2) 291
   0.009203	[D2] HID descriptor length 291
   0.012375	[D2] Report Descriptor size = 291
   0.012452	[D3] Report Descriptor: (291 bytes) => 05 84 09 04 a1 01 09 24 a1 00 05 84 85 01
   0.012539	[D3]  09 25 09 1f 75 04 95 02 15 00 25 0f b1 03 09 fd 79 01 75 08 95 01 15 00 26
   0.012621	[D3]  ff 00 b1 03 09 fe 79 02 b1 23 09 ff 79 02 b1 23 05 85 09 89 b1 22 05 85 09
   0.012703	[D3]  d1 09 8b 25 01 75 01 95 02 b1 a2 95 06 b1 83 05 85 09 2c 75 08 25 ff 95 01
   0.012782	[D3]  b1 83 09 83 25 64 b1 82 09 67 b1 82 09 8c b1 83 09 66 b1 83 09 68 75 20 55
   0.012864	[D3]  00 15 00 27 ff ff ff 75 66 01 10 95 01 b1 83 05 84 09 40 67 21 d1 f0 00 55
   0.012948	[D3]  05 75 10 95 01 27 ff ff 00 00 b1 83 09 30 95 01 b1 83 05 84 09 02 a1 02 05
   0.013029	[D3]  85 09 d0 09 44 09 45 09 42 09 61 09 d1 09 69 65 00 55 00 25 01 75 01 95 07
   0.013111	[D3]  b1 83 95 01 b1 a3 c0 05 84 09 57 85 09 75 20 15 ff 27 ff ff ff 7f 66 01 10
   0.013234	[D3]  b1 82 09 56 85 0a b1 82 85 02 75 08 95 01 81 a3 05 85 09 66 15 00 25 64 75
   0.013315	[D3]  08 95 01 81 a3 09 68 66 01 10 55 00 75 20 95 01 15 00 27 ff ff ff fe 81 83
   0.013453	[D3]  05 84 09 30 67 21 d1 f0 00 55 05 75 10 95 01 81 82 85 06 75 08 95 20 81 83
   0.013517	[D3]  c0 c0
   0.013700	HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x01
   0.013845	HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -16777217 as 4278190079 in ReportID: 0x02
   0.014107	Using subdriver: EXPLORE HID 0.2
   0.014232	[D1] 28 HID objects found
   0.014293	[D4] Entering libusb_get_report
[... cut ...]
   0.041555	[D5] send_to_all: ADDCMD driver.reload-or-error
   0.041706	[D5] send_to_all: SETINFO driver.state "quiet"
   0.041728	[D1] Driver initialization completed, beginning regular infinite loop
   0.041752	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.041783	[D5] send_to_all: SETINFO driver.state "updateinfo"
   0.041801	[D1] upsdrv_updateinfo...
...
```

Not sure if the data beyond the first expected 9 bytes has anything interesting yet.

jonpaterson, 2025-08-05:
> Here's the driver debug data from my 1000VA/600W version:
> 
> ```
> Network UPS Tools - Generic HID driver 0.52 (2.8.1)
> USB communication driver (libusb 1.0) 0.46
>    0.000000	[D1] Built-in default or configured user for drivers 'root' was ignored due to 'root' specified on command line
>    0.000240	[D3] main_arg: var='explore' val='<null>'
>    0.000381	[D5] send_to_all: SETINFO driver.flag.explore "enabled"
>    0.000515	[D3] main_arg: var='vendorid' val='075d'
>    0.000589	[D5] send_to_all: SETINFO driver.parameter.vendorid "075d"
>    0.000756	[D3] main_arg: var='port' val='auto'
>    0.000839	[D5] send_to_all: SETINFO driver.parameter.port "auto"
>    0.000875	[D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28784-155eea44e7) 13.3.0 and configured with flags: --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --sysconfdir=/etc/nut --datadir=/usr/share/nut --with-dev --with-usb --without-avahi --without-snmp --with-serial --without-doc --with-neon --without-powerman --without-wrap --with-hotplug-dir=/etc/hotplug --with-cgi --without-ipmi --without-freeipmi --without-linux-i2c --without-ssl --without-libltdl --without-macosx_ups --with-statepath=/var/run/nut --with-pidpath=/var/run --with-drvpath=/lib/nut --with-user=root --with-group=root --with-gd-includes='-I/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include/freetype2 -I/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/include/libpng16' --with-gd-libs='-L/builder/shared-workdir/build/sdk/staging_dir/target-x86_64_musl/usr/lib -lgd'
>    0.000978	[D1] debug level is '5'
>    0.001043	[D5] send_to_all: SETINFO driver.debug "5"
>    0.001150	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
>    0.001653	[D1] Succeeded to become_user(root): now UID=0 GID=0
>    0.001745	[D5] send_to_all: SETINFO device.type "ups"
>    0.001810	[D5] send_to_all: SETINFO driver.state "init.device"
>    0.001869	[D1] upsdrv_initups (non-SHUT)...
>    0.001931	[D2] Initializing an USB-connected UPS with library libusb-1.0.27 (API: 0x100010a) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
>    0.004187	[D2] Checking device 1 of 9 (075D/0300)
>    0.006638	[D2] - VendorID: 075d
>    0.006700	[D2] - ProductID: 0300
>    0.006872	[D2] - Manufacturer: -BMS-
>    0.007045	[D2] - Product: Smart-Battery
>    0.007178	[D2] - Serial Number: XXXXXXXX
>    0.007240	[D2] - Bus: 007
>    0.007370	[D2] - Bus Port: 001
>    0.007432	[D2] - Device: 002
>    0.007599	[D2] - Device release number: 0001
>    0.007660	[D2] Trying to match device
>    0.007818	[D2] match_function_subdriver (non-SHUT mode): matching a device...
>    0.007943	[D3] match_function_regex: matching a device...
>    0.008016	[D2] Device matches
>    0.008149	[D2] Reading first configuration descriptor
>    0.008220	[D3] libusb_kernel_driver_active() returned 0: Success
>    0.008315	[D2] Claimed interface 0 successfully
>    0.008377	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
>    0.008820	[D2] Retrieved HID descriptor (expected 9, got 9)
>    0.008891	[D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 23 01
>    0.008954	[D3] HID descriptor length (method 1) 291
>    0.009013	[D4] i=0, extra[i]=09, extra[i+1]=21
>    0.009080	[D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 23 01
>    0.009144	[D3] HID descriptor length (method 2) 291
>    0.009203	[D2] HID descriptor length 291
>    0.012375	[D2] Report Descriptor size = 291
>    0.012452	[D3] Report Descriptor: (291 bytes) => 05 84 09 04 a1 01 09 24 a1 00 05 84 85 01
>    0.012539	[D3]  09 25 09 1f 75 04 95 02 15 00 25 0f b1 03 09 fd 79 01 75 08 95 01 15 00 26
>    0.012621	[D3]  ff 00 b1 03 09 fe 79 02 b1 23 09 ff 79 02 b1 23 05 85 09 89 b1 22 05 85 09
>    0.012703	[D3]  d1 09 8b 25 01 75 01 95 02 b1 a2 95 06 b1 83 05 85 09 2c 75 08 25 ff 95 01
>    0.012782	[D3]  b1 83 09 83 25 64 b1 82 09 67 b1 82 09 8c b1 83 09 66 b1 83 09 68 75 20 55
>    0.012864	[D3]  00 15 00 27 ff ff ff 75 66 01 10 95 01 b1 83 05 84 09 40 67 21 d1 f0 00 55
>    0.012948	[D3]  05 75 10 95 01 27 ff ff 00 00 b1 83 09 30 95 01 b1 83 05 84 09 02 a1 02 05
>    0.013029	[D3]  85 09 d0 09 44 09 45 09 42 09 61 09 d1 09 69 65 00 55 00 25 01 75 01 95 07
>    0.013111	[D3]  b1 83 95 01 b1 a3 c0 05 84 09 57 85 09 75 20 15 ff 27 ff ff ff 7f 66 01 10
>    0.013234	[D3]  b1 82 09 56 85 0a b1 82 85 02 75 08 95 01 81 a3 05 85 09 66 15 00 25 64 75
>    0.013315	[D3]  08 95 01 81 a3 09 68 66 01 10 55 00 75 20 95 01 15 00 27 ff ff ff fe 81 83
>    0.013453	[D3]  05 84 09 30 67 21 d1 f0 00 55 05 75 10 95 01 81 82 85 06 75 08 95 20 81 83
>    0.013517	[D3]  c0 c0
>    0.013700	HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x01
>    0.013845	HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -16777217 as 4278190079 in ReportID: 0x02
[... cut ...]
>    0.041728	[D1] Driver initialization completed, beginning regular infinite loop
>    0.041752	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
>    0.041783	[D5] send_to_all: SETINFO driver.state "updateinfo"
>    0.041801	[D1] upsdrv_updateinfo...
> ...
> ```
> 
> Not sure if the data beyond the first expected 9 bytes has anything interesting yet.

@sovanyio looks like you were able to extract more data, can you tell me how you were able to get this?

mogorman, 2025-08-12:
can someone tell me how to use this locally? would appreciate it.

JefferMC, 2025-12-05:
For what it's worth, I have the 1000/800W Pro.  It showed up in nut-scanner -U.  I configured it in UPS.CONF using that information.  I came up and seemed to be working fine as it charged.  However, a few minutes after it finished charging, it went into status OB DISCHARGE, although it's still connected to AC.  This is the current NUT uspc output for it.  I'd gladly provide other information if helpful.

```
battery.charge: 99
battery.runtime: 960
battery.type: Lion
device.mfr: -BMS-
device.model: Smart-Battery
device.serial: J8D52120E5FBE615854:25
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: Smart-Battery
driver.parameter.productid: 0300
driver.parameter.serial: J8D52120E5FBE615854:25
driver.parameter.synchronous: auto
driver.parameter.vendorid: 075D
driver.state: quiet
driver.version: 2.8.1
driver.version.data: iDowell HID 0.2
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
ups.delay.shutdown: 20
ups.delay.start: 30
ups.mfr: -BMS-
ups.model: Smart-Battery
ups.productid: 0300
ups.serial: J8D52120E5FBE615854:25
ups.status: OB DISCHRG
ups.timer.shutdown: 1
ups.timer.start: 257
ups.vendorid: 075d
```
Details... I just realized that all of the above happened with the power OFF.  It was not supplying power to the outlets.  The front panel was not exactly per the book, but it lit up and showed the battery status on a button press.  I turned it on (by holding the button for 3 seconds) and the display panel went to book-normal and the ups.status  switched to OL.  So OB DISCHRG might mean device turned off.

pponce, 2025-12-09:
I also used winnut client just to check goldenmate status, and it also is showing 220V input/output for me ( I'm in the US so it should be about 120V)

jimklimov, 2025-12-10:
In the comment https://github.com/networkupstools/nut/issues/3015#issuecomment-3395494672 above with logs of both WinNUT and actual `upsc`, there seem to be no reports from the device about input/output voltage. In fact, it may be WinNUT client (a third-party project, some one of several by that or similar name as it was unmaintained and forked and reimplemented all over again over the years) using fallbacks (a lot of them mentioned in the log above), e.g.:
```
10/12/2025 8:00:47 PM Pid: 128580 UPS_Network : Apply Fallback Value when retrieving input.voltage
10/12/2025 8:00:47 PM Pid: 128580 UPS_Network : Apply Fallback Value when retrieving output.voltage
...
```

So I guess "220V" may be seen just because it is the hardcoded fallback default in that client program. Per analysis above, the UPS does not really report much (in that descriptor with some 27 entries or so), so this data is just not there originally.

DeadEnded, 2026-01-12:
Just installed a Goldenmate UPS this weekend and found my way to this thread...
SO - since it seems a few people have been poking around on this, I decided to try and give it a whirl...

I spent a day digging and failing trying to figure out how to get a dev container running nut... then how and what to change where... even as far as installing Roxx and HIDRDD (which on Alpine container was not straight forward... lots of compile errors that I don't know how I got through)... 

Long story short... I think I have a working first draft.
```
/nut_dev/nut-2.8.4 # upsc goldenmate
battery.capacity.design: 100
battery.capacity.full: 100
battery.charge: 99
battery.charge.low: 5
battery.runtime: 125040
battery.type: Lion
battery.voltage: 13.2
battery.voltage.nominal: 3.7
device.mfr: -BMS-
device.model: Smart-Battery
device.serial: #######
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.4
driver.version.data: Goldenmate HID 0.1
driver.version.internal: 0.67
driver.version.usb: libusb-1.0.29 (API: 0x0100010B)
ups.delay.shutdown: 20
ups.delay.start: 30
ups.mfr: -BMS-
ups.model: Smart-Battery
ups.productid: 0300
ups.serial: #######
ups.status: OL CHRG
ups.vendorid: 075d
```

Connecting to it with Home-Assistant integration it pulls in the values (most) as well:

[image]

I have the following files created/updated:
`drivers/goldenmate-hid.c`
`drivers/goldenmate-hid.h`
`drivers/usbhid-ups.c`
`drivers/Makefile.am`
`data/driver.list.in`

I'd like to take this to completion with a PR, but need to know what other files I need to update to make it complete... for example, does it require any document updates? Also, I guessed the `support level` in `driver.list.in` would be 1. I wasn't successful finding any outline for updates - so if anyone would be so kind as to push me in the right direction, I would really like to finish this out and submit it.

Thanks!
DeadEnd

jimklimov, 2026-01-12:
@DeadEnded : thanks, that looks encouraging! generally there's also `NEWS.adoc` to add a notice in, maybe the `docs/man/usbhid-ups.txt` man page, and a DRIVER_VERSION bump in `usbhid-ups.c`. Maybe more, but can't say well before a PR :)

jimklimov, 2026-01-13:
In NEWS, this would be a (new) sub-driver under `usbhid-ups`. See earlier releases, e.g. in 2.8.4 section there are several `Introduced XYZ protocol/subdriver/support...` lines for new features in large drivers.

DeadEnded, 2026-01-14:
Yes, unfortunately I started with no knowledge of what I was looking at. I worked my way through all of this learning how it worked as I went.

I'm sure others knew what I have learned already - this is what it took for me to get there, lol.

jimklimov, 2026-01-15:
Doing this in a claim method sounds like running down the correct path :) Hard to say more without seeing the actual code (as a PR proposal).