networkupstools/nut #3018: Nut 2.8.3 building from source, binaries end up in /usr/local/ups/bin in spite of different assignment in configure file

DanoBoca, 2025-07-15:
Nut 2.8.3 building from source, binaries end up in /usr/local/ups/bin in spite of different assignment in configure file.
I've updated the configure file from 2.8.1, assigned
--prefix=/usr \
--exec_prefix=/usr \
--sbindir=/usr/sbin \
--bindir=/usr/bin \

however the binaries end up in /usr/local/ups/bin and /usr/local/ups/sbin

below is the configure file.   Any pointers on what is not correct or stopping the assignments frm workding

./configure \
--prefix=/usr \
--includedir=/usr/include \
--mandir=/usr/share/man \
--infodir=/usr/share/info \
--exec_prefix=/usr \
--sysconfdir=/etc/nut \
--sbindir=/usr/sbin \
--bindir=/usr/bin \
--libexecdir=/usr/lib/nut  \
--srcdir=. \
--disable-silent-rules \
--libdir=/usr/lib/`gcc -print-multiarch` \
--with-drvpath=/lib/nut \
--datadir=/usr/share/nut \
--with-udev-dir=/lib/udev \
--with-systemdsystemunitdir=/lib/systemd/system \
--with-user=root  \
--with-group=root \
--with-usb \
--with-cgi \
--with-cgipath=/usr/lib/cgi-bin/nut \
--with-drivers=usbhid-ups,dummy-ups \
--with-doc=man \
--with-serial=no \
--with-pkgconfig-dir=/usr/lib/'gcc -print-multiarch'/pkgconfig \
--with-nut_monitor \
--enable-inplace-runtime

Debian Trixie, Raspberry PI 5.

jimklimov, 2025-07-15:
Is the file literally as spelled above? If there are no trailing backslashes and multiple lines, these are two dozen independent lines as far as shell is concerned - so you get default `./configure` without any tweaks.

jimklimov, 2025-07-15:
Ah, lack of backslashes here was because this block was not pre-formatted in source. Added backticks around, and they got revealed.

For pre-requisites see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests => the https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt document (also [as a rendered page on NUT website](https://networkupstools.org/docs/qa-guide.chunked/_prerequisites_for_building_nut_on_different_oses.html)).

Are you building from git checkout, or from tarball?

In the end of the `configure` script, it should print a summary with paths etc. and the `CONFIG_FLAGS` it recognized. Does it show the settings you intended it to have?

ragd011, 2025-07-25:
I'm having the same issue.  I'm doing a very basic install.  usb, user, group.  What is the minimum of the above to see if i can resolve this because after a succesful make install, the system can find any of the commands.  make install-as-root fails on not finding certain files.