networkupstools/nut #3020: [TS SHARA 4450] Incorrect output.voltage reported

mhalano, 2025-07-16:
I have a TS SHARA 4450 using nutdrv_qx.

It reports the wrong value for `output.voltage`. Out-of-the-box, the voltage is 115V, but I changed, using the switch below the device, so the output became 220V, but I still get reported around 115V. I tested with a multimeter and it's arout 220V the output.

Please, let me know if you need any tests, like a patch, another driver, etc.

I'm still tweaking the device, and it's for my home office, so tests can be done easily.

```
mhalano@skynet:/var/lib$ upsc tsshara
Init SSL without certificate database
battery.charge: 100
battery.voltage: 27.6
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr: TS -222  240923
device.model: Senoid  32
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyACM0
driver.parameter.protocol: megatec
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Megatec 0.07
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.current.nominal: 100.0
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 226.0
input.voltage.fault: 226.0
output.voltage: 115.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V01020405A
ups.load: 0
ups.status: OL
ups.temperature: 27.0
ups.type: offline / line interactive
```

jimklimov, 2025-07-16:
Hello, can you please try running the driver with high debug verbosity (e.g. `systemctl stop nut-driver@tsshara.service`, run the program like `/lib/nut/nutdrv_qx -a tsshara -DDDDDD -d1`, copy the output here, and restart the service). This should show strings reported by the device, and whether NUT misinterprets the output or those reports are wrong.

Also, just in case, check in the manual if something has to be cycled in the device after you flip the switch. I wonder if there is a need, and some way, to reboot its controller? Maybe unplug the UPS, disconnect batteries, wait and add them back - assuming there is no extra power for the controller to be useful in case the main batteries die off?

jimklimov, 2025-07-16:
Thanks! So `112.0` is literally what the device says, several times :\

If you feel adventurous, try building current NUT code base per https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests to test with some of the newer protocol subdrivers introduced after NUT v2.8.1 release, e.g. check if Q6 is supported by the device and gives different answers?

mhalano, 2025-07-16:
I will try to get a newer version and test other drivers then. I also contacted the manufacturer to check if there is any way to do a reset in the microprocessor to check if the information returns correctly. I will let you know.

mhalano, 2025-07-16:
I bumped the version, but nothing changed:
```
mhalano@skynet:~$ sudo /lib/nut/nutdrv_qx -a tsshara -DDDDDD -d1
   0.000001     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.3 release - Generic Q* USB/Serial driver 0.42
USB communication driver (libusb 1.0) 0.50
   0.000041     [D1] upsdrv_makevartable...
   0.000053     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x0100010B)"
   0.000055     [D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010B)
   0.000105     [D5] do_upsconf_args: confupsname=(null), var=maxretry, val=3
   0.000110     [D5] do_upsconf_args: call do_global_args()
   0.000112     [D3] do_global_args: var='maxretry' val='3'
   0.000161     [D5] do_upsconf_args: confupsname=tsshara, var=driver, val=nutdrv_qx
   0.000165     [D5] do_upsconf_args: call main_arg()
   0.000167     [D3] main_arg: var='driver' val='nutdrv_qx'
   0.000170     [D5] do_upsconf_args: not a main_arg()
   0.000172     [D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000175     [D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.000178     [D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000181     [D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000185     [D5] do_upsconf_args: confupsname=tsshara, var=desc, val=tsshara
   0.000189     [D5] do_upsconf_args: call main_arg()
   0.000192     [D3] main_arg: var='desc' val='tsshara'
   0.000196     [D5] do_upsconf_args: confupsname=tsshara, var=port, val=/dev/ttyACM0
   0.000199     [D5] do_upsconf_args: call main_arg()
   0.000201     [D3] main_arg: var='port' val='/dev/ttyACM0'
   0.000204     [D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/ttyACM0, reloadable=0, reload_flag=0
   0.000208     [D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000213     [D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyACM0"
   0.000221     [D5] do_upsconf_args: confupsname=tsshara, var=protocol, val=megatec
   0.000227     [D5] do_upsconf_args: call main_arg()
   0.000231     [D3] main_arg: var='protocol' val='megatec'
   0.000234     [D5] do_upsconf_args: not a main_arg()
   0.000238     [D5] do_upsconf_args: process as value
   0.000241     [D6] testvar_reloadable: searching for var=protocol, vartype=2, reload_flag=0
   0.000246     [D6] testvar_reloadable: found var=protocol, val='(null)' => 'megatec', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000252     [D6] testvar_reloadable: verdict for (re)loading var=protocol value: 1
   0.000265     [D5] send_to_all: SETINFO driver.parameter.protocol "megatec"
   0.000274     [D1] Network UPS Tools version 2.8.3 release built with gcc (Ubuntu 14.3.0-1ubuntu1) 14.3.0 and configured with flags: --build=x86_64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/x86_64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --prefix=/usr --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=${prefix}/lib/x86_64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/usr/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/usr/lib/udev --with-systemdsystemunitdir=/usr/lib/systemd/system --with-systemdshutdowndir=/usr/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d --with-python=python3 --with-python3=/usr/bin/python3 --with-libsystemd --with-doc=man,html-single,html-chunked,pdf
   0.000305     [D1] debug level is '6'
   0.000311     [D5] send_to_all: SETINFO driver.debug "6"
   0.000317     [D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000932     [D1] Succeeded to become_user(nut): now UID=113 GID=114
   0.000941     [D1] Signalling UPS [tsshara]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.000949     Can't open /run/nut/nutdrv_qx-tsshara: No such file or directory
   0.000951     [D1] Request for other driver to exit returned code -1
   0.000954     [D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.000958     [D5] send_to_all: SETINFO device.type "ups"
   0.000961     [D5] send_to_all: SETINFO driver.state "init.device"
   0.000964     [D1] upsdrv_initups...
   0.101766     [D2] Skipping protocol Voltronic 0.10
   0.101804     [D2] Skipping protocol Voltronic-QS 0.09
   0.101811     [D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.101816     [D2] Skipping protocol Mustek 0.08
   0.101821     [D2] Skipping protocol Megatec/old 0.08
   0.101826     [D2] Skipping protocol BestUPS 0.07
   0.101831     [D2] Skipping protocol Mecer 0.08
   0.101902     [D3] send: 'Q1'
   0.310148     [D5] read: (47 bytes) => 28 32 32 31 2e 30 20 32 32 31 2e 30 20 31 31 32 2e 30
   0.310192     [D5]  20 30 30 30 20 36 30 2e 30 20 32 37 2e 32 20 32 38 2e 30 20 30 30 30 30 31
   0.310199     [D5]  30 30 31 0d
[... cut ...]
input.current.nominal: 100.0
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 221.0
input.voltage.fault: 221.0
output.voltage: 112.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V01020405A
ups.load: 0
ups.status: OL
ups.temperature: 27.0
ups.type: offline / line interactive
   1.586901     [D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   1.586910     [D1] upsdrv_cleanup...
   1.589042     [D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

closed 2025-07-16 by mhalano

jimklimov, 2025-07-16:
Ok, thanks for the update and good luck with the support :)

mhalano, 2025-07-17:
I think a new firmware would do a different calculation to consider the switch position. The value returned doesn't consider the switch position right now. I measured with a multimeter, and the output was around 220V, but the result of the query still was around 115V... The strange part is the reported value fluctuates, some times is 113V, sometimes 112V, etc. so I don't know where they are getting this value right now.