networkupstools/nut #3038: EcoFlow Delta 3 Plus Support

ragd011, 2025-07-30:
Just wanted to open this since it appears different than the EcoFlow River 3 issue.

I've installed a VM of Ubuntu Questing in proxmox and passed USB through so I could try 2.8.3 with my EcoFlow.

In the syslog I've been seeing the following about every minute:
2025-07-30T17:54:51.781719-04:00 nutmaster nut-monitor[1239]: UPS ecoflow@localhost on battery
2025-07-30T17:54:51.791444-04:00 nutmaster nut-monitor[10674]: Warning: no shutdown command defined!
2025-07-30T17:54:56.794699-04:00 nutmaster nut-monitor[1239]: UPS ecoflow@localhost on line power
2025-07-30T17:54:56.796252-04:00 nutmaster nut-monitor[1239]: UPS ecoflow@localhost battery is low
2025-07-30T17:54:56.805714-04:00 nutmaster nut-monitor[10680]: Warning: no shutdown command defined!
2025-07-30T17:54:56.806631-04:00 nutmaster nut-monitor[10679]: Warning: no shutdown command defined!

I don't have it configured do do anything yet but from this it appears to think the UPS status is changing every minute or less.  I haven't seen a loss of connection or anything.

The unit is plugged in with no alerts form the app.  Battery shows 93%.

I'm using the default upshid-usb driver.

Let me know if there is anything else woth posting.  Thanks.

jimklimov, 2025-07-30:
For now I can only direct to other discussions about the vendor - e.g. following the label https://github.com/networkupstools/nut/issues?q=state%3Aopen%20label%3A%22EcoFlow%22

My understanding is that some of their firmwares got USB thoroughly wrong, and that generally it is a quickly evolving brand (or model lineage). Not sure from that if there is anything typically wrong with their hardware though.

I guess a first good step would be to check that your installed firmware is the latest one for the model?..

ragd011, 2025-07-30:
ok Thanks.  Unfortunately, I'm learning this model is fairly new and they aren't very good about firmware updates.  Most people open up support cases to get the latest firmware pushed.  At this point can't recommend this model or brand.

closed 2025-07-31 by ragd011

jimklimov, 2025-08-01:
Nice! Commendable, indeed!