networkupstools/nut #3043: Revise auto-detection of dependency library versions, to have a consistent mesh

jimklimov, 2025-08-04:
CONTEXT: *BSD builds do tend to warn about mismatch of library variants pulled by different paths through dependencies; did so since "forever". This may be an incoherence induced by the distro though, not sure how leaf projects like NUT can impact that properly - it is probably more of a quest for packagers to use `--with-XXX-libs=...` sort of options responsibly.

As https://github.com/networkupstools/nut/pull/3042/checks?check_run_id=47288172772 goes, "Annotations" are:
```
[Check warning on line 0 in -](https://github.com/networkupstools/nut/pull/3042/files#annotation_37380870046)

@nut-ci-farm-jenkins nut-ci-farm-jenkins / All Issues - Aggregated

GCC4 Linker Error

NORMAL:
libfontconfig.so.2, needed by /usr/pkg/lib/libgd.so, may conflict with libfontconfig.so.1
```

```
[Check warning on line 0 in -](https://github.com/networkupstools/nut/pull/3042/files#annotation_37380870048)

@nut-ci-farm-jenkins nut-ci-farm-jenkins / All Issues - Aggregated

GCC4 Linker Error

NORMAL:
libintl.so.1, needed by /usr/pkg/lib/libfontconfig.so, may conflict with libintl.so.8
```

```
[Check warning on line 0 in -](https://github.com/networkupstools/nut/pull/3042/files#annotation_37380870051)

@nut-ci-farm-jenkins nut-ci-farm-jenkins / All Issues - Aggregated

GCC4 Linker Error

NORMAL:
libfreetype.so.19, needed by /usr/X11R7/lib/libfontconfig.so.2, may conflict with libfreetype.so.6
```

```
[Check warning on line 0 in -](https://github.com/networkupstools/nut/pull/3042/files#annotation_37380870052)

@nut-ci-farm-jenkins nut-ci-farm-jenkins / All Issues - Aggregated

GCC4 Linker Error

NORMAL:
libintl.so.1, needed by /usr/pkg/lib/libneon.so, may conflict with libintl.so.8
```

```
[Check warning on line 0 in -](https://github.com/networkupstools/nut/pull/3042/files#annotation_37380870054)

@nut-ci-farm-jenkins nut-ci-farm-jenkins / All Issues - Aggregated

GCC4 Linker Error

NORMAL:
libcrypto.so.14, needed by /usr/pkg/lib/libnetsnmp.so, may conflict with libcrypto.so.3
```

```
[Check warning on line 0 in -](https://github.com/networkupstools/nut/pull/3042/files#annotation_37380870055)

@nut-ci-farm-jenkins nut-ci-farm-jenkins / All Issues - Aggregated

GCC4 Linker Error

NORMAL:
libstdc++.so.9, needed by /usr/pkg/lib/libcppunit.so, may conflict with libstdc++.so.7
```

_Originally posted by @jimklimov in https://github.com/networkupstools/nut/issues/3042#issuecomment-3149368454_
            

QUEST: Our `configure` script in "auto" mode could try to test for such messages and/or use `ldd`, `elfdump` et al. to produce a minimally-conflicting set of linker options. This may however be too platforms/toolkits-dependent.

jimklimov, 2025-08-04:
I'd say this is an inconsistency on specific distro's packaging side more than on NUT side. For example, we build `snmp-ups` driver with crypto support (for SNMPv3 etc.) and so require `libnetsnmp` and `openssl` packages. What does `pkgconf` and `ldd` tell us?

```
-bash-5.2$ pkgconf --libs netsnmp
-L/usr/pkg/lib -lnetsnmp

-bash-5.2$ ldd /usr/pkg/lib/libnetsnmp.so
/usr/pkg/lib/libnetsnmp.so:
        -lm.0 => /usr/lib/libm.so.0
        -lc.12 => /usr/lib/libc.so.12
        -lcrypto.14 => /usr/lib/libcrypto.so.14
        -lcrypt.1 => /lib/libcrypt.so.1
        -lpthread.1 => /usr/lib/libpthread.so.1

-bash-5.2$ pkgconf --libs openssl
-L/usr/pkg/lib -lssl -lcrypto

-bash-5.2$ ls /usr/pkg/lib/libcrypto.*
libcrypto.a     libcrypto.so    libcrypto.so.3
```

So on one hand we indeed require (per `pkgconf` output) to directly link with `/usr/pkg/lib/libcrypto.3`, on another - indirectly `/usr/lib/libcrypto.so.14` via `/usr/pkg/lib/libnetsnmp.so`...

jimklimov, 2025-08-04:
There is however a choice of at least some of those pkgconfig metadata:
```
-bash-5.2$  find / -name '*snmp*.pc' 2>/dev/null
/usr/pkg/lib/pkgconfig/netsnmp-agent.pc
/usr/pkg/lib/pkgconfig/netsnmp.pc

-bash-5.2$ find / -name '*openssl*.pc' 2>/dev/null
/usr/lib/pkgconfig/openssl.pc
/usr/pkg/lib/pkgconfig/openssl.pc
```

gdt, 2025-08-04:
The mismatch issue can be serious, but it's basically because some libraries are provided by the base system, and then there are newer versions in pkgsrc/ports.   When a package is installed that can't use the base system version, then the new one is present, and that can get picked up.

This is really not a nut problem.   Nut's responsibility is to be able to build with any non-ancient version, so that it doesn't force new.

Overall the combination of base/pkg needs to figure out which version to use and stick with it.  This is hard because often there is a library A in base that depends on base ssl, and then other programs want to link against openssl and A.

Generally -L/-R should load from pkgsrc/ports first, and then /usr.  It's adding a kludge that's going to have other negative consequences to fight that, and I strongly recommend against it.

pkgsrc has a plan to decide if the base lib is good enough and if so use it, and if not force installation of a new one in pkgsrc.  And, to only make libraries visible to the build if they are declared as dependencies.

I would say that these issue are probably best raised with the packaging systems.

Linux has the same problem really, but it's resolved by "we expect the world to build with whatever old libs our Laggard Term Stable release has, and claim that it's the upstream's packages fault if it doesn't", and thus isn't the option to have newer.

As for pc files, -L/usr/lib is built in and not needed.  If they are buggy, fair enough and bugs can be filed -- but I don't see it as a nut bug or something to fix.  I don't think there is anything special about nut here, other than that the configure stuff in general seems much more complicated than it needs to be.   Adding more wouldn't be an improvement!