networkupstools/nut #3045: Inaccurate data reported by Powercom BNT-600AP (VID:PID 0d9f:0004) with usbhid-ups driver

vasmarfas, 2025-08-05:
Using a Powercom UPS with VID:PID 0d9f:0004 on a Raspberry Pi running Linux with NUT version 2.8.0 and the usbhid-ups driver.

The UPS is detected correctly, and the upsc command returns status and battery info, but many values look incorrect or outdated (for example, input.frequency is 0.0, input.voltage is 320.0, ups.date and battery.date are from 2010/12/20, which is obviously not current).

Driver config:

```
[bnt600ap]
driver = "usbhid-ups"
port = "auto"
vendorid = "0D9F"
productid = "0004"
```

Output from upsc:

```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 30
battery.date: 2010/12/20
battery.runtime: 832
battery.type: PbAc
device.mfr: POWERCOM Co.,LTD
device.model: HID UPS Battery
device.serial: 004-0D9F-000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0004
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0D9F
driver.version: 2.8.0
driver.version.data: PowerCOM HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 0.0
input.voltage: 320.0
input.voltage.nominal: 100
output.frequency: 48.0
output.voltage: 0.0
output.voltage.nominal: 100
ups.beeper.status: enabled
ups.date: 2010/12/20
ups.delay.shutdown: 20
ups.delay.start: 60
ups.load: 100
ups.mfr: POWERCOM Co.,LTD
ups.model: HID UPS Battery
ups.productid: 0004
ups.serial: 004-0D9F-000
ups.status: OL CHRG
ups.test.result: Done and passed
ups.timer.shutdown: 0
ups.timer.start: 0
ups.vendorid: 0d9f
```

Issues and questions:

Many returned values appear incorrect , especially input voltage/frequency parameters.

Would like to understand if it’s possible to improve parsing for this UPS model to get more accurate and current data.

Interested if there are plans to add official support for this model in the usbhid-ups driver with full parameter handling.

Happy to provide additional logs or USB traces if needed.

jimklimov, 2025-08-05:
Can you please check if current NUT (build from master branch) has this issue? Release v2.8.4 is brewing to hopefully be out this week.

vasmarfas, 2025-08-05:
There is output from `sudo /lib/nut/usbhid-ups -DDD -a bnt600ap`:

<details>
  <summary>Long code warning</summary>

```
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] do_global_args: var='maxretry' val='3'
   0.003229     [D3] main_arg: var='driver' val='usbhid-ups'
   0.006365     [D3] main_arg: var='port' val='auto'
   0.006561     [D3] main_arg: var='vendorid' val='0D9F'
   0.006692     [D3] main_arg: var='productid' val='0004'
   0.008129     [D1] debug level is '3'
   0.012141     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.018564     [D1] upsdrv_initups (non-SHUT)...
   0.041314     [D2] Checking device 1 of 2 (0D9F/0004)
   0.052347     [D2] - VendorID: 0d9f
   0.053784     [D2] - ProductID: 0004
   0.054450     [D2] - Manufacturer: POWERCOM Co.,LTD
   0.055800     [D2] - Product: HID UPS Battery
   0.056448     [D2] - Serial Number: 004-0D9F-000
   0.056514     [D2] - Bus: 001
   0.056560     [D2] - Device: unknown
   0.056612     [D2] - Device release number: 0001
   0.056658     [D2] Trying to match device
   0.056711     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.058118     [D3] match_function_regex: matching a device...
   0.059479     [D2] Device matches
   0.060935     [D2] Reading first configuration descriptor
   0.061071     [D3] libusb_kernel_driver_active() returned 0
   0.061209     [D2] Claimed interface 0 successfully
   0.061279     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.064328     [D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 c6 03
   0.064639     [D3] HID descriptor length (method 1) 966
   0.066068     [D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 c6 03
   0.066770     [D3] HID descriptor length (method 2) 966
   0.067260     [D2] HID descriptor length 966
   0.111964     [D2] Report Descriptor size = 966
   0.113342     [D3] Report Descriptor: (966 bytes) => 05 84 09 04 a1 01 09 24 a1 02 85 01 09 fe
   0.113480     [D3]  79 01 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 02 b1 22 05 85 85 05
   0.113587     [D3]  09 89 79 04 b1 22 85 07 09 8b b1 22 85 08 09 2c 25 03 b1 22 85 09 09 83 25
   0.113694     [D3]  64 b1 22 09 67 b1 22 85 0e 09 66 81 22 09 66 b1 a2 85 0b 09 8c b1 22 85 0c
   0.113797     [D3]  09 29 b1 a2 85 0d 09 85 75 10 27 ff ff 00 00 b1 a2 85 37 09 68 81 22 09 68
   0.113899     [D3]  b1 a2 05 84 85 0f 09 57 75 10 16 00 00 27 ff ff 00 00 b1 a2 85 10 09 56 75
   0.114007     [D3]  20 16 00 00 27 9f 86 01 00 b1 a2 85 11 09 40 75 10 16 64 00 26 f0 00 b1 22
   0.114110     [D3]  85 12 09 30 75 10 16 00 00 26 40 01 b1 a2 85 2c 09 36 75 08 25 5a b1 a2 85
   0.114213     [D3]  13 09 5a 75 08 15 00 25 03 b1 a2 09 02 a1 02 05 85 85 0a 09 44 75 01 15 00
   0.114314     [D3]  25 01 81 a2 09 44 b1 a2 09 45 81 a2 09 45 b1 a2 09 d0 81 a2 09 d0 b1 a2 09
   0.114417     [D3]  d1 81 a2 09 d1 b1 a2 09 42 81 a2 09 42 b1 a2 09 43 81 a2 09 43 b1 a2 09 4b
   0.114525     [D3]  81 a2 09 4b b1 a2 09 db 81 a2 09 db b1 a2 05 84 09 68 81 a2 09 68 b1 a2 09
   0.114627     [D3]  69 81 a2 09 69 b1 a2 09 73 81 a2 09 73 b1 a2 09 65 81 a2 09 65 b1 a2 95 03
   0.114730     [D3]  81 01 b1 01 95 01 06 86 ff 09 80 81 a2 09 80 b1 a2 c0 c0 05 84 09 12 a1 00
   0.114832     [D3]  85 15 09 58 75 08 25 66 b1 a2 85 16 09 70 75 08 25 03 b1 a2 05 85 85 17 09
   0.114934     [D3]  85 75 10 27 ff ff 00 00 b1 a2 85 18 09 66 75 08 15 00 25 64 b1 a2 05 84 85
   0.115038     [D3]  19 09 40 75 10 16 00 0c 26 00 30 b1 22 85 1a 09 30 75 10 16 00 09 26 00 24
   0.115143     [D3]  b1 a2 85 1b 09 56 75 20 16 00 00 27 9f 86 01 00 b1 a2 85 2d 09 36 75 08 25
   0.115244     [D3]  5a b1 a2 c0 05 84 09 1a a1 00 85 1c 09 40 75 10 16 64 00 26 f0 00 b1 22 85
   0.115346     [D3]  1d 09 30 75 10 16 00 00 26 40 01 b1 a2 85 2e 09 30 75 10 16 00 00 26 40 01
   0.115447     [D3]  b1 a2 85 1e 09 32 75 08 15 00 25 46 b1 a2 c0 05 84 09 1c a1 00 85 2f 09 41
   0.115554     [D3]  75 08 15 01 25 0c b1 22 85 1f 09 35 75 08 15 00 25 64 b1 a2 85 20 09 40 75
   0.115657     [D3]  10 16 64 00 26 f0 00 b1 22 85 21 09 30 75 10 16 00 00 26 40 01 b1 a2 85 30
   0.115760     [D3]  09 42 75 08 15 32 25 3c b1 22 85 22 09 32 75 08 15 00 25 46 b1 a2 85 23 09
   0.115861     [D3]  57 75 08 15 00 25 12 b1 a2 85 24 09 56 75 20 16 00 00 27 9f 86 01 00 b1 a2
   0.115962     [D3]  c0 05 84 85 06 09 01 79 05 b1 a2 85 25 09 5a 75 08 15 00 25 03 b1 a2 09 02
   0.116091     [D3]  a1 02 05 85 85 26 09 44 75 01 15 00 25 01 b1 a2 09 45 b1 a2 09 d0 b1 a2 09
   0.120056     [D3]  d1 b1 a2 09 42 b1 a2 09 43 b1 a2 09 4b b1 a2 09 db b1 a2 05 84 09 68 b1 a2
   0.125565     [D3]  09 69 b1 a2 09 73 b1 a2 09 65 b1 a2 09 6e b1 a2 09 6f b1 a2 09 71 b1 a2 06
   0.127505     [D3]  86 ff 09 80 b1 a2 c0 85 27 05 84 09 69 75 08 15 00 25 01 b1 a2 09 1e a1 00
   0.128634     [D3]  85 a0 09 01 75 08 95 07 15 00 26 ff 00 b1 03 85 a1 09 01 75 08 95 07 15 00
[... 2203 lines cut ...]
 104.847112     [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 104.847281     [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 104.847449     [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 104.847611     [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
 104.860511     [D3]  c5 54 91 d0 be b0 aa 05 00 f8 b5 05 00 b8 72 db b6 43 02 92 68
 104.860606     [D1] Got 1 HID objects...
 104.860731     [D3] Report[buf]: (2 bytes) => 0e 32
 104.860854     [D2] Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID: 0x0e, Offset: 0, Size: 8, Value: 50
 104.860976     [D1] Quick update...
 104.867547     [D3] Report[get]: (3 bytes) => 0f 00 00
 104.871035     [D2] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x0f, Offset: 0, Size

jimklimov, 2025-08-05:
Check here for hints and prerequisites: https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

vasmarfas, 2025-08-06:
@jimklimov  After night there is the results. This UPS connected to 12v40Ah LiFePo4 battery, which enough to handle about 20-25 hours of current load (4 mikrotik routers), so it can not be discharged to 60% in 1 hour.

[image]

[image]

[image]

[image]

vasmarfas, 2025-08-06:
It's quite strange, because the bnt600ap is connected to the rpi zero w, which is located within a radius of 1 meter from the wifi router, i.e. there should be no network outages. Is there any way to fix these unstable indicators? Even now I decided to check what indicators change if you pull the UPS out of the network (leave it on the battery) and realized that there are actually none. The incoming voltage can be either 0 or not 0. The status is still Online, Battery Charging. Except that only the percentage of battery charge seems to decrease, in general, it is difficult to monitor without stability. Although it has improved relative to version 2.8.0, it is still bad, maybe there are ways to improve stability?

The parameters in the photo are from top to bottom: 
- Input voltage
- Output voltage
- Status Data
- Battery charge
- Load
- Status

[image]
[image]