networkupstools/nut #3047: New PID needed for APC UPS

jspitael, 2025-08-07:
Hi,

I work for Schneider Electric and we will be introducing a new APC UPS that will use a VID of 0x051D and a PID of 0x0005.  We request that you add those values to https://[raw.githubusercontent.com/networkupstools/nut/master/scripts/upower/95-upower-hid.hwdb](https://raw.githubusercontent.com/networkupstools/nut/master/scripts/upower/95-upower-hid.hwdb).

If other information is required to support this UPS in NUT, please let me know.

Thanks,
Jim

jimklimov, 2025-08-07:
If it supports USB HID, please also check the `apc_usb_device_table[]` at https://github.com/networkupstools/nut/blob/a3e0664179a114d606aa104140ed444fd105f190/drivers/apc-hid.c#L89-L104 - most other series needed this or that tweak method to be referenced. Is one of those applicable here?

Otherwise, a likely suitable driver is https://github.com/networkupstools/nut/blob/master/drivers/apc_modbus.c but it has certain stability issues (largely due to libmodbus, but for end-users it either works or not overall).

jimklimov, 2025-08-08:
Per offline discussion, answering the questions above would need some more time so pushing to next NUT release.

jspitael, 2025-09-08:
> @jspitael<https://github.com/jspitael> : I'm getting back from the offline vacation soon; are there any news on your side? (Which of the protocol drivers would the new USB IDs best apply to?)

Hi Jim,

UPSs with PID of 0x0005 will support the HID power summary.  Exactly which usages are still not finalized.

Is that sufficient for now?

Regards,

Jim

jimklimov, 2025-09-08:
I guess so, thanks. Eventually, check with `drivers/apc-hid.c` when the device is ready, to see if needed data points are handled (mapping NUT concept names per `docs/nut-names.txt` to USB paths).

FWIW, I'm also working on a solution for issue #3082 to help tracking down usages that are not mapped (seen in the report, not referenced by run-time walk of the mapping table).