networkupstools/nut #3048: Question => My UPS is working with blazer_ser but I cannot make it work with utdrv_qx driver

rosset, 2025-08-07:
Hi folks, this is my current issue, my UPS (Brazil / China Coletek brand) is "working"  with  the blazer_ser driver, but I cannot make it work with utdrv_qx driver.

The UPS is recognized as ENERMAX PG 2500VA

All details below, so are:
- Assumption #1, per the logs, is this UPS working with the blazer_ser / Megatec / Q1 protocols.
- Question #1, is it possible to use the utdrv_qx with the Megatec / Q1 subdriver ?
- Question #2, is the blazer_ser development on hold or it's still getting updates / fixes ?
- Question #3, should I try to setup with the new driver or just keep the Nut running with the blazer_ser driver ?

This is my first nut / UPS usage, can someone help to clarify if it's possible to migrate to the "new" driver ?

Thanks!

UPS / Nobreak model => Coletek UPS PG II 3200va

I was unable to find the datasheet for the UPS, I suspect internally it's another "brand".

No software / drivers provided by the manufacturer (text in PT-BR): 
https://www.coletek.com.br//nobreak-ups-pg-ii-3200va-bivolt-coletek-energia/p

I did purchased through this seller:
https://www.kabum.com.br/produto/609640/nobreak-coletek-ups-pg-ii-3200va-biv-4-baterias-10-tomadas-preto-pgii-3200ba-4b10t

Working with Nut 2.8.2 from Homebrew with blazer_ser driver.

UPS identified as Enermax PG 2500VA and using Megatec/Q1 protocol commands.

For now I was unable to get the UPS work with utdrv_qx / Megatec / Q1 protocol commands.

/opt/homebrew/bin/upsc myups@localhost
battery.charge: 100
battery.voltage: 27.40
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr: ENERMAX
device.model: PG 2500VA
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: blazer_ser
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/cu.usbserial-A50285BI
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2
driver.version.internal: 1.61
input.current.nominal: 3.0
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 214.0
input.voltage.fault: 214.0
input.voltage.nominal: 110
output.voltage: 119.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: VER 0.7
ups.load: 2
ups.mfr: ENERMAX
ups.model: PG 2500VA
ups.status: OL
ups.temperature: 31.0
ups.type: offline / line interactive

jimklimov, 2025-08-07:
What errors do your attempts with `nutdrv_qx` show? Notably, did you spell its name right - in this post it was `utdrv_qx` all over the place; could that be the trouble on your computer also?

rosset, 2025-08-07:
> What errors do your attempts with `nutdrv_qx` show? Notably, did you spell its name right - in this post it was `utdrv_qx` all over the place; could that be the trouble on your computer also?

typo, I'll share the logs when I back home w/ access to the HW, thanks!

jimklimov, 2025-08-07:
Note: I'll be on offline-mostly vacation till end of month, but maybe someone else would chime in with suggestions during this time.