networkupstools/nut #3053: NUT 2.8.4 ./configure error

teoberi, 2025-08-08:
./configure \
CPPFLAGS="-O2 -D_FORTIFY_SOURCE=3" \
CFLAGS="-fPIC -pie -fstack-protector-strong" \
CXXFLAGS="-fPIC -pie -fstack-protector-strong" \
LDFLAGS="-Wl,-z,now,-s,--enable-new-dtags" \
--build=x86_64-slackware-linux \
--disable-static \
--without-python2 \
--without-ssl \
--without-usb \
--with-user=nut \
--with-group=nut

> configure: Got explicit list of document formats to build or not: man=auto; formats not listed will be silently skipped
checking whether to requested to build and install documentation... 'man=auto' => 'man=auto'
checking desire and ability to build man documentation... auto
checking if we can build man... yes
checking whether to would build specific documentation format(s)... yes  man-man mans-progs-built
configure: error: Can not enable WITH_CHANGELOG_HTML_SINGLE due to build settings or environment

teoberi, 2025-08-08:
Asciidoc and a2x do not exist in Slackware only rubygem-asciidoctor!

closed 2025-08-08 by teoberi

jimklimov, 2025-08-08:
That *is* a reasonable request; unfortunately for a few decades NUT semver usage is a bit skewed in that regard. Like 3.x is "reserved" for a rewrite to C++ (or maybe Rust nowadays?) which is logged for 10 or 15 years or so. And scripts expecting more or less alphanumeric comparisons (so single-digit version components) do not help either. And releases being now sped up to have more than one in a year certainly still exceed the tolerance for what a "patch" number bump wold be by just collecting a lot of stuff since the last release (being snapshots of a master branch - and that also does not lend itself to proper tracking of e.g. Major.Minor releases as such, and Patch ones as branches for hotfixes).

I agree these are all true problems (at least in not being what the world expects of semver, or myself in other projects for that matter), and that recent hiccup with several errata in what was named NUT v2.8.3 could have been a few committed fixes for next patch versions, while "real work" marches on to be released in 2.9.x... But with the NUT team as small as it is, some corners get cut.

Maybe a later Major.Minor release would set the healthier trend.

As for the build breakage per se - I was surprised too; CI builds (of not-tagged code) went along with that PR's change just fine.