networkupstools/nut #3056: pynutclient: missing sdist

mtelka, 2025-08-09:
The recent releases (2.8.3 and 2.8.4) of `pynutclient` at PyPI are missing the sdist tarball.  Please add the sdist to PyPI to make downstream integration of `pynutclient` easier.  Thank you.

jimklimov, 2025-08-09:
Away for vacation, PR welcome.

mtelka, 2025-08-09:
> Also, was it present in earlier releases? I think we started fiddling with its recipes before 2.8.3?

Yes, 2.8.2 does have it: https://pypi.org/project/pynutclient/2.8.2/#files

jimklimov, 2025-09-10:
v2.8.1 also has the source tarball published. Curious, I don't see any significant recipe/script changes that would have impacted this.

I suppose earlier publications were rather manual (running `make upload*` goals on the workstation, etc.) and something got not-called with CI driving this based on release tags.

jimklimov, 2025-09-10:
Comparing local and CI logs, the local setup can use the "build module" and CI can not, probably leading to different outcomes:

* local workstation (note there are both `tar.gz` and `whl` files in `dist` dir):
```
$ make V=1 clean-local  dist
rm -rf PyNUTClient README.txt LICENSE-GPL3 dist build *.egg-info
rm -f .pypi-src .pypi-dist*
  PYPI  Generate PyPI module source
make[1]: Entering directory '/home/jim/nut/scripts/python/module'
  PYPI  Generate PyPI module distribution (using 'build' module)
* Building sdist...
running sdist
running egg_info
creating pynutclient.egg-info
writing pynutclient.egg-info/PKG-INFO
writing dependency_links to pynutclient.egg-info/dependency_links.txt
writing top-level names to pynutclient.egg-info/top_level.txt
writing manifest file 'pynutclient.egg-info/SOURCES.txt'
reading manifest file 'pynutclient.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE-GPL3'
writing manifest file 'pynutclient.egg-info/SOURCES.txt'
running check
creating pynutclient-2.8.4.44.7
creating pynutclient-2.8.4.44.7/PyNUTClient
creating pynutclient-2.8.4.44.7/pynutclient.egg-info
copying files to pynutclient-2.8.4.44.7...
copying LICENSE-GPL3 -> pynutclient-2.8.4.44.7
copying MANIFEST.in -> pynutclient-2.8.4.44.7
copying README.txt -> pynutclient-2.8.4.44.7
copying setup.py -> pynutclient-2.8.4.44.7
copying tox.ini -> pynutclient-2.8.4.44.7
copying PyNUTClient/PyNUT.py -> pynutclient-2.8.4.44.7/PyNUTClient
copying PyNUTClient/__init__.py -> pynutclient-2.8.4.44.7/PyNUTClient
copying PyNUTClient/test_nutclient.py -> pynutclient-2.8.4.44.7/PyNUTClient
copying pynutclient.egg-info/PKG-INFO -> pynutclient-2.8.4.44.7/pynutclient.egg-info
copying pynutclient.egg-info/SOURCES.txt -> pynutclient-2.8.4.44.7/pynutclient.egg-info
copying pynutclient.egg-info/dependency_links.txt -> pynutclient-2.8.4.44.7/pynutclient.egg-info
copying pynutclient.egg-info/top_level.txt -> pynutclient-2.8.4.44.7/pynutclient.egg-info
Writing pynutclient-2.8.4.44.7/setup.cfg
Creating tar archive
removing 'pynutclient-2.8.4.44.7' (and everything under it)
* Building wheel from sdist
* Building wheel...
running bdist_wheel
running build
running build_py
creating build
creating build/lib
creating build/lib/PyNUTClient
copying PyNUTClient/test_nutclient.py -> build/lib/PyNUTClient
copying PyNUTClient/PyNUT.py -> build/lib/PyNUTClient
copying PyNUTClient/__init__.py -> build/lib/PyNUTClient
installing to build/bdist.linux-x86_64/wheel
running install
running install_lib
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/wheel
creating build/bdist.linux-x86_64/wheel/PyNUTClient
copying build/lib/PyNUTClient/test_nutclient.py -> build/bdist.linux-x86_64/wheel/PyNUTClient
copying build/lib/PyNUTClient/PyNUT.py -> build/bdist.linux-x86_64/wheel/PyNUTClient
copying build/lib/PyNUTClient/__init__.py -> build/bdist.linux-x86_64/wheel/PyNUTClient
running install_egg_info
running egg_info
writing pynutclient.egg-info/PKG-INFO
writing dependency_links to pynutclient.egg-info/dependency_links.txt
writing top-level names to pynutclient.egg-info/top_level.txt
reading manifest file 'pynutclient.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE-GPL3'
[... 127 lines cut ...]
   802798     36 -rw-r--r--   1 runner   docker      35068 Aug  8 09:50 ./LICENSE-GPL3
   802818      0 -rw-r--r--   1 runner   docker          0 Aug  8 09:50 ./.pypi-dist
   802561     20 -rw-r--r--   1 runner   docker      19296 Aug  8 09:50 ./PyNUT.py.in
   802780      4 drwxr-xr-x   2 runner   docker       4096 Aug  8 09:50 ./PyNUTClient
   802806      4 -rw-r--r--   1 runner   docker         20 Aug  8 09:50 ./PyNUTClient/__init__.py
   802789      8 -rwxr-xr-x   1 runner   docker       6748 Aug  8 09:50 ./PyNUTClient/test_nutclient.py
   802781     20 -rw-r--r--   1 runner   docker      19296 Aug  8 09:50 ./PyNUTClient/PyNUT.py
   802746      0 -rw-r--r--   1 runner   docker          0 Aug  8 09:50 ./.pypi-tools-dist-wheel
   802772      8 -rw-r--r--   1 runner   docker       6060 Aug  8 09:50 ./Makefile
   802559      4 -rw-r--r--   1 runner   docker         16 Aug  8 09:50 ./MANIFEST.in
   802562     12 -rw-r--r--   1 runner   docker      12019 Aug  8 09:50 ./README.adoc
   802797     12 -rw-r--r--   1 runner   docker      12019 Aug  8 09:50 ./README.txt
   802816      4 drwxr-xr-x   2 runner   docker       4096 Aug  8 09:50 ./dist
   803592     28 -rw-r--r--   1 runner   docker      24970 Aug  8 09:50 ./dist/pynutclient-2.8.4rc1-py3-none-any.whl
   802817      0 -rw-r--r--   1 runner   docker          0 Aug  8 09:50 ./.pypi-dist-pip-wheel
   802565      4 -rw-r--r--   1 runner   docker        971 Aug  8 09:50 ./tox.ini
   802563      4 -rw-r--r--   1 runner   docker       2100 Aug  8 09:50 ./setup.py.in
   802807      0 -rw-r--r--   1 runner   docker          0 Aug  8 09:50 ./.pypi-src
   802558      4 -rw-r--r--   1 runner   docker        133 Aug  8 09:50 ./.gitignore
   803577      4 drwxr-xr-x   2 runner   docker       4096 Aug  8 09:50 ./pynutclient.egg-info
   803640     16 -rw-r--r--   1 runner   docker      13297 Aug  8 09:50 ./pynutclient.egg-info/PKG-INFO
   803579      4 -rw-r--r--   1 runner   docker          1 Aug  8 09:50 ./pynutclient.egg-info/dependency_links.txt
   803580      4 -rw-r--r--   1 runner   docker         12 Aug  8 09:50 ./pynutclient.egg-info/top_level.txt
   803581      4 -rw-r--r--   1 runner   docker        267 Aug  8 09:50 ./pynutclient.egg-info/SOURCES.txt
```

jimklimov, 2025-09-10:
Comments to `scripts/python/module/Makefile.am` goal `.pypi-dist-pip-wheel` even state that:
> Does not support "sdist" though. Oh well.

jimklimov, 2025-09-10:
So, adding the module to CI helped, see https://test.pypi.org/project/pynutclient/2.8.4.74/#files for example

Strangely, the stage name indicated it should have been there. But for some reason was not, maybe failed to install or run in earlier iterations?..