networkupstools/nut #3061: Incorrect online status value for KSTAR

ftoledo, 2025-08-14:
Hello, I have this UPS.
It's connected to the power line, but the status shows OB.
It should be OL.

Bus 003 Device 003: ID 09d6:0001 KSTAR UPS

```
root@pvmexico1:~# upsc ups_mexico
battery.charge: 1500
battery.charge.low: 150
battery.type: KSTAR?UPS
device.mfr: KSTAR
device.model: UPS HI
device.serial: KHID0
device.type: ups
driver.debug: 5
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0001
driver.parameter.subdriver: MGE HID
driver.parameter.synchronous: auto
driver.parameter.vendorid: 09d6
driver.state: quiet
driver.version: 2.8.4
driver.version.data: MGE HID 1.55
driver.version.internal: 0.67
driver.version.usb: libusb-0.1 (or compat)
ups.mfr: KSTAR
ups.model: UPS HI
ups.productid: 0001
ups.serial: KHID0
ups.status: OB
ups.vendorid: 09d6
root@pvmexico1:~# 
root@pvmexico1:~# upsc ups_mexico
battery.charge: 1500
battery.charge.low: 150
battery.type: KSTAR?UPS
device.mfr: KSTAR
device.model: UPS HI
device.serial: KHID0
device.type: ups
driver.debug: 5
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0001
driver.parameter.subdriver: MGE HID
driver.parameter.synchronous: auto
driver.parameter.vendorid: 09d6
driver.state: updateinfo
driver.version: 2.8.4
driver.version.data: MGE HID 1.55
driver.version.internal: 0.67
driver.version.usb: libusb-0.1 (or compat)
ups.mfr: KSTAR
ups.model: UPS HI
ups.productid: 0001
ups.serial: KHID0
ups.status: OB
ups.vendorid: 09d6

```
I run nut  2.8.4 built from sources on proxmox (debian 12)

pve-manager/8.4.5/57892e8e686cb35b (running kernel: 6.8.12-13-pve)

I'm attach the debug log

[ups_log.txt](https://github.com/user-attachments/files/21780919/ups_log.txt)

jimklimov, 2025-09-08:
I was looking at where it actually reports the power status, and it seems `UPS.PowerSummary.ACPresent` is available (logged early in report descriptor walk) but MGE HID mapping does not look at that. Actually just two subdrivers do: `APC HID` and `Belkin/Liebert HID`. So first try your luck with those...

ftoledo, 2025-09-08:
Hello, thanks for paying attention to this issue! and I hope it's been a few good vacation days! 
I config as Belkin driver. It's still missing some basic input/outpus values, but  now report OL as status. 
```
[ups_mexico]
    driver = usbhid-ups
    port = auto
    vendorid = 09d6
    productid = 0001
    debug_min = 5
    subdriver = "Belkin/Liebert HID 0.22"
```
```
# upsc ups_mexico
battery.charge: 1500
battery.charge.low: 150
battery.charge.warning: 450
battery.type: KSTAR?UPS
battery.voltage.nominal: 12.0
device.mfr: KSTAR
device.model: UPS HI
device.serial: KHID0
device.type: ups
driver.debug: 5
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0001
driver.parameter.subdriver: Belkin/Liebert HID 0.22
driver.parameter.synchronous: auto
driver.parameter.vendorid: 09d6
driver.state: updateinfo
driver.version: 2.8.4
driver.version.data: Belkin/Liebert HID 0.22
driver.version.internal: 0.67
driver.version.usb: libusb-0.1 (or compat)
ups.mfr: KSTAR
ups.model: UPS HI
ups.productid: 0001
ups.serial: KHID0
ups.status: OL
ups.vendorid: 09d6

```
this is the label data from the box (is sold under the name LYONN CTB-800) :

[image]

Also I attach new log from this driver

[nut-driver.log.gz](https://github.com/user-attachments/files/22211963/nut-driver.log.gz)

ftoledo, 2026-09-14:
I update to 2.8.5 tag. I found stale issue still pressent. I do a restart driver via cron every hour

do you thin that increasing the polling interval will help?

my actual config:

```
[ups_mexico]
    driver = "usbhid-ups"
    port = "auto"
    vendorid = "09D6"
    productid = "0001"
    product = "UPS HI"
    serial = "KHID0"
    vendor = "KSTAR"
    subdriver = "Belkin/Liebert HID 0.23"
    override.battery.charge.scale = 0.1
    debug_min = 5
```
I also test with  the with MGE HID 1.58 but its don't report status

[ups.log](https://github.com/user-attachments/files/32208176/ups.log)