networkupstools/nut #3062: NUT 2.8.4 tries to install a python module outside of its PREFIX

bjornfor, 2025-08-20:
A regression between 2.8.3 and 2.8.4, seen from a packager view point, is that the latest NUT release tries to install python modules outside of the given `./configure --prefix=PREFIX`:

```
$ ./configure --prefix=/nix/store/g8cdal4ikk8wzq0nh9l59a8h8clqqqmq-nut-2.8.4 [...]
[...]
$ make [...]
[...]
make[4]: Nothing to be done for 'install-exec-am'.                                                                                                                                                                                                                                                                                                                                                                                 
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/mkdir -p '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                                                  
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/mkdir -p '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                                                  
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/mkdir -p '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                                                  
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/mkdir -p '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                                                  
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/install -c -m 644 module/PyNUT.py '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                         
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/install -c module/test_nutclient.py '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                       
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/install -c -m 644 module/PyNUT.py '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                         
 /nix/store/ih779chzzag1nm91fgnrndml4mghm3la-coreutils-9.7/bin/install -c module/test_nutclient.py '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages'                                                                                                                                                                                                                                       
install: install: cannot create regular file '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages/PyNUT.py'cannot create regular file '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages/test_nutclient.py': Permission denied                                                                                                                           
: Permission denied                                                                                                                                                                                                                                                                                                                                                                                                                
install: cannot create regular file '/nix/store/9yh9ak97gn659bk4d3n411fx6c0ng7s2-python3-3.13.5/lib/python3.13/site-packages/PyNUT.py': Permission denied                                                                                                                                                                                                                                                                          
make[4]: *** [Makefile:1037: install-pynut_py3_siteDATA] Error 1                                                                                                                                                                                                                                                                                                                                                                   
```

I ran `./configure --help` and looked for "python" options, but didn't find anything about where to install those files. (I only found `--with-python(|2|3)` options.)

Please default to a path under the given `./configure --prefix=PREFIX` and/or add an option for specifying where to put those files.

This is blocking upgrading NUT in [nixpkgs](https://github.com/nixos/nixpkgs/), because it doesn't allow packages to mutate the install tree of other packages (which is the reason for that "Permission denied" error).

gdt, 2025-08-26:
@bjornfor If you are inclined, and you can figure out where this should have been explained so that another copy of you that hadn't had this discussion would have found it, that would help.

I agree with Jim that it's hard to satisfy both prefix and python norms.

It might be that pynut should be split into its own repo which is just a python module, following python norms, so that nut proper doesn't install python anything.   That would probably help a lot of people and be easier in the long run.

jimklimov, 2025-08-28:
I've had a moment near a charger to do something despicable, what anyone could do in the past weeks: `git diff -bu v2.8.{3,4} *.ac`

Apparently the only change around Python was to add support for NUT-Monitor with Py3+QT6 (to older Py3+QT5 and Py2+GTK2). So it is not as much that recipes changed to suddenly install stuff that was optional before, as that the build environment for earlier NUT releases did not suit the checked requirements of the UI client (was "too new" for that code base, kudos).

And the module default is "auto" meaning here it is (only IIRC?) installed if needed (e.g. the consumer client is to be installed).

As for separate packages, there's now a PyPI repo package for PyNUTClient. Probably some NUT configure options combo can (now? after some dev?) ensure that a client could be installed if a satisfactory module version exists. Or just be installed opportunistically (e.g. UI may be dysfunctional until module is installed somehow).

jimklimov, 2025-08-29:
My understanding is it was explored in the past and collapsed into a monorepo of sorts. I think a big part of rationale for bindings (like python or augeas) was to ensure they fit the protocol, recently covered by testing as code evolves and not catching up after a point release... The C (and C++ now) client libs are in tree - why won't those other languages be? jNut was expelled (or bever added inside?) before me though, and IIRC Perl is third-party still (might be a handful of them), so there is no 100% consistency here.

There are many other optionalities in scripts dir that are enabled due to platform abilities (e.g. different udev-like integrations for USB), and much other code too (docs, spellchecks, cgi all have huge but optional build- and/or run- time dependency footprints), so this client is not an outlier.

The WMNut widget was born and remained external though. This was actually a problem when adding CI for it, because packaged NUT in GHA Linux images was a decade old, and current features like verified support for #840 concept-renaming could not be used/tested unless a modern one was custom-built (every time) in the running image. Which is wasteful too.

I do have a packager hat elsewhere and agree with the woes and see that so many OS recipes have screenful-long incantations for the `configure` scripts to deliver a specific choice, and those lines may have to be revised for a new release (SEMVER debacle aside), but also I do have a user hat and want "as much as possible" built and delivered with least effort (auto modes that were in NUT before me); I think most of individual options are covered by `--with-all={yes,no,auto}` though, and can discuss if it is a bug or not when something else (not about docs that are another can of worms) is now non-deterministic by default. I think python module/UI are covered there, so one gets what they ask for - may be different as code evolves.

A few releases ago you wouldn't get say modbus or i2c drivers, the concept was not there, and now separate sub-packages with certain run-time dependencies may be in order (adding work for package maintainers, and highlighted in `UPGRADING.adoc`). Would you argue that if they were not present in v1 release, they should not be built by default (deps in build env permitting) years down the road, just because surprise for packagers? Would they even notice it was added? In any case they should adjust something (config? file lists? sub-package defs?) to deliver the results of that work if not ignoring them (or to ignore deliberately and with awareness - documenting why), so anyhow something would trigger their work.

jimklimov, 2025-08-31:
Huh, I mentioned Perl binding code as external prohect(s) - but there's in fact an in-tree Perl client module `scripts/perl/Nut.pm` (not `make install`'ed anywhere though at the moment... hmmm...)

bjornfor, 2025-09-05:
@jimklimov: Thanks for fixing!

gdt, 2025-09-08:
I'd like to point out that "bindings as optional component of main build" and "in separate repo" are separable.

It is possible to have  a repo where the py-bindings are in a subdir and one can build them with normal py-setup.   That way you get repo coherency and uncoupled builds for packaging.

That does mean that gui clients are also in subdirs, and you need to build/install nut, build/instal bindings, and then build/install GUI.  But ~all packaging systems are going to want those packaged separately anyway.