networkupstools/nut #3063: upsdrvctl.exe uses incorrect path separator on Windows

twoeater, 2025-08-20:
Description
I'm trying to run upsdrvctl.exe on a Windows system. However, the application seems to be using a forward slash (/) as a path separator instead of a backslash (\\), which is the standard for Windows.

This leads to a "No such file or directory" error because the full path C:\NUT-for-Windows-x86_64-SNAPSHOT-2.8.4.443-master\mingw64\sbin/usbhid-ups.exe is invalid on Windows.

Steps to Reproduce:

1.  Run the following command:
```shell
PS C:\NUT-for-Windows-x86_64-SNAPSHOT-2.8.4.443-master\mingw64\sbin> .\upsdrvctl.exe start
```

2. Observe the error message:
```shell
Network UPS Tools upsdrvctl.exe - UPS driver controller 2.8.4 release
Can't start C:\NUT-for-Windows-x86_64-SNAPSHOT-2.8.4.443-master\mingw64\sbin/usbhid-ups.exe: No such file or directory [지정된 파일을 찾을 수 없습니다. ]
```

Additional Context
I attempted to use the -r option to specify the path, but this did not resolve the issue. The application still seems to construct the path internally using the incorrect separator.

Example with -r option:
```shell
PS C:\NUT-for-Windows-x86_64-SNAPSHOT-2.8.4.443-master\mingw64\sbin> .\upsdrvctl.exe -r "C:\test\" start
```

The output remains the same, showing the invalid path with the forward slash:
```shell
Network UPS Tools upsdrvctl.exe - UPS driver controller 2.8.4 release
Can't start C:\NUT-for-Windows-x86_64-SNAPSHOT-2.8.4.443-master\mingw64\sbin/usbhid-ups.exe: No such file or directory [지정된 파일을 찾을 수 없습니다. ]
```

jimklimov, 2025-08-20:
Windows for a couple decades supports both slashes; not sure indeed about mixing them though. Probably a platform-specific "pathsep" macro or similar trick is needed. Can you experiment and PR a solution?

jimklimov, 2025-08-20:
Sorry just saw PR #3064.

Also, which Windows version did this happen on?