networkupstools/nut #3067: Missing data for APC UPS on Home assistant using NUT

lordwizzard, 2025-08-29:
With the latest update to Home Assistant. NUT is not reporting a lot of data. It does not matter whether using either snmp-ups or  usbhid-ups. The same pieces are missing. This was working prior to August 1, 2025

The UPS is a APC SU1000XL with a AP9606 network card.

I've add a screen shot of what snmp sees versus NUT (integration and the server add-on):

[image]

Log attached below:

[a0d7b954_nut_2025-08-29T01-44-09.255Z.log](https://github.com/user-attachments/files/22035927/a0d7b954_nut_2025-08-29T01-44-09.255Z.log)

jimklimov, 2025-09-08:
FWIW, attached logs indicate your HA runs NUT v2.8.0; the latest release (which was released Aug 8, 2025) is 2.8.4. So it is not quite a latest update there, as far as NUT project is concerned.

My random guess would be that the HA plugin update is broken, ignoring some readings that NUT apparently gave and may still give to you, but HA no longer understands.

lordwizzard, 2025-09-09:
Looks like this has been fixed with the HASSOS 16.2 release, both for snmp-ups and apcups-hid for alot of different UPSes

closed 2025-09-09 by lordwizzard

jimklimov, 2025-09-09:
FWIW, every time you typed the `usbhid-ups` driver name, it was very different and creative :)