networkupstools/nut #3079: [HCL] Salicru SPS 500 ONE supported by nutdrv_qx

alejsanc, 2025-09-07:
```
# upsc salicru

Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.6
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Voltronic-QS 0.09
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
input.voltage: 232.9
input.voltage.fault: 232.9
output.current.nominal: 1.0
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 232.9
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware.aux: PM-H
ups.load: 0
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665
```

```
# upscmd -l salicru
Instant commands supported on UPS [salicru]:

beeper.toggle - Toggle the UPS beeper
driver.killpower - Tell the driver daemon to initiate UPS shutdown; should be unlocked with driver.flag.allow_killpower option or variable setting
driver.reload - Reload running driver configuration from the file system (only works for changes in some options)
driver.reload-or-error - Reload running driver configuration from the file system (only works for changes in some options); return an error if something changed and could not be applied live (so the caller can restart it with new options)
driver.reload-or-exit - Reload running driver configuration from the file system (only works for changes in some options); exit the running driver if something changed and could not be applied live (so service management framework can restart it with new options)
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.quick - Start a quick battery test
```

jimklimov, 2025-09-08:
Thanks for the report. What stands out about this one to me is that there is a `salicru` subdriver for `usbhid-ups` (should have been in NUT v2.8.1 already), but it chose a Voltronic-QS instead. In fact, the USB VID:PID `0665:5161` in sources only points to a few devices that talk Megatec protocols, not USB HID. The `drivers/salicru-hid.c` goes with vendor `0x2e66` and several products...

Just in case, wondering how did you decide to use `usbhid-pus` here at all - just by the vendor+driver names in HCL table, or `nut-scanner` suggestion, or some forum, or experimental iterations, or... ? :)

jimklimov, 2025-09-08:
Got it, thanks for the updated reports :)

alejsanc, 2025-09-08:
For SPS ONE devices, the manufacturer provides ViewPower software. For other devices, it offers other software. These other devices must be those that work with usbhid-ups.