networkupstools/nut #3091: Incorrect voltage on CyberPower DX800E

pmalecek, 2025-09-16:
Hello,

I am using a CyberPower DX800E UPS and Debian distribution. NUT displays all the necessary information, but the battery voltage is significantly lower (approx. 7.8V) than in reality (approx. 13.7V) - measured with a multimeter.

Is there any way to fix this? Any patch?

Thanks.

> root@server:/etc/nut# upsc myups@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 1380
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 7.6
battery.voltage.nominal: 12
device.mfr: CPS
device.model: DX800E
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: DX800E
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 0
input.transfer.low: 0
input.voltage: 230.0
input.voltage.nominal: 230
output.voltage: 238.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 22
ups.mfr: CPS
ups.model: DX800E
ups.productid: 0501
ups.realpower.nominal: 450
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: 0
ups.vendorid: 0764

pmalecek, 2025-09-16:
I only use DEB packages within the repository to maintain a clean environment. After upgrading to Debian 13.1 (Trixie) and NUT version 2.8.1, no changes occurred.

Is there an official APT (Debian/Ubuntu) repository for NUT Server available somewhere? I haven't found anything usable. Thanks.

jimklimov, 2025-09-16:
Not at the moment, idea to use OBS was long backlogged somewhere; recently revived but not done yet.

pmalecek, 2025-09-16:
Unfortunately, despite our efforts, the new version does not solve the problem. It is strange that the battery voltage value is approximately 2/3 of the actual value.

> root@server:/usr/local/ups/bin# ./upsc myups@localhost
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 900
battery.runtime.low: 300
battery.status: 100%
battery.type: PbAcid
battery.voltage: 7.8
battery.voltage.nominal: 12
device.mfr: CPS
device.model: DX800E
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.interrupt_pipe_no_events_tolerance: -1
driver.parameter.pollfreq: 12
driver.parameter.pollinterval: 2
driver.parameter.port: /var/state/ups/usbhid-ups-myups
driver.parameter.product: DX800E
driver.parameter.productid: 0501
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.state: quiet
driver.version: 2.8.4
driver.version.data: CyberPower HID 0.84
driver.version.internal: 0.67
driver.version.usb: libusb-0.1 (or compat)
input.transfer.high: 0
input.transfer.low: 0
input.voltage: 230.0
input.voltage.nominal: 230
output.voltage: 240.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 120
ups.load: 38
ups.mfr: CPS
ups.model: DX800E
ups.productid: 0501
ups.realpower.nominal: 480
ups.status: OL
ups.timer.shutdown: -60
ups.timer.start: 0
ups.vendorid: 0764

`Network UPS Tools upsc 2.8.4 release
`

jimklimov, 2025-09-16:
Thanks... can you post a debug log from the driver startup? Maybe some fixup is needed (or breaks stuff, if already happens needlessly)?